mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #429: 128 X 64 resolution LED Matrix Displayed split

Jiwno, 2023-03-25:
Hi, I'm using 128 X 64 resolution LED Matrix and I have some displaying issue with it.
I've been trying all the examples in the library but it all displays split like the image.
The image is the result of "2_PatternPlasma.ino" file.
I changed  panel_height, panel_width, panel_number, and also defined pin_e to 32.
Down below is the code that I used and I just changed the values I mentioned right above.

[image]

/*
 * Portions of this code are adapted from Aurora: https://github.com/pixelmatix/aurora
 * Copyright (c) 2014 Jason Coon
 *
 * Portions of this code are adapted from LedEffects Plasma by Robert Atkins: https://bitbucket.org/ratkins/ledeffects/src/26ed3c51912af6fac5f1304629c7b4ab7ac8ca4b/Plasma.cpp?at=default
 * Copyright (c) 2013 Robert Atkins
 *
 * Permission is hereby granted, free of charge, to any person obtaining a copy of
 * this software and associated documentation files (the "Software"), to deal in
 * the Software without restriction, including without limitation the rights to
 * use, copy, modify, merge, publish, distribute, sublicense, and/or sell copies of
 * the Software, and to permit persons to whom the Software is furnished to do so,
 * subject to the following conditions:
 *
 * The above copyright notice and this permission notice shall be included in all
 * copies or substantial portions of the Software.
 *
 * THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS OR
 * IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY, FITNESS
 * FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT. IN NO EVENT SHALL THE AUTHORS OR
 * COPYRIGHT HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER LIABILITY, WHETHER
 * IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING FROM, OUT OF OR IN
 * CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER DEALINGS IN THE SOFTWARE.
 */
 

// HUB75E pinout
// R1 | G1
// B1 | GND
// R2 | G2
// B2 | E
//  A | B
//  C | D
// CLK| LAT
// OE | GND

/*  Default library pin configuration for the reference
  you can redefine only ones you need later on object creation

#define R1 25
#define G1 26
#define BL1 27
#define R2 14
#define G2 12
#define BL2 13
#define CH_A 23
#define CH_B 19
#define CH_C 5
#define CH_D 17
#define CH_E -1 // assign to any available pin if using two panels or 64x64 panels with 1/32 scan
#define CLK 16
#define LAT 4
#define OE 15

*/

#include <ESP32-HUB75-MatrixPanel-I2S-DMA.h>
#include <FastLED.h>

// Configure for your panel(s) as appropriate!
#define PANEL_WIDTH 128
#define PANEL_HEIGHT 64  	// Panel height of 64 will required PIN_E to be defined.
#define PANELS_NUMBER 1 	// Number of chained panels, if just a single panel, obviously set to 1
#define PIN_E 32

#define PANE_WIDTH PANEL_WIDTH * PANELS_NUMBER
#define PANE_HEIGHT PANEL_HEIGHT

// placeholder for the matrix object
MatrixPanel_I2S_DMA *dma_display = nullptr;

uint16_t time_counter = 0, cycles = 0, fps = 0;
unsigned long fps_timer;

CRGB currentColor;
CRGBPalette16 palettes[] = {HeatColors_p, LavaColors_p, RainbowColors_p, RainbowStripeColors_p, CloudColors_p};
CRGBPalette16 currentPalette = palettes[0];

CRGB ColorFromCurrentPalette(uint8_t index = 0, uint8_t brightness = 255, TBlendType blendType = LINEARBLEND) {
  return ColorFromPalette(currentPalette, index, brightness, blendType);
}

void setup() {
  
  Serial.begin(115200);
  
  Serial.println(F("*****************************************************"));
  Serial.println(F("*        ESP32-HUB75-MatrixPanel-I2S-DMA DEMO       *"));
  Serial.println(F("*****************************************************"));

  /*
    The configuration for MatrixPanel_I2S_DMA object is held in HUB75_I2S_CFG structure,
    pls refer to the lib header file for full details.
    All options has it's predefined default values. So we can create a new structure and redefine only the options we need 

    // those are the defaults
    mxconfig.mx_width = 64;                   // physical width of a single matrix panel module (in pixels, usually it is always 64 ;) )
    mxconfig.mx_height = 32;                  // physical height of a single matrix panel module (in pixels, usually almost always it is either 32 or 64)
    mxconfig.chain_length = 1;                // number of chained panels regardless of the topology, default 1 - a single matrix module
    mxconfig.gpio.r1 = R1;                    // pin mappings
    mxconfig.gpio.g1 = G1;
    mxconfig.gpio.b1 = B1;                    // etc
    mxconfig.driver = HUB75_I2S_CFG::SHIFT;   // shift reg driver, default is plain shift register
    mxconfig.double_buff = false;             // use double buffer (twice amount of RAM required)
    mxconfig.i2sspeed = HUB75_I2S_CFG::HZ_10M;// I2S clock speed, better leave as-is unless you want to experiment
  */

  /*
    For example we have two 64x64 panels chained, so we need to customize our setup like this

  */
  HUB75_I2S_CFG mxconfig;
  mxconfig.mx_height = PANEL_HEIGHT;      // we have 64 pix heigh panels
  mxconfig.chain_length = PANELS_NUMBER;  // we have 2 panels chained
  mxconfig.gpio.e = PIN_E;                // we MUST assign pin e to some free pin on a board to drive 64 pix height panels with 1/32 scan
  //mxconfig.driver = HUB75_I2S_CFG::FM6126A;     // in case that we use panels based on FM6126A chip, we can change that

  /*
    //Another way of creating config structure
    //Custom pin mapping for all pins
    HUB75_I2S_CFG::i2s_pins _pins={R1, G1, BL1, R2, G2, BL2, CH_A, CH_B, CH_C, CH_D, CH_E, LAT, OE, CLK};
    HUB75_I2S_CFG mxconfig(
                            64,   // width
                            64,   // height
                             4,   // chain length
                         _pins,   // pin mapping
      HUB75_I2S_CFG::FM6126A      // driver chip
    );

  */

  // OK, now we can create our matrix object
  dma_display = new MatrixPanel_I2S_DMA(mxconfig);

  // let's adjust default brightness to about 75%
  dma_display->setBrightness8(192);    // range is 0-255, 0 - 0%, 255 - 100%

  // Allocate memory and start DMA display
  if( not dma_display->begin() )
      Serial.println("****** !KABOOM! I2S memory allocation failed ***********");
 
  // well, hope we are OK, let's draw some colors first :)
  Serial.println("Fill screen: RED");
  dma_display->fillScreenRGB888(255, 0, 0);
  delay(1000);

  Serial.println("Fill screen: GREEN");
  dma_display->fillScreenRGB888(0, 255, 0);
  delay(1000);

  Serial.println("Fill screen: BLUE");
  dma_display->fillScreenRGB888(0, 0, 255);
  delay(1000);

  Serial.println("Fill screen: Neutral White");
  dma_display->fillScreenRGB888(64, 64, 64);
  delay(1000);

  Serial.println("Fill screen: black");
  dma_display->fillScreenRGB888(0, 0, 0);
  delay(1000);

  // Set current FastLED palette
  currentPalette = RainbowColors_p;
  Serial.println("Starting plasma effect...");
  fps_timer = millis();
}

void loop() {
  
    for (int x = 0; x < PANE_WIDTH; x++) {
            for (int y = 0; y <  PANE_HEIGHT; y++) {
                int16_t v = 0;
                uint8_t wibble = sin8(time_counter);
                v += sin16(x * wibble * 3 + time_counter);
                v += cos16(y * (128 - wibble)  + time_counter);
                v += sin16(y * x * cos8(-time_counter) / 8);

                currentColor = ColorFromPalette(currentPalette, (v >> 8) + 127); //, brightness, currentBlendType);
                dma_display->drawPixelRGB888(x, y, currentColor.r, currentColor.g, currentColor.b);
            }
    }

    ++time_counter;
    ++cycles;
    ++fps;

    if (cycles >= 1024) {
        time_counter = 0;
        cycles = 0;
        currentPalette = palettes[random(0,sizeof(palettes)/sizeof(palettes[0]))];
    }

    // print FPS rate every 5 seconds
    // Note: this is NOT a matrix refresh rate, it's the number of data frames being drawn to the DMA buffer per second
    if (fps_timer + 5000 < millis()){
      Serial.printf_P(PSTR("Effect fps: %d\n"), fps/5);
      fps_timer = millis();
      fps = 0;
    }
} // end loop

board707, 2023-03-25:
Please show the clear photo of the panel reverse side with readable text near HUB75 connector and panel ID

Jiwno, 2023-03-25:
[image]

Here!

board707, 2023-03-25:
What the ending of the panel ID in the lower part of the photo: "SOFT-1.875-..." ?
I think your panel use shift register for multiplexing the rows... this would explain why only three channels are used - A B C.
I don't know is your panel supported by this library.  It is question to maintainer - @mrfaptastic

mrfaptastic, 2023-03-25:
Not that I'm aware of. Never had a board with these chips.

DarrylStrong, 2023-03-25:
I encountered this with some modules a year or so back. I also modified the library is to handle these chips. FM6124 if memory serves me right.

Jiwno, 2023-03-25:
You mean that use "3_FM6126Panel" example? right?
I've already tried it but it just displays like the image below.
4 split change to 2 split.

[image]

DarrylStrong, 2023-03-26:
Do you have photographs of the ICs on the pcb?

mrfaptastic, 2023-03-26:
> I encountered this with some modules a year or so back. I also modified the library is to handle these chips. FM6124 if memory serves me right. 

Got any code that I could take a look at?

Jiwno, 2023-03-26:
> Do you have photographs of the ICs on the pcb?
It's written 'RUL5158C OAC52883' and 'CHIPONE ICN2037BP D10194NA"
[image]

Jiwno, 2023-03-26:
> > I encountered this with some modules a year or so back. I also modified the library is to handle these chips. FM6124 if memory serves me right.
> 
> Got any code that I could take a look at?

I only changed this part of FM6126 example.

// Output resolution and panel chain length configuration
#define PANEL_RES_X 128 	   // Number of pixels wide of each INDIVIDUAL panel module. 
#define PANEL_RES_Y 64 	   // Number of pixels tall of each INDIVIDUAL panel module.
#define PANEL_CHAIN 1      // Total number of panels chained one to another

mrfaptastic, 2023-03-26:
What does the simple test shapes look like? Only so I can understand how the panel does the row addressing better.

Jiwno, 2023-03-26:
> What does the simple test shapes look like? Only so I can understand how the panel does the row addressing better.

It looks like this, and I only changed the resolution part to res_x 128, res_y 64.

[image]

DarrylStrong, 2023-03-26:
> > I encountered this with some modules a year or so back. I also modified the library is to handle these chips. FM6124 if memory serves me right. 
> 
> Got any code that I could take a look at?

It is already in the library 😉

DarrylStrong, 2023-03-26:
Those ICs are shift register type drivers not 3 to 8 line demultiplexers.

DarrylStrong, 2023-03-26:
Those ICs should work in the same way as the SM5266 so set that type in mxconfig.

https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-DMA/issues/164

That is where I sorted that issue...

board707, 2023-03-26:
> I only changed this part of FM6126 example.

no, your panel has nothing to do with 6126 chip (and 6126 example)

Jiwno, 2023-03-26:
> Those ICs should work in the same way as the SM5266 so set that type in mxconfig.
> 
> #164
> 
> That is where I sorted that issue...

Thanks. But I have some compile error after I replaced the cpp file. 

Could you tell me what version are you using?

DarrylStrong, 2023-03-26:
I am using an old version which I have further modified for my specific use case.

Perhaps you could post up the error messages you get?

Jiwno, 2023-03-26:
this is the error message.

In file included from /Users/easyone/Documents/Arduino/libraries/ESP32_HUB75_LED_MATRIX_PANEL_DMA_Display/src/ESP32-HUB75-MatrixPanel-I2S-DMA.h:12,
                 from /private/var/folders/7g/6qwxccz9389842nzmbx43mbh0000gn/T/.arduinoIDE-unsaved2023226-5811-v2jvnz.jxnq/2_PatternPlasma/2_PatternPlasma.ino:58:
/Users/easyone/Documents/Arduino/libraries/ESP32_HUB75_LED_MATRIX_PANEL_DMA_Display/src/platforms/platform_detect.hpp:58:19: note: #pragma message: Compiling for original ESP32 (released 2016)
   #pragma message "Compiling for original ESP32 (released 2016)"
                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /Users/easyone/Documents/Arduino/libraries/FastLED/src/FastLED.h:75,
                 from /private/var/folders/7g/6qwxccz9389842nzmbx43mbh0000gn/T/.arduinoIDE-unsaved2023226-5811-v2jvnz.jxnq/2_PatternPlasma/2_PatternPlasma.ino:59:
/Users/easyone/Documents/Arduino/libraries/FastLED/src/fastspi.h:157:23: note: #pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output
 #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"
                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /Users/easyone/Documents/Arduino/libraries/ESP32_HUB75_LED_MATRIX_PANEL_DMA_Display/src/ESP32-HUB75-MatrixPanel-I2S-DMA.h:12,
                 from /Users/easyone/Documents/Arduino/libraries/ESP32_HUB75_LED_MATRIX_PANEL_DMA_Display/src/ESP32-HUB75-MatrixPanel-I2S-DMA 복사본.cpp:1:
/Users/easyone/Documents/Arduino/libraries/ESP32_HUB75_LED_MATRIX_PANEL_DMA_Display/src/platforms/platform_detect.hpp:58:19: note: #pragma message: Compiling for original ESP32 (released 2016)
   #pragma message "Compiling for original ESP32 (released 2016)"
                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /Users/easyone/Documents/Arduino/libraries/ESP32_HUB75_LED_MATRIX_PANEL_DMA_Display/src/ESP32-HUB75-MatrixPanel-I2S-DMA.h:12,
                 from /Users/easyone/Documents/Arduino/libraries/ESP32_HUB75_LED_MATRIX_PANEL_DMA_Display/src/ESP32-HUB75-MatrixPanel-I2S-DMA.cpp:3:
/Users/easyone/Documents/Arduino/libraries/ESP32_HUB75_LED_MATRIX_PANEL_DMA_Display/src/platforms/platform_detect.hpp:58:19: note: #pragma message: Compiling for original ESP32 (released 2016)
   #pragma message "Compiling for original ESP32 (released 2016)"
                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /Users/easyone/Documents/Arduino/libraries/ESP32_HUB75_LED_MATRIX_PANEL_DMA_Display/src/ESP32-HUB75-MatrixPanel-I2S-DMA.h:12,
                 from /Users/easyone/Documents/Arduino/libraries/ESP32_HUB75_LED_MATRIX_PANEL_DMA_Display/src/ESP32-HUB75-MatrixPanel-leddrivers.cpp:13:
/Users/easyone/Documents/Arduino/libraries/ESP32_HUB75_LED_MATRIX_PANEL_DMA_Display/src/platforms/platform_detect.hpp:58:19: note: #pragma message: Compiling for original ESP32 (released 2016)
   #pragma message "Compiling for original ESP32 (released 2016)"
                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/easyone/Documents/Arduino/libraries/ESP32_HUB75_LED_MATRIX_PANEL_DMA_Display/src/ESP32-HUB75-MatrixPanel-I2S-DMA.cpp: In member function 'bool MatrixPanel_I2S_DMA::allocateDMAmemory()':
/Users/easyone/Documents/Arduino/libraries/ESP32_HUB75_LED_MATRIX_PANEL_DMA_Display/src/ESP32-HUB75-MatrixPanel-I2S-DMA.cpp:60:113: error: 'PIXEL_COLOR_DEPTH_BITS' was not declared in this scope
 #define rowBitStructBuffSize        sizeof(ESP32_I2S_DMA_STORAGE_TYPE) * (PIXELS_PER_ROW + CLKS_DURING_LATCH) * PIXEL_COLOR_DEPTH_BITS
                                                                                                                 ^~~~~~~~~~~~~~~~~~~~~~
/Users/easyone/Documents/Arduino/libraries/ESP32_HUB75_LED_MATRIX_PANEL_DMA_Display/src/ESP32-HUB75-MatrixPanel-I2S-DMA.cpp:61:54: note: in expansion of macro 'rowBitStructBuffSize'
 #define frameStructBuffSize         ROWS_PER_FRAME * rowBitStructBuffSize
                                                      ^~~~~~~~~~~~~~~~~~~~
/Users/easyone/Documents/Arduino/libraries/ESP32_HUB75_LED_MATRIX_PANEL_DMA_Display/src/ESP32-HUB75-MatrixPanel-I2S-DMA.cpp:89:51: note: in expansion of macro 'frameStructBuffSize'
     size_t _frame_buffer_memory_required        = frameStructBuffSize * _num_frame_buffers;
                                                   ^~~~~~~~~~~~~~~~~~~
/Users/easyone/Documents/Arduino/libraries/ESP32_HUB75_LED_MATRIX_PANEL_DMA_Display/src/ESP32-HUB75-MatrixPanel-I2S-DMA.cpp:60:113: note: suggested alternative: 'PIXEL_COLOR_DEPTH_BITS_MAX'
 #define rowBitStructBuffSize        sizeof(ESP32_I2S_DMA_STORAGE_TYPE) * (PIXELS_PER_ROW + CLKS_DURING_LATCH) * PIXEL_COLOR_DEPTH_BITS
                                                                                                                 ^~~~~~~~~~~~~~~~~~~~~~
/Users/easyone/Documents/Arduino/libraries/ESP32_HUB75_LED_MATRIX_PANEL_DMA_Display/src/ESP32-HUB75-MatrixPanel-I2S-DMA.cpp:61:54: note: in expansion of macro 'rowBitStructBuffSize'
 #define frameStructBuffSize         ROWS_PER_FRAME * rowBitStructBuffSize
                                                      ^~~~~~~~~~~~~~~~~~~~
/Users/easyone/Documents/Arduino/libraries/ESP32_HUB75_LED_MATRIX_PANEL_DMA_Display/src/ESP32-HUB75-MatrixPanel-I2S-DMA.cpp:89:51: note: in expansion of macro 'frameStructBuffSize'
[... cut ...]
 #define getRowDataPtr(row, _dpth, buff_id) &(dma_buff.rowBits[row]->data[_dpth * dma_buff.rowBits[row]->width + buff_id*(dma_buff.rowBits[row]->width * dma_buff.rowBits[row]->color_depth)])
                                              ^~~~~~~~
/Users/easyone/Documents/Arduino/libraries/ESP32_HUB75_LED_MATRIX_PANEL_DMA_Display/src/ESP32-HUB75-MatrixPanel-I2S-DMA.cpp:932:39: note: in expansion of macro 'getRowDataPtr'
       ESP32_I2S_DMA_STORAGE_TYPE *p = getRowDataPtr(_y, color_depth_idx, back_buffer_id);
                                       ^~~~~~~~~~~~~

exit status 1

Compilation error: exit status 1

mrfaptastic, 2023-03-26:
Doesn't make any sense to me. Start from a clean library install, use the current release. Double check your wiring. 

If the 6126 config doesn't work then perhaps it's a new variant of panel.

tsctrl, 2023-03-26:
> [image]
> 
> Here!

> [image]
> 
> Here!

i have outdoor P4 80x40 icn 2037bp 1/10s panel. has this one merge to master? i could test once

board707, 2023-03-26:
@Jiwno  and @tsctrl  - are you the one same person? The matrix on your photos looks very similar

tsctrl, 2023-03-26:
sorry to interrupt. i take back what i said. this panel has different scale than the standard 32x64px which require all display to be remap to this resolution. hope this 80x40px board to be supported by @board707 library, but sadly it will run on stm arduino or rpi only, wish stm arduino could run on stmcubemx too, long discussion regarding this from me on the stm libs repo. maybe i will take a visit later for an update..

tsctrl, 2023-03-26:
@mrfaptastic ,

```
platform_detect.hpp:58:19: note: #pragma message: Compiling for original ESP32 (released 2016)
#pragma message "Compiling for original ESP32 (released 2016)"
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```
this message keep showing during compiling the codes, on my end around at least 3 times. it will be more on clean build. on every class that callling the virtual matrix class. could this message to be display once? probably issue on my compiler linker. not sure anyone else have this problem, not an issue i just complaining..

tsctrl, 2023-03-26:
@Jiwno 

this https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-DMA/blob/58610563ee7a7084db1a58fd715faea35904cf17/src/ESP32-HUB75-MatrixPanel-I2S-DMA.h#L246

as per @DarrylStrong suggested,
SM5266 IC changes was already in the latest library. you dont have to modify the library or replace .cpp files. only need to specify SM5266 type in mxconfig and retest.

mrfaptastic, 2023-03-27:
> @mrfaptastic ,
>
> ```
> platform_detect.hpp:58:19: note: #pragma message: Compiling for original ESP32 (released 2016)
> #pragma message "Compiling for original ESP32 (released 2016)" [...]

I have it there to let me know if I'm compiling for the right hardware target. I'll comment out the compile time message for the esp32 only.

Also, will delete your message and this response after you've read it as well as I'm trying to keep this git issue readable for future readers. Hopefully the SM5366 mxconfig parameter suggestion is the fix @Jiwno

Jiwno, 2023-03-27:
> nd @tsctrl - are you the one same person? The matrix on your photos looks very similar

No, but it seems similar though

Jiwno, 2023-03-27:
> @Jiwno
> 
> https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-DMA/blob/58610563ee7a7084db1a58fd715faea35904cf17/src/ESP32-HUB75-MatrixPanel-I2S-DMA.h#L246
> 
> as per @DarrylStrong suggested, SM5266 IC changes was already in the latest library. you dont have to modify the library or replace .cpp files. only need to specify SM5266 type in mxconfig and retest.
> 
> `mxconfig.driver = HUB75_I2S_CFG::SM5366;`

It doesn't work, unfortunately.
LED displays nothing...
But thank you for helping. I really appreciate.