mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #430: Feature request: Display scaling on multiple resolution

tsctrl, 2023-03-27:
hi @mrfaptastic ,

i would like to request for some sort of configuration for display scaling. not sure if it is possible or if this was currently supported, but  i am not able to get the current config to work on this feature. 
motivation: i have multiple display resolution like 32x64px single panel and 4 chained panels 32x64 which resulting 64x128px. since 64x128px is 2x the scale of 32x64px. i would like to reuse display design i have done for 32x64px without recreating new display design for 64x128px. so a configuration with display scaling of 2 will resulting larger display scaling for the same as smaller display without recreating design for larger display from scratch. this will open up more possibilities and option to reduce duplicates codes and lesser development effort. this will be similar as pixel manipulation for image resizing.

graphical representation as below:
[image]

tq!

mrfaptastic, 2023-03-27:
I get the idea, but this is an gfx library matter. This library is only focused on the electrical / DMA bitstream to a panel.

With that said, a wrapper class like the virtual matrix class could possibly do this, but all the scaling would be done by the adafruit or other gfx library.

board707, 2023-03-27:
> gfx libs does not controls the resolution and the chains. pixel mappings as such as pixel scaling are directly related to the hardware itseft 

I don't think that it related to hardware. For doing that you do not need control resolution and chains, it can be truly logical function - just map any point A(x,y) to 4 points 
```
A1(2*x, 2*y,  A2(2*x +1, 2*y),  A3(2*x, 2*y +1), A4(2*x+1, 2*y +1)
```

tsctrl, 2023-03-27:
thank you @board707,

@mrfaptastic wondering where this could be added. if the 2 is the scaling factor than this could be added as configuration option? similar as ic selector or frequency settings?  does it make sense?

could be a base x and y manipulation on prior of panel and mapping settings..

mrfaptastic, 2023-03-27:
> thank you @board707,
> 
> @mrfaptastic wondering where this could be added. if the 2 is the scaling factor than this could be added as configuration option? similar as ic selector or frequency settings?  does it make sense?
> 
> edit: could be a base x and y manipulation on prior of panel and mapping settings.. i am ok if this does not goes in library. i need assistance on how this could be added or better if possible to be added without library modification. i am still 50% agree if this is not hardware related. as it does set how many led to lit. 😆
> 
> got it! it is a way how to redundant the drawing from one to multiple pixel. yes it does or might not hardware related. but how to achieve this?

I have a soft spot for you @tsctrl . Will look into it, can add it to the virtual matrix panel class... BUT I have no guarantees on how it will actually look. Stuff might look totally distorted.

mrfaptastic, 2023-04-01:
Refer to #432 - The red text 'HJ' is after I used the new 'setScaleFactor(2)'

closed 2023-04-01 by mrfaptastic

mrfaptastic, 2023-04-01:
@tsctrl - If you can avoid it, please don't raise issues regarding to weird GFX output from using the scaling. What has been implemented is technically sound and what @board707 also suggested, but the underlying Adafruit_GFX library does the drawing.

tsctrl, 2023-04-01:
hi @mrfaptastic, understood. from the changes ive seen it was adding the x and y to the scaling factor which will increase the size or the lit to the x and y direction. which can cause distortion as mentioned.

`scaled_x_start_pos = x * _scale_factor;`

example: add 2 scale factor to the x axis for coordinate of a pixel at position 0, which will draw from 0 and 1 at x.

it is suppose to increase the x to the positive and negative direction. where by example above it should will draw at -0.5, 0 and +0.5 on x.

@board707 do you get the idea?

tsctrl, 2023-04-01:
hi @mrfaptastic,

just ignore my previous comment on this, it is what it is suppose to work based on the requested feature. what kind of awkward request is this btw.

i have tested the changes, image, gif works well, font are pixelated(like minecraft texture) expected result as small font have small resolution but scaled, still acceptable for this change. but one issue i have figure out the getTextBounds() function didnt work anymore.

i guess the getTextBounds() result should be adjusted as per the scale changes. if this possible, do you have suggestion on fow to fix this? or if this is not possible or too bad for this library, i am ok for this scaling changes to be reverted and discarded.

tsctrl, 2023-04-02:
hi @mrfaptastic,

is it possible to add display descaling too? to fit 64x32 display into 32x16 panels? allowing negative scaling?

mrfaptastic, 2023-04-02:
I will not be implementing any more non-core GFX related stuff as this goes beyond the scope of this library. Descaling sounds like just more work as what happens when there's only a '1px' width line (or text) and then one wants to descale it by 50%? It'll dissapear.

I will probably rename the setScaleFactor to setZoomFactor to get rid of any ambiguity. 

I suggest you make a custom wrapper library over this library to implement the relevant anti-aliasing / blending you will need

tsctrl, 2023-04-02:
yes, 1px in descale will resulting the text or line to be dissappear. is expected as descaling is reduce pixel density to at least the minimum pixel to equal the descale factor.

i am aware that this is beyond the scope of this project. i understand and accept the decision on this, the wrapper blending and interpolation suggested are overkill for my usecase. thank you again for enabling this scaling function in. thanks!