mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #432: Rotate display memory

TechnikFreak92, 2023-04-01:
Hello everyone and best regards, I'm the new guy

I have a problem, maybe you can help me.
 
According to the example "VirtualMatrixPanel.pdf" (in the Doc directory)
I have the example "Example 3) Vertical serpentine ‘S’ chain / stack"
constructed with 3 panels instead of 4 panels.

The whole thing is then rotated 90 degrees hardware moderately.
So from height 96 pixels, width 64 pixels (portrait) to (landscape) 64x96 pixels.

Now all I really have to do is run the command "virtualDisp->setRotation(1);"
enter before I write the text on the display.
However, it only works with the original width of 64 pixels.

If the command setRotation() does not use the whole memory, e.g. rotated 90 degrees,
i.e. from 96 high and 64 wide, to 64 high and 96 wide (Portrait -> Landscape)
[image]

mrfaptastic, 2023-04-01:
I'll look into this with #430

mrfaptastic, 2023-04-01:
Done

[image]

mrfaptastic, 2023-04-01:
Implemented:

```
// 1) Include key virtual display library
    #include <ESP32-VirtualMatrixPanel-I2S-DMA.h>

// 2) Set configuration
    #define PANEL_RES_X 64 // Number of pixels wide of each INDIVIDUAL panel module. 
    #define PANEL_RES_Y 32 // Number of pixels tall of each INDIVIDUAL panel module.

    #define NUM_ROWS 3 // Number of rows of chained INDIVIDUAL PANELS
    #define NUM_COLS 1 // Number of INDIVIDUAL PANELS per ROW
    #define PANEL_CHAIN NUM_ROWS*NUM_COLS    // total number of panels chained one to another

    /* Configure the serpetine chaining approach. Options are:
        CHAIN_TOP_LEFT_DOWN
        CHAIN_TOP_RIGHT_DOWN
        CHAIN_BOTTOM_LEFT_UP
        CHAIN_BOTTOM_RIGHT_UP

        The location (i.e. 'TOP_LEFT', 'BOTTOM_RIGHT') etc. refers to the starting point where 
        the ESP32 is located, and how the chain of panels will 'roll out' from there.

        In this example we're using 'CHAIN_BOTTOM_LEFT_UP' which would look like this in the real world:

        Chain of 4 x 64x32 panels with the ESP at the BOTTOM_LEFT:

        +---------+---------+
        |    4    |    3    |
        |         |         |
        +---------+---------+
        |    1    |    2    |
        |  (ESP)  |         |
        +---------+---------+      

    */
    #define VIRTUAL_MATRIX_CHAIN_TYPE CHAIN_TOP_LEFT_DOWN 

// 3) Create the runtime objects to use

    // placeholder for the matrix object
    MatrixPanel_I2S_DMA *dma_display = nullptr;

    // placeholder for the virtual display object
    VirtualMatrixPanel  *virtualDisp = nullptr;

/******************************************************************************
 * Setup!
 ******************************************************************************/
void setup() {
  
  delay(2000);
  Serial.begin(115200);
  Serial.println(""); Serial.println(""); Serial.println("");
  Serial.println("*****************************************************");
  Serial.println(" HELLO !");
  Serial.println("*****************************************************");

  /******************************************************************************
   * Create physical DMA panel class AND virtual (chained) display class.   
   ******************************************************************************/

  /*
    The configuration for MatrixPanel_I2S_DMA object is held in HUB75_I2S_CFG structure,
    All options has it's predefined default values. So we can create a new structure and redefine only the options we need

	Please refer to the '2_PatternPlasma.ino' example for detailed example of how to use the MatrixPanel_I2S_DMA configuration
  */

  HUB75_I2S_CFG mxconfig(
                PANEL_RES_X,   // module width
                PANEL_RES_Y,   // module height
                PANEL_CHAIN    // chain length
  );

  //mxconfig.driver = HUB75_I2S_CFG::FM6126A;     // in case that we use panels based on FM6126A chip, we can set it here before creating MatrixPanel_I2S_DMA object

  // Sanity checks
  if (NUM_ROWS <= 1) {
    Serial.println(F("There is no reason to use the VirtualDisplay class for a single horizontal chain and row!"));
  }

  // OK, now we can create our matrix object
  dma_display = new MatrixPanel_I2S_DMA(mxconfig);

  // let's adjust default brightness to about 75%
  dma_display->setBrightness8(192);    // range is 0-255, 0 - 0%, 255 - 100%

  // Allocate memory and start DMA display
  if( not dma_display->begin() )
      Serial.println("****** !KABOOM! I2S memory allocation failed ***********");

  // create VirtualDisplay object based on our newly created dma_display object
  virtualDisp = new VirtualMatrixPanel((*dma_display), NUM_ROWS, NUM_COLS, PANEL_RES_X, PANEL_RES_Y, VIRTUAL_MATRIX_CHAIN_TYPE);

  // So far so good, so continue
  virtualDisp->fillScreen(virtualDisp->color444(0, 0, 0));
  virtualDisp->drawDisplayTest(); // draw text numbering on each screen to check connectivity

 // delay(1000);

  Serial.println("Chain of 4x 64x32 panels for this example:");
  Serial.println("+---------+---------+");
  Serial.println("|    4    |    3    |");
  Serial.println("|         |         |");
  Serial.println("+---------+---------+");
  Serial.println("|    1    |    2    |");
  Serial.println("| (ESP32) |         |");
  Serial.println("+---------+---------+");

   // draw blue text
   virtualDisp->setFont(&FreeSansBold12pt7b);

   virtualDisp->setTextColor(virtualDisp->color565(0, 0, 255));

   // Red text inside red rect (2 pix in from edge)
   virtualDisp->drawRect(1,1, virtualDisp->width()-2, virtualDisp->height()-2, virtualDisp->color565(255,0,0));

   // White line from top left to bottom right
   virtualDisp->drawLine(0,0, virtualDisp->width()-1, virtualDisp->height()-1, virtualDisp->color565(255,255,255));

   virtualDisp->drawDisplayTest(); // re draw text numbering on each screen to check connectivity

   virtualDisp->setRotation(1);
   virtualDisp->setTextSize(1); 
   virtualDisp->setCursor(0, virtualDisp->height()- ((virtualDisp->height()-45)/2));    
   virtualDisp->print("ABCDFG");

   // Scale factor ! Everything will be double size now. including Text
    virtualDisp->setCursor(8, 20);
    virtualDisp->setScaleFactor(2);   
    virtualDisp->setTextColor(virtualDisp->color565(255, 0, 0));   
    virtualDisp->print("HJ");
   
}

void loop() {
  

} // end loop
```

[image]

closed 2023-04-01 by mrfaptastic

TechnikFreak92, 2023-04-02:
Seems to be working, thanks

commit 0a08debc (2023-04-01): Close #432 #430

Implement VirtualDisplayClass rotation (and let AdaFruit_GFX) be aware of it as well for font printing.

Ability to 'scale' up (zoom) display as a quick and dirty magnification mode.