mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #433: Huidu WF2 / HD-WF2 ESP32-S3 controller card

hn, 2023-04-06:
The [Huidu HD-WF2](https://www.huidu.cn/product_131.html) is a cheap LED Matrix controller card based on an 8MB ESP32-S3, a [Datasheet](https://huidu-cn.oss-ap-southeast-1.aliyuncs.com/eData/File/Specifications/Single_color/WFX_series/HD-WF2%20Specification%20V6.1.1.pdf) is available for download. It can be found in various cheap 128x32 displays sold on Amazon (e.g. brand name 'Leadleds').

The PCB has a serial port and GPIO0 exposed (label silkscreend on PCB back), so it's quite easy to up- and download firmware.

The pin config is as follows:

```
#define R1_PIN 2
#define R2_PIN 3
#define G1_PIN 6
#define G2_PIN 7
#define B1_PIN 10
#define B2_PIN 11
#define A_PIN 39
#define B_PIN 38
#define C_PIN 37
#define D_PIN 36
#define E_PIN 21
#define LAT_PIN 33
#define CLK_PIN 34
#define OE_PIN 35

#define PANEL_RES_X 64     // Number of pixels wide of each INDIVIDUAL panel module. 
#define PANEL_RES_Y 32     // Number of pixels tall of each INDIVIDUAL panel module.
#define PANEL_CHAIN 2      // Total number of panels chained one to another

HUB75_I2S_CFG::i2s_pins _pins = {R1_PIN, G1_PIN, B1_PIN, R2_PIN, G2_PIN, B2_PIN, A_PIN, B_PIN, C_PIN, D_PIN, E_PIN, LAT_PIN, OE_PIN, CLK_PIN};
HUB75_I2S_CFG mxconfig(
  PANEL_RES_X,   // module width
  PANEL_RES_Y,   // module height
  PANEL_CHAIN,   // Chain length
  _pins          // pin mapping
);

Basic tests done, LED matrix seems to work fine. Currently having WiFi issues but this seems to be a different story.

```

DarrylStrong, 2023-04-06:
That PCB has 2 HUB75 ports, how are they mapped/addressed?

mrfaptastic, 2023-04-06:
Would be interesting to see if this library is what this product uses.

hn, 2023-04-06:
I have an 8MB dump of the stock firmware. I can't post it to the public because it may contain intelectual property of Huidu.

From a quick string grep: based on ESP-IDF, no obvious signs of other non-core libraries.

I think Huidu uses some proprietary LED matrix "operating system" for their products. I actually never used it, so when Amazon delivered the package, I directly unscrewed the case and flashed via serial port.

mrfaptastic, 2023-04-06:
How much are one of these controllers?

hn, 2023-04-06:
On AliExpress, the WF2 is about 15 €/$ and I just saw they also have a 4 port version (WF4) for roughly the same price (uh?).

The complete 128x32 display (2 panels, metal case, controller and power supply) was about 140€ on Amazon.

mrfaptastic, 2023-04-07:
The specs in the PDF say it drive a chain up to 768*64 px...

I'd love to see how it does that. Would be interesting to see with an oscilloscope.

Kamal-Sonani, 2023-04-07:
Can you post that controller's snap shot?

I don't think that WF2 have Esp32S3

board707, 2023-04-07:
> The specs in the PDF say it can drive a chain up to 768*64 px...

Pixel figures means nothing without of color depth.

hn, 2023-04-07:
Ok guys, if you want it, some PCB pron for you ;-)

WF2 V7.1.0-2, ESP32-S3, GND-RX-TX-5V serial port below MCU, GPIO0 and EN (didn't test EN pin) next to it:

[image]

Full PCB inside panel housing (panel label P4-2121-64*32-16S-HL1.2, 2PS688B), serial port connected with jumper cables:

[image]

Stock firmware boot log (no more output after this):

```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0xb (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd0108,len:0x1730
load:0x403b6000,len:0xbcc
load:0x403ba000,len:0x3304
entry 0x403b6250
I (24) boot: ESP-IDF v4.4-274-g09c8eef954-dirty 2nd stage bootloader
I (25) boot: compile time 15:06:14
I (25) boot: chip revision: 0
I (28) boot.esp32s3: Boot SPI Speed : 80MHz
I (33) boot.esp32s3: SPI Mode       : DIO
I (38) boot.esp32s3: SPI Flash Size : 8MB
I (42) boot: Enabling RNG early entropy source...
I (48) boot: Partition Table:
I (51) boot: ## Label            Usage          Type ST Offset   Length
I (59) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (66) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (73) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (81) boot:  3 ota_0            OTA app          00 10 00010000 00100000
I (88) boot:  4 ota_1            OTA app          00 11 00110000 00100000
I (96) boot:  5 app_data         unknown          40 06 00210000 00500000
I (103) boot: End of partition table
I (108) esp_image: segment 0: paddr=00110020 vaddr=3c0a0020 size=1150ch ( 70924) map
I (129) esp_image: segment 1: paddr=00121534 vaddr=3fc9a2a0 size=0414ch ( 16716) load
I (133) esp_image: segment 2: paddr=00125688 vaddr=40378000 size=0a990h ( 43408) load
I (144) esp_image: segment 3: paddr=00130020 vaddr=42000020 size=95548h (611656) map
I (254) esp_image: segment 4: paddr=001c5570 vaddr=40382990 size=0790ch ( 30988) load
I (261) esp_image: segment 5: paddr=001cce84 vaddr=50000000 size=00010h (    16) load
I (268) boot: Loaded app from partition at offset 0x110000
I (268) boot: Disabling RNG early entropy source...
```

mrfaptastic, 2023-04-07:
Good information. Even if I didn't want a hub controller, it's a decent way to get an esp32-s3 board with the gpio pins exposed.

I guess there's a real time clock connected to one of the pins given the battery.

hn, 2023-04-07:
I've absolutely no clue about LED panels or controllers or if there are better options on the market. For this thing, I had an alternative open-source hardware running on the device about two hours after the parcel arrived, so I'm pretty happy with my purchase. Thanks again for your work on this library!

And yes, it has a buffered RTC, time is correct even after power-cut.

mrfaptastic, 2023-04-07:
It's a good find so thanks for sharing. Best part is it's actually quite price competitive to buying an S3 devkit.

I'm going to buy one.

hn, 2023-04-07:
The HUB75E ports are routed through 74hc245 transcievers (likely for 3v3-5v level shifting) and I _think_ you might not be able to use them as inputs (so they are more GPO rather than GPIO ...). Only if you want to use it for something other than driving LED panels.

You see, I'm not a Huidu sales representative ;-)

mrfaptastic, 2023-04-07:
> You see, I'm not a Huidu sales representative ;-)

Hahahaha!

Also. The Huidu software must do GPIO bit banging using a CPU as the LCD mode can only do a single 16bit channel at once, which this library uses.

So one will never be able to use DMA and drive more than one of the hub75e ports.

Basically, their software that drives the panels must be more akin to PxMatrix library. So you basically upgraded by using this library 🤣

tavalin, 2023-04-10:
@mrfaptastic assuming these don't come with the 16MB PSRAM, is there any tangible performance advantages of this over a normal ESP32?

@hn have you got a link to any sellers with this version of the board? A lot of the sellers on AliExpress have pictures of old versions of the PCB and you can't tell from the photos which ESP32 version is in them

hn, 2023-04-11:
@tavalin I bought mine as part of a complete LED display including case and power supply, please see very first post.

hn, 2023-04-16:
Added pin mapping for WF2 port X2, see update to top post. Pins A/B/C/D/OE/CLK/LAT seem to be common with both X1 and X2.

I'm currently unable to quickly find E pin for port X2. Strange.

ergindemir, 2023-04-17:
> > The HUB75E ports are routed through 74hc245 transcievers (likely for 3v3-5v level shifting) and I _think_ you might not be able to use them as inputs (so they are more GPO rather than GPIO ...). Only if you want to use it for something other than driving LED panels.
> 
> Very good point. With that said I've ordered myself the WF1 because I like cheap. Hopefully it's an S3 device as well as I can use it to continue testing S3 support without the need for jumper cables. Not sure if it'll come with any psram though.

My wf1 comes with a Chinese microcontroller. It costs less than $5, so s3 is impossible.

mrfaptastic, 2023-04-17:
Damn. Looks like I wasted 5 bucks then !

tavalin, 2023-04-19:
I have received my HD-WF2 from the seller I Iinked to previously and can confirm it has an ESP32-S3 🙂

@mrfaptastic is assuming no octal PSRAM, is there any performance benefit over a normal ESP32?

@hn how did you flash new firmware? Using something like an FTDI on the  pins next to the power terminal block?

hn, 2023-04-19:
@tavalin Yes, a 3v3 usb-serial converter. I updated https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-DMA/issues/433#issuecomment-1500208556 with more pinout infos. The link you mentioned is dead, can you name the seller here again?

mrfaptastic, 2023-04-23:
@hn @ergindemir 

Recieved my $5 WF1, which is actually based on the ESP32-S2. Main practical drawback to the WF2 (other than S2 vs S3) is there are no obvious pins exposed to flash the device via Serial RX/TX, however there is GPIO0 exposed to put the ESP32 into download mode.

Found the key traces though.

I'm quite impressed. Probably a cheap way now to get up and running with a HUB75E panel without having to design a custom PCB and stuff around with wiring. 

Wrote some code, which also uses the RTC. Will work on it to actually display that to screen etc. Just a functional test of bits at the moment.

https://github.com/mrfaptastic/HD-WF1-LED-MatrixPanel-DMA

mrfaptastic, 2023-04-23:
> @mrfaptastic assuming these don't come with the 16MB PSRAM, is there any tangible performance advantages of this over a normal ESP32?

Nope. Just an easier way to connect to an hub75e panel really.  Downside is being restricted to only really being able to use it for hub 75e driving.

ergindemir, 2023-04-23:
Did you determine the gpio pin for the button?

hn, 2023-04-24:
@mrfaptastic Nice. The I2C_BM8563 was missing here, double-checked: pins are the same for the WF2.

Have you got OTA working so you don't have to fiddle with the pins all the time? When I upload a firmware via OTA (HTTPUpdateServer.h), a message appears something like "Read Flash failed".

mrfaptastic, 2023-04-24:
Yep. Let me update my GitHub sketch. I use "ElegantOTA" library to do the work for me.

mrfaptastic, 2023-04-25:
I have updated my sketch at the other git repository. Feel free to contribute as well.

Using ElegantOTA amongst other libraries and it's working perfectly. 

[image]

```
[209213][V][Parsing.cpp:389] _parseForm(): PostArg FileName: firmware
[209218][V][Parsing.cpp:394] _parseForm(): PostArg Name: firmware
[209225][V][Parsing.cpp:405] _parseForm(): PostArg Type: application/octet-stream
[209232][V][Parsing.cpp:435] _parseForm(): Start File: firmware Type: application/octet-stream
Update Received: firmware
[209250][D][Updater.cpp:133] begin(): OTA Partition: app1
Update Success: 1156656
Rebooting...
ESP-ROM:esp32s2-rc4-20191025
```

GrimorJander, 2023-05-15:
How can we plug sensors on WF2 ?? [https://www.youtube.com/embed/iWv9zkH-qFs?start=80](https://www.youtube.com/embed/iWv9zkH-qFs?start=80)
If we flash custom firm with FTDI adapter, can we restore back to  original software??

mrfaptastic, 2023-05-15:
You can't as the HUB75 output pins are output only.

Using "esptool" you can download the flash as a binary image and write it back... But don't ask me how as I haven't done that myself

GrimorJander, 2023-05-18:
> You can't as the HUB75 output pins are output only.
> 

I see, and probably other version of Huidu boards has free GPIOs pins for sensors connection.
Like RTC I2C lines, that are wired to GPIO 41 and 42 on the pcb.
Does the USB port read update packages only? or does it read media files too?
I can´t see any update or OTA on the Huidu site

mrfaptastic, 2023-05-19:
The USB port is just connected to the USB capable pins of the esp. Can do whatever your want in software.

E.g. For the WF1 its pins 19 & 20

GrimorJander, 2023-05-19:
> The USB port is just connected to the USB capable pins of the esp. Can do whatever your want in software.
> 
> E.g. For the WF1 its pins 19 & 20

I see. I'm waiting my WF2 to arrive. Meanwhile, I just install Esptool and now I can download and restore binary image s from esp32 S2, as you said early.  Don't know if will work with S3 aswell

mrfaptastic, 2023-05-19:
Unless you want to do some serious micro soldering on the esp32-s3 chip, you'll be unable to expose other GPIOs.

But whatever has been documented so far is what you can do.

That is:

Set and use the RTC 
Drive a hub75 panel

hn, 2023-05-20:
@GrimorJander It might be possible to connect an (i2c-)sensor to the USB pins. That would be an unusual connector for i2c, but if it works ... :)

GrimorJander, 2023-05-22:
WF2 arrived. I'm trying to make it run with "ledart" app but still no luck with 3 different panels. 2 of then 64x32 and one 128x64.
Only show some rambdom horizontal lines and auto recognition function doesn't find the correct setting for my panels.

mrfaptastic, 2023-05-22:
Are you using the pin mapping @hn provided?

GrimorJander, 2023-05-22:
> Are you using the pin mapping @hn provided?

I'm running stock firmware, running data wire
[image]
 from 75EX1 port to hub75 in of the pannel

GrimorJander, 2023-05-23:
My bad. I was plugging the flex hub75 cable in the wrong way

GrimorJander, 2023-05-24:
I must say that I am impressed with the original software being very powerful for such a small processor.
I am using a single panel 128x64 with scan rate 1/30 and it moves freely. The ease of being able to modify the content from the phone in real time, make different programs and custom overlays, makes everything easier. (especially for someone like me who doesn't know how to program)

mrfaptastic, 2023-05-25:
> I must say that I am impressed with the original software being very powerful for such a small processor.
> I am using a single panel 128x64 with scan rate 1/30 and it moves freely. The ease of being [...]

Agreed. If you just want to display text with effects then the the card is great as it is!

GrimorJander, 2023-06-05:
I continue to delve deeper into the original software and I have seen that the "HD2020" pc application is much more complete than "Ledart" for smartphones. My only real complaint with this system is the little bit depth of color. There are only 8 colors available (3bit?) so any jpg or bmp image is unusable unless it is a low color pixelart or Logo. I have tried to capture application update packages but they are only available for the HD60 model which is incompatible (it is the predecessor to WF)

GrimorJander, 2023-06-15:
> 
> Using "esptool" you can download the flash as a binary image and write it back... But don't ask me how as I haven't done that myself

I can confirm that the original software can be restored even with a modified version. I just changed the SSID of my module.
[image]

beta-tester, 2023-07-14:
does sombody has already one of those HD-WF4 boards?
[image]

[HD-WF4](https://www.huidu.cn/product_132.html)
[HD-WF4 Specification V6.0.1](https://huidu-cn.oss-ap-southeast-1.aliyuncs.com/eData/File/Specifications/Single_color/WFX_series/HD-WF4%20Specification%20V6.0.1.pdf)

any information about how the ports are connected?
because it is written:
> HD-WF4 is equipped with 2-wire hub75e port.

- is each of the 4 ports independend to each others?
- or are only 2 ports independen and the others are clones of each independen others?

GrimorJander, 2023-07-16:
> > HD-WF4 is equipped with 2-wire hub75e port.
> 
> * is each of the 4 ports independend to each others?
> * or are only 2 ports independen and the others are clones of each independen others?

WF2 has two Hub75 ports for two lines of 32 leds tall. So maximun is 64 pixels tall .
WF4 probably drives 4 lines of 32 pixels/leds each one.  I think 128 px is the taller you can go with this board

mrfaptastic, 2023-07-16:
There's no point buying a WF4 or even a WF2 and use this library because your can't drive more than one chain using DMA.

The software that comes with these cards bitbangs the GPIO output using software and hence only a colour depth of 3 bits, and being able to bit bang 30+ GPIOs at once.

beta-tester, 2023-07-16:
> There's no point buying a WF4 or even a WF2 and use this library because your can't drive more than one chain using DMA.

good point.

you mean they drive LED panels the ESP32-S in software instead of DMA with their stock firmware?

and i thought they are using DMA but with 32bit or 64bit width or DMA per CPU. you see i have no clue how the ESP32-S and your library is working. (shame on me)

would it be possible to drive 4x RGB LED panels with 64x64 picwls with an ESP32-S3-WROOM-1U-N16R8 with 8Mbyte PSRAM (octal) at 60Hz frame rate at 6bit per pixel?

hn, 2023-07-20:
I just pushed https://github.com/hn/esphome-configs/blob/master/led-matrix-hub75-huidu-wf2/led-matrix-hub75-huidu-wf2.yaml which makes it super easy to replace the WF2 stock firmware. Thanks to @TillFleisch for the ESPhome wrapper code!

Kalrkloss, 2023-08-22:
Does anybody know where to get the V7? All offers on Aliexpress show pictures of the V6, with clearly anything else than an ESP32.

rigorka, 2023-08-22:
> Does anybody know where to get the V7? All offers on Aliexpress show pictures of the V6, with clearly anything else than an ESP32.

Talk to the sellers. Most have V7 these days despite of showing the old V6 pics online. Just ask the seller before buying.

rwahyur, 2023-08-25:
> The [Huidu HD-WF2](https://www.huidu.cn/product_131.html) is a cheap LED Matrix controller card (with 2 HUB75E ports, labeled X1 and X2) based on an 8MB ESP32-S3, a [Datasheet](https://huidu-cn.oss-ap-southeast-1.aliyuncs.com/eData/File/Specifications/Single_color/WFX_series/HD-WF2%20Specification%20V6.1.1.pdf) is available for download. It can be found in various 128x32 displays sold on Amazon (e.g. 'Leadleds P4 WiFi LED shield', ASIN B08XYT1JDL).
> 
> The PCB has a serial port and GPIO0 exposed (label silkscreend on PCB back), so it's quite easy to up- and download firmware. There also is a USB-A port available and [USB OTG](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-guides/usb-otg-console.html) may work, didn't test that.
> 
> The pin config (reverse engineered) is as follows:
> 
> ```
> #define WF2_X1_R1_PIN 2
> #define WF2_X1_R2_PIN 3
> #define WF2_X1_G1_PIN 6
> #define WF2_X1_G2_PIN 7
> #define WF2_X1_B1_PIN 10
> #define WF2_X1_B2_PIN 11
> #define WF2_X1_E_PIN 21
> 
> #define WF2_X2_R1_PIN 4
> #define WF2_X2_R2_PIN 5
> #define WF2_X2_G1_PIN 8
> #define WF2_X2_G2_PIN 9
> #define WF2_X2_B1_PIN 12
> #define WF2_X2_B2_PIN 13
> #define WF2_X2_E_PIN -1        // Currently unknown, so X2 port will not work (yet) with 1/32 scan panels
> 
> #define WF2_A_PIN 39
> #define WF2_B_PIN 38
> #define WF2_C_PIN 37
> #define WF2_D_PIN 36
> #define WF2_OE_PIN 35
> #define WF2_CLK_PIN 34
> #define WF2_LAT_PIN 33
> 
> #define WF2_BUTTON_TEST     17  // Test key button on PCB, 1=normal, 0=pressed
> #define WF2_LED_RUN_PIN     40  // Status LED on PCB
> #define WF2_BM8563_I2C_SDA  41  // RTC BM8563 I2C port
> #define WF2_BM8563_I2C_SCL  42
> 
> #define PANEL_RES_X 64     // Number of pixels wide of each INDIVIDUAL panel module. 
> #define PANEL_RES_Y 32     // Number of pixels tall of each INDIVIDUAL panel module.
> #define PANEL_CHAIN 2      // Total number of panels chained one to another
> 
> HUB75_I2S_CFG::i2s_pins _pins_x1 = {WF2_X1_R1_PIN, WF2_X1_G1_PIN, WF2_X1_B1_PIN, WF2_X1_R2_PIN, WF2_X1_G2_PIN, WF2_X1_B2_PIN, WF2_A_PIN, WF2_B_PIN, WF2_C_PIN, WF2_D_PIN, WF2_X1_E_PIN, WF2_LAT_PIN, WF2_OE_PIN, WF2_CLK_PIN};
> HUB75_I2S_CFG::i2s_pins _pins_x2 = {WF2_X2_R1_PIN, WF2_X2_G1_PIN, WF2_X2_B1_PIN, WF2_X2_R2_PIN, WF2_X2_G2_PIN, WF2_X2_B2_PIN, WF2_A_PIN, WF2_B_PIN, WF2_C_PIN, WF2_D_PIN, WF2_X2_E_PIN, WF2_LAT_PIN, WF2_OE_PIN, WF2_CLK_PIN};
> 
> HUB75_I2S_CFG mxconfig(
>   PANEL_RES_X,   // module width
>   PANEL_RES_Y,   // module height
>   PANEL_CHAIN,   // Chain length
>   _pins_x1       // pin mapping for port X1
> );
> ```
> 
> Basic tests done, LED matrix seems to work fine. Currently having WiFi issues but this seems to be a different story.

About wifi issue, you can try to set i2s speed to 20Mhz.
Change the div_num to 2 on dma.cpp

rwahyur, 2023-08-25:
> The [Huidu HD-WF2](https://www.huidu.cn/product_131.html) is a cheap LED Matrix controller card (with 2 HUB75E ports, labeled X1 and X2) based on an 8MB ESP32-S3, a [...]

how do you find the wiring?

hn, 2023-08-26:
@rwahyur Well, no magic involved, with a multimeter and a loop like `digitalWrite(i, HIGH); delay(1000); digitalWrite(i, LOW);`.

I recommend using the ESPHome solution. WiFi, flashing, everything works out of the box.

GrimorJander, 2023-08-31:
> I just received a HD-WF2 from this seller: https://www.aliexpress.com/item/1005002258734810.html, and it is version V7.1.0-2. It has an ESP32-S3 and 8MB of flash.
> 
> And BTW: Flashing is even easier than I thought. Just pull GPIO 0 to ground, and then connect the board to the PC via an USB-A to A cable (yes, those things exist, although they shouldn't). The board will then appear as "USB JTAG/serial debug unit (COMxx)", and you can flash the board using a web installer (I used https://wled-install.github.io).

That's way is even easier to flash anything. First time I dump the original software, only have to shortcut gpio 0 and ground and Esptool could find the COM port from USB port. Later I solder header pins to the serial port and dump again from a UART-TTL adapter and dump again with same result (same dumped image)

hn, 2023-08-31:
@Kalrkloss Nice, thanks for reporting! I updated my top post accordingly (OTG update).

Kalrkloss, 2023-08-31:
[image]

Kalrkloss, 2023-08-31:
BTW, you need to short GPIO0 to GND only when your firmware reconfigures the USB port, which the original firmware seems to do, because when connecting the board to a PC, it appears as JTAG for a short time, and then dissappears. I assume, that the originaol firmware configures the USB as host, to be able to read pictures from USB sticks.
After flashing a firmware, that doesn't reconfigure the USB, you can leave GPIO0 alone, the USB port will always stay in JTAG mode.

mrfaptastic, 2023-08-31:
Just a reminder: Using the DMA library the maximum parallel data is 2 bytes (16 bits/GPIOs) 

So it is not possible to drive both HUB75 ports using DMA at the same time.

rwahyur, 2023-09-15:
[image]

rwahyur, 2023-09-15:
[image]

rwahyur, 2023-09-15:
[image]

[16 more replies]