mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #434: Indexing error in x axis (SM5166PC Chipset)

Jiwno, 2023-04-07:
Hi, I have an issue while using the led matrix with the library.

I'm using a 96x48 pixel matrix with SM5166PC chipset and have index error in x axis.

As you can see in the image (0, 0) of the matrix doesn't start from left top corner.

I found that the index of x axis starts with 32-63 and then goes to 0-63.

32 to 63 index are repeated.

Is there any ways to set indexes of the matrix again or any other ways to solve this problem?
[image]

DarrylStrong, 2023-04-07:
Do you have a photo of the rear of the pcb?

Jiwno, 2023-04-07:
> Do you have a photo of the rear of the pcb?

[image]
Here!!

But when I tested the example code '1_SimpleTestShapes', index seemed fine(after I defined pin e to 32 and I also defined pin e for the plasma example code as well)

Like this.
[image]

mrfaptastic, 2023-04-07:
So SM5166 compatibility works then? Looks good to me.

Jiwno, 2023-04-07:
Yeah, index aligned well on shape test example code but not with the plasma code.

I don't know why but I think it works fine though

mrfaptastic, 2023-04-24:
It could be a bug with the pattern plasma code to be honest.

BaiTian6641, 2023-06-16:
I have the same issue with 96 x 48 panel and none of example code cannot display properly. However, there is a modify version of ESP32-HUB75-MatrixPanel-DMA library, that library seem work fine, but it only support ESP32.
https://www.waveshare.com/wiki/RGB-Matrix-P2.5-96x48-F#Demo_2

mrfaptastic, 2023-06-16:
> https://www.waveshare.com/wiki/RGB-Matrix-P2.5-96x48-F#Demo_2

Good find. It would have been nice if they could have contributed to this library than make a custom copy.

BaiTian6641, 2023-06-16:
[image]

So, this is what panel display when I using none modified library. (Example is the BouncingSquares)

BaiTian6641, 2023-06-16:
[image]

For simple test shape example.

BaiTian6641, 2023-06-16:
I make modify on waveshare's library, using current esp32_i2s_parallel_dma.hpp and esp32_i2s_parallel_dma.cpp. And result is this:
[image]
This is SimpleTestShapes example.
[image]
This is BouncingSquares example.

BaiTian6641, 2023-06-17:
After set the invert_pclk to true in the gdma_lcd_parallel16.hpp, it works! @mrfaptastic @Jiwno 
[image]
[image]
[image]
[image]
So, here are some examples. The AuroraDemo example and BouncingSquares  example.

> Hi, I have an issue while using the led matrix with the library.
>
> I'm using a 96x48 pixel matrix with SM5166PC chipset and have index error in x axis.
>
> As you can see in the image (0, 0) of the [...]

mrfaptastic, 2023-06-23:
You found a bug that seemingly hasn't been detected for quite some time since I made changes to this library to include ESP32-S3 support.

You shouldn't need to edit the .hpp file to invert the clock (but you had to due toa bug).

You can pass to the `mxconfig.clkphase = true;` as part of the library config on initialisation.

closed 2024-01-09 by mrfaptastic