mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #435: Sometimes display doesn't start on ESP32-S3

noflip95, 2023-04-08:
Hi,

This is a very strange bug. 

Most of the time (9 times out of 10) everything works perfectly, but sometimes the display does not start, while the ESP32-S3 continu to work normally.

Context:
- Arduino IDE
- Lib. release **3.0.8**
- YD-ESP32-23 module from www.vcc-gnd.com (ESP32-S3-N16R8 / ESP32-S3-WROOM-1) powered by a stable independant 5V
- DMD : two panels P3 64x32 = 128x32 / separate power 30W / driven by 74HCT245 level adapter
- tested on single/dual core, with/without USB CDC interface (the issue is always the same whatever the configuration)
- classic pinout, no reserved pin used (also tested with a different pin configuration)
- "driver" fm6124 used (nice to track behaviour on HUB75 !).
- Several devices tested to avoid a deffective component. The issue happen on all five modules I have.
- SPIRAM not used (disabled in IDE Arduino).

About one time on ten, the ESP32-S3 start normaly, except that I got no display.
Using the debug mode, the initialisation sequence report no error. All look normal, and messages are exactly the same as when the display works.

For investigations, I connect a logic analyser (LA1010) on HUB75:
- when the display works, I can see the initialization sequence (fm6124 - values for reg11/12 are correctly transmitted), then follow the classic multiplex RGB/LAT/CLK/OE.
- when the display don't work, I see the fm6124 initialization sequence, but after ... nothing !

So, for me, the issue is in the start of the DMA / LCD-mode. As I can see the fm6124 init seq., I can exclude an issue from hardware/power/panels. If the multiplex RGB/LAT/CLK/OE is absent after the fmt6124 sequence, it means that there is something wrong with the DMA/LCD-Mode, but what ?

Pressing the boot button on module doesn't solve the problem. Sometimes, pressing it one time, and the display is active again. But sometimes, I need to press button five to ten times to get the display reactived.

Here is the log of the start, when the display doesn't works.
As you can see, there is no error message reported, nothing abnormal... but the display doesn't start !

16:38:23.686 -> ESP-ROM:esp32s3-20210327
16:38:23.733 -> Build:Mar 27 2021
16:38:23.733 -> rst:0x1 (POWERON),boot:0x18 (SPI_FAST_FLASH_BOOT)
16:38:23.733 -> SPIWP:0xee
16:38:23.733 -> mode:DIO, clock div:1
16:38:23.733 -> load:0x3fce3808,len:0x44c
16:38:23.733 -> load:0x403c9700,len:0xbe4
16:38:23.733 -> load:0x403cc700,len:0x2a38
16:38:23.733 -> entry 0x403c98d4
16:38:23.781 -> [    98][D][esp32-hal-tinyusb.c:675] tinyusb_enable_interface(): Interface CDC enabled
16:38:23.828 -> [   111][D][esp32-hal-tinyusb.c:564] tinyusb_load_enabled_interfaces(): Load Done: if_num: 2, descr_len: 75, if_mask: 0x10
16:38:23.828 -> [   117][D][esp32-hal-rmt.c:615] rmtInit():  -- TX RMT - CH 0 - 1 RAM Blocks - pin 48
16:38:23.828 -> [   124][W][esp32- al2ti[Ir[E:232] HUm75AtaaciInanrl-ptS-agA.::4DG] temen(i: eregit()s Ust g ppIO 1d!fSr ti_P N
16:38:23.828 -> LE  L.3.
I][ESP32-HUB75-MatrixPanel-I2S-DMA.h:415] begin(): [begin()] Using GPIO 16 for G1_PIN
16:38:23.875 -> [   141][I][ESP32-HUB75-MatrixPanel-I2S-DMA.h:416] begin(): [begin()] Using GPIO 7 for B1_PIN
16:38:23.875 -> [   150][I][ESP32-HUB75-MatrixPanel-I2S-DMA.h:417] begin(): [begin()] Using GPIO 6 for R2_PIN
16:38:23.875 -> [   158][I][ESP32-HUB75-MatrixPanel-I2S-DMA.h:418] begin(): [begin()] Using GPIO 17 for G2_PIN
16:38:23.875 -> [   166][I][ESP32-HUB75-MatrixPanel-I2S-DMA.h:419] begin(): [begin()] Using GPIO 5 for B2_PIN
16:38:23.875 -> [   174][I][ESP32-HUB75-MatrixPanel-I2S-DMA.h:420] begin(): [begin()] Using GPIO 12 for A_PIN
16:38:23.875 -> [   183][I][ESP32-HUB75-MatrixPanel-I2S-DMA.h:421] begin(): [begin()] Using GPIO 8 for B_PIN
16:38:23.923 -> [   191][I][ESP32-HUB75-MatrixPanel-I2S-DMA.h:422] begin(): [begin()] Using GPIO 11 for C_PIN
16:38:23.923 -> [   199][I][ESP32-HUB75-MatrixPanel-I2S-DMA.h:423] begin(): [begin()] Using GPIO 3 for D_PIN
16:38:23.923 -> [   207][I][ESP32-HUB75-MatrixPanel-I2S-DMA.h:424] begin(): [begin()] Using GPIO -1 for E_PIN
16:38:23.923 -> [   215][I][ESP32-HUB75-MatrixPanel-I2S-DMA.h:425] begin(): [begin()] Using GPIO 46 for LAT_PIN
16:38:23.923 -> [   224][I][ESP32-HUB75-MatrixPanel-I2S-DMA.h:426] begin(): [begin()] Using GPIO 9 for OE_PIN
16:38:23.923 -> [   232][I][ESP32-HUB75-MatrixPanel-I2S-DMA.h:427] begin(): [begin()] Using GPIO 10 for CLK_PIN
16:38:23.971 -> [   241][I][ESP32-HUB75-MatrixPanel-leddrivers.cpp:46] fm6124init(): [LEDdrivers] MatrixPanel_I2S_DMA - initializing FM6124 driver...
16:38:23.971 -> [   253][I][ESP32-HUB75-MatrixPanel-I2S-DMA.cpp:38] allocateDMAmemory(): [I2S-DMA] Free heap: 350616
16:38:23.971 -> [   261][I][ESP32-HUB75-MatrixPanel-I2S-DMA.cpp:39] allocateDMAmemory(): [I2S-DMA] Free SPIRAM: 0
16:38:23.971 -> [   270][I][ESP32-HUB75-MatrixPanel-I2S-DMA.cpp:43] allocateDMAmemory(): [I2S-DMA] allocating rowBitStructs with pixel_color_depth_bits of 8
16:38:23.971 -> [   282][I][ESP32-HUB75-MatrixPanel-I2S-DMA.cpp:70] allocateDMAmemory(): [I2S-DMA] Allocating 32768 bytes memory for DMA BCM framebuffer(s).
16:38:24.018 -> [   294][I][ESP32-HUB75-MatrixPanel-I2S-DMA.cpp:75] allocateDMAmemory(): [I2S-DMA] Minimum visual refresh rate (scan rate from panel top to bottom) requested: 60 Hz
16:38:24.018 -> [   309][W][ESP32-HUB75-MatrixPanel-I2S-DMA.cpp:93] allocateDMAmemory(): [I2S-DMA] lsbMsbTransitionBit of 0 gives 28 Hz refresh rate.
16:38:24.018 -> [   320][W][ESP32-HUB75-MatrixPanel-I2S-DMA.cpp:93] allocateDMAmemory(): [I2S-DMA] lsbMsbTransitionBit of 1 gives 57 Hz refresh rate.
16:38:24.065 -> [   332][W][ESP32-HUB75-MatrixPanel-I2S-DMA.cpp:93] allocateDMAmemory(): [I2S-DMA] lsbMsbTransitionBit of 2 gives 112 Hz refresh rate.
16:38:24.065 -> [   344][W][ESP32-HUB75-MatrixPanel-I2S-DMA.cpp:106] allocateDMAmemory(): [I2S-DMA] lsbMsbTransitionBit of 2 used to achieve refresh rate of 60 Hz. Percieved colour depth to the eye may be reduced.
16:38:24.065 -> [   361][I][ESP32-HUB75-MatrixPanel-I2S-DMA.cpp:109] allocateDMAmemory(): [I2S-DMA] DMA has pixel_color_depth_bits of 6
16:38:24.065 -> [   372][I][ESP32-HUB75-MatrixPanel-I2S-DMA.cpp:123] allocateDMAmemory(): [I2S-DMA] Recalculated number of DMA descriptors per row: 32
16:38:24.112 -> [   384][D][gdma_lcd_parallel16.cpp:340] allocate_dma_desc_memory(): [S3] Allocating 6144 bytes memory for DMA descriptors.
16:38:24.112 -> [   397][I][ESP32-HUB75-MatrixPanel-I2S-DMA.cpp:276] configureDMA(): [I2S-DMA] 512 DMA descriptors linked to buffer data.
16:38:24.112 -> [   405][I][gdma_lcd_parallel16.cpp:136] init(): [S3] Clock divider is 12
16:38:24.112 -> [   412][D][gdma_lcd_parallel16.cpp:138] init(): [S3] Resulting output clock frequency: 13333333 Mhz
16:38:24.112 -> [   421][I][ESP32-HUB75-MatrixPanel-I2S-DMA.cpp:316] configureDMA(): [I2S-DMA] DMA setup completed

I'm wondering, if under some unclear circumstances, the LCD-mode is disabled in the SOC chip ???

As no error message is reported, it's difficult to identify if this issue is related to the ESP32-S3 chip itself, or something wrong in the library when using the DMA/LCD-mode.

The compilation said the the device used is ESP32-S3, no doubt with that (I got the #pragma message) !
May be there is something missing/incorrect in the **gdma_lcd_parallel16.cpp** source for S3 ? It report that all is OK, but sometimes, there is nothing on outputs !

mrfaptastic, 2023-04-08:
Bizarre. Can you check to see if you getting any output on the clock pin?

noflip95, 2023-04-08:
Indeed very bizarre.

When the display works, this is what the logic analyzer get.
We can see the FM6124 sequence at the begining, and about 180 ms later, the normal RGB mux.
[image]

Detail of the FM6124 : we can see that REG11 and REG12 are correctly addressed.
[image]

When the display don't work, we got the FM6124 sequence, but after 180 ms ... nothing, or just one clock pulse.
[image]

What's annoying is that it's not systematic. It works fine 9 times out of 10, and it doesn't work the tenth time.

I may be wrong, but I suspect a problem in the initialization of the LCD-CAM (in gdma_lcd_parallel16.cpp, at Bus_Parallel16::init). May be, adding some delay before "periph_module_enable(PERIPH_LCD_CAM_MODULE);" or after the reset, could change this strange behaviour.

mrfaptastic, 2023-04-08:
Yeah. I think add a few more delays and also double the reset sequence. See what happens.

noflip95, 2023-04-09:
I first made a rather brutal modification without thinking too much, adding a 1 ms delay before and after the reset. And obviously it worked better. I didn't do intensive tests, but I didn't notice the display fault once again.
I then removed the first delay (before the reset), and the same thing seemed to work fine.

Then I read in the technical reference manual §29.3.4 ,that the clock had to be configured **BEFORE** doing the reset.

[image]
So I moved the sequence after the initialization of the clock, I also added the reset of the FIFO and I put a delay back to 100 µs:

[image]
**BUT then it didn't work again!**

So I finally set the delay to 1 ms and so far it works fine. I'll do some long-term testing to see if the issue is fixed.

[image]

The documentation says that setting the bit to 1 triggers the reset and then it is cleared by the hardware. But I did not find any specification on the duration of this operation, which I presume is not instantaneous. A delay of 100 µs sometimes seems insufficient. There is no way of knowing when the reset is complete. If this is longer than 100 µs, then there is a risk of starting to write in the registers too early.

Note that, I also changed the frequency unit in a debug line, because otherwise we will work in the hyper-frequency domain !!!
[image]

I need to test in the long term (a few days) to see if the problem reoccurs or not. I will keep you informed of my investigations.

mrfaptastic, 2023-04-09:
Great work. Thanks for looking into this, was going to read the TRM when I had a chance as well. Certainly looks like the need for a delay to be honest. 

If you get it working without this issue - feel free to raise a pull request.

mrfaptastic, 2023-04-17:
@noflip95 - Any luck?

noflip95, 2023-04-17:
Unfortunately no.
I spent a lot of time trying to put timeouts everywhere (up to 10ms), to move the reset sequence, but nothing helped.
Sooner or later there is a boot where the LCD-Mode does not start while the ESP32 is working properly.

I have no idea where the problem is, what is certain is that it happens quite often. As my project requires a lot of I/O ports, I cannot transpose it to a simple ESP32. It must run on a S3 (but SPI RAM is not mandatory).

I'm using an ESP32-S3-N16R8, with SPI RAM disabled, so I don't think that's the problem

Is this a hardware bug specific to the S3? I haven't seen any reports of it, but maybe it hasn't been used too much on this version of the ESP yet.

mrfaptastic, 2023-04-17:
Are you able to share privately a cut down version/sketch that might trigger this bug?

rigorka, 2023-04-17:
Hi! I can confirm that this bug is completely reproducible on my ESP32-S3-N16R8 too. 

Nothing fancy in the sketch at all: just a display init and the output of one line of text. 
My observations show about 20% failure rate when the ESP is restarted after uploading the new firmware, and ~5% failure rate on casual restarts.

If we can't figure out what's causing this behavior, I think I'll have to use a slightly overly complicated but fairly bullet proof approach: I can monitor the current consumed by the display with the hall sensor. I can show some kind of test pattern on every boot, measure the power consumption, and trigger the ESP reboot if the current consumption does not meet my expectations (==display is blank or showing gibberish).

Same current monitoring approach could be used to diagnose "loss of display" in production for any reason.

But of course I'd rather we find the root cause of what's happening.

mrfaptastic, 2023-04-17:
Does a panel even need to be connected to trigger this issue? I presume it isn't a current related issue.

mrfaptastic, 2023-04-17:
No, what I was trying to say is not related to your idea to detect an issue. It's whether the startup power draw from the LED panel is what causes the S3 to play up.

But having looked again at the original analysis that was done, the issue occurs even before a valid "row" of data can be sent out. So ignore that.

mrfaptastic, 2023-04-17:
@paintyourdragon actually explained the issue in the S3 gdma.cpp file and I must have accidentally changed something to break it. Looks like lcd_dummy needs to be enabled

So change:

```
LCD_CAM.lcd_user.lcd_dummy = 0;          // Dummy phase(s) @ LCD start
    LCD_CAM.lcd_user.lcd_dummy_cyclelen = 0; // 1 dummy phase
```

Needs to be.
```
LCD_CAM.lcd_user.lcd_dummy = 1; // Dummy phase(s) @ LCD start LCD_CAM.lcd_user.lcd_dummy_cyclelen = 64; // 64 dummy phase
```

Could be the fix?

mrfaptastic, 2023-04-17:
The comment he wrote:

```
    // "Dummy phases" are initial LCD peripheral clock cycles before data
    // begins transmitting when requested. After much testing, determined
    // that at least one dummy phase MUST be enabled for DMA to trigger
    // reliably. A problem with dummy phase(s) is if we're also using the
    // LCD_PCLK_IDX signal (not used in this code, but Adafruit_Protomatter
    // does)...the clock signal will start a couple of pulses before data,
    // which may or may not be problematic in some situations. You can
    // disable the dummy phase but need to keep the LCD TX FIFO primed
    // in that case, which gets complex.
    // always_out_en is set above to allow aribtrary-length transfers,
    // else lcd_dout_cyclelen is used...but is limited to 8K. Long (>4K)
    // transfers need DMA linked lists, not used here but mentioned later.

```

mrfaptastic, 2023-04-18:
Try the latest git master and see if that helps

noflip95, 2023-04-18:
Tests in progress...

I've first, made some tests (about 15 sequences of boots/power on) with that setting:
LCD_CAM.lcd_user.lcd_dummy = **1**; 
LCD_CAM.lcd_user.lcd_dummy_cyclelen = **1**;
and never notice a black display.

Now, I'm testing with the latest changes:
Extended delay to 1 ms after lcd_CAM reset and gdma_reset ( esp_rom_delay_us(1000); )
LCD_CAM.lcd_user.lcd_dummy = **1**; 
LCD_CAM.lcd_user.lcd_dummy_cyclelen = **100**;

It's still too early to tell if this fixes the problem for good, but so far it seems to work and I haven't had a black screen yet.
I will continue long-term testing before confirming that it works well. 
Stay tuned !

> ... It's whether the startup power draw from the LED panel is what causes the S3 to play up.

No, I think the consumption of the LED panel has nothing to do. In my case, to comply with the specifications of the panels operating under **5V**, I use two 74HCT245 drivers. The ESP32-S3 is therefore not directly connected to the panels and we could also test very well without connecting them to the HUB75.

_I had thought as a workaround, to sacrifice an input pin of the ESP32-S3, connected to the OE or LAT output. And to detect if the signal was indeed present after the start, and reboot if not. It's not very complicated to do, but it uses an extra input. The best thing is obviously to find a software solution so that the DMA starts every time._

rigorka, 2023-04-18:
Testing the latest master for a few hrs now - no failures occurred in the last 60 or so ESP reboots.
Will keep observing...

noflip95, 2023-04-19:
I tested on a long series (more than 100 !) of reboot, power up/down, and reprogramming and found no display fault again.
The problem therefore seems to have been definitively solved.

Thank you for the support.

closed 2023-04-19 by noflip95

mrfaptastic, 2023-04-19:
Awesome! Thanks for confirming.

jwee1369, 2024-03-16:
I don't know if my problem is related, but i'm not getting anything on my 64x32 led matrix from Adafruit. Very rarely i get 3 white dots, and one time i got a red broken line. I've verified on the Mega with Adafruit example and the panel works fine, so nothing broken there.

led matrix: [https://www.adafruit.com/product/2276](url)
ESP32-S3-DevKitC-1-N8R2: [https://www.amazon.com/dp/B0B6HT7V7P?psc=1&ref=ppx_yo2ov_dt_b_product_details](url)

 I used the pinout in the esp32s3-default-pins.hpp on the SImpleTestShapes.ino example. reinitialized the dma_display with the following. 

`  MatrixPanel_I2S_DMA *dma_display = nullptr;

    HUB75_I2S_CFG::i2s_pins _pins={R1_PIN, G1_PIN, B1_PIN, R2_PIN, G2_PIN, B2_PIN, A_PIN, B_PIN, C_PIN, D_PIN, E_PIN, LAT_PIN, OE_PIN, CLK_PIN};
    HUB75_I2S_CFG mxconfig(
      PANEL_RES_X, // Module width
      PANEL_RES_Y, // Module height
      PANEL_CHAIN, // chain length
      _pins // pin mapping
    );

    dma_display = new MatrixPanel_I2S_DMA(mxconfig);
`

I've checked my pin connection multiple times.

Any help would be appreciated, thanks!

mrfaptastic, 2024-03-16:
Most likely power related. Try a different power supply.