mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #436: GIFs not looping for 8 seconds in AnimatedGIFPanel_SPIFFS example

UltrashRicco, 2023-04-10:
Hi there and thank you so much for this library, I am having fun with it!

I noticed an issue (I think) with the AnimatedGIFPanel_SPIFFS example sketch. I believe it should be playaing all the gifs inside the /gifs directory, for 8000 milliseconds before moving on to the next GIF in the directory, according to the following code, right?

```
void ShowGIF(char *name)
{
  start_tick = millis();
   
  if (gif.open(name, GIFOpenFile, GIFCloseFile, GIFReadFile, GIFSeekFile, GIFDraw))
  {
    x_offset = (MATRIX_WIDTH - gif.getCanvasWidth())/2;
    if (x_offset < 0) x_offset = 0;
    y_offset = (MATRIX_HEIGHT - gif.getCanvasHeight())/2;
    if (y_offset < 0) y_offset = 0;
    Serial.printf("Successfully opened GIF; Canvas size = %d x %d\n", gif.getCanvasWidth(), gif.getCanvasHeight());
    Serial.flush();
    while (gif.playFrame(true, NULL))
    {      
      if ( (millis() - start_tick) > 8000) { // we'll get bored after about 8 seconds of the same looping gif
        break;
      }
    }
    gif.close();
  }

} /* ShowGIF() */
```

Well actually, the behaviour is as follows: every GIF int the directory is indeed being played, but the next GIF is played as soon as the previous GIF played only once (only 1 loop). I believe it is intended to be played for 8 seconds (8000 milliseconds) instead, but it does not seem to be working, and I am unable to understand how the code works to try and fix it. ShowGIF is a function that is called in the loop function. I tried adding the following serial output, but it is not working either:

```
while (gif.playFrame(true, NULL))
    {      
      if ( (millis() - start_tick) > 8000) { // we'll get bored after about 8 seconds of the same looping gif
        Serial.printf("reached 8000 milliseconds");
        Serial.flush();
        break;
      }
```

[image]

Also, I took a look at the "ESP32-LGFX-SDCard-GifPlayer" example from the AnimatedGIF library, and it shows an interesting way of controlling how much the GIFs have to loop:

```
#include "AnimatedGIF.h"

AnimatedGIF gif;

// rule: loop GIF at least during 3s, maximum 5 times, and don't loop/animate longer than 30s per GIF
const int maxLoopIterations =     5; // stop after this amount of loops
const int maxLoopsDuration  =  3000; // ms, max cumulated time after the GIF will break loop
const int maxGifDuration    = 30000; // ms, max GIF duration
```

```
void loop()
{

  tft.clear();

  const char * fileName = GifFiles[currentFile++%totalFiles].c_str();

  int loops = maxLoopIterations; // max loops
  int durationControl = maxLoopsDuration; // force break loop after xxx ms

  while(loops-->0 && durationControl > 0 ) {
    durationControl -= gifPlay( (char*)fileName );
    gif.reset();
  }

}
```

I would like to use this logic inside the AnimatedGIFPanel_SPIFFS example, but I really am not familiar with coding and I do not know how to increment a loop counter, a loop duration counter, etc.

Could someone please help? (at least fixing the intended behaviour of playing each GIF for 8 seconds before moving on to the next!).
Many thanks in advance!

tsctrl, 2023-04-10:
basically its depends on your gif files, gif has loop settings during gif creation and how long your gif file was looping. for AnimatedGif library, it has max loop count to break the while or you can use file frames amount that was return by the library to check duration of the selected gif to render and count current loop number.

mrfaptastic, 2023-04-10:
As per @tsctrl, perhaps the GIF isn't designed to loop.

Perhaps change this code:

```
    while (gif.playFrame(true, NULL))
    {      
      if ( (millis() - start_tick) > 8000) { // we'll get bored after about 8 seconds of the same looping gif
        break;
      }
    }
```
to this:

```
while(  (millis() - start_tick) < 8000 )
{
    while (gif.playFrame(true, NULL))
    {      
  //    if ( (millis() - start_tick) > 8000) { // we'll get bored after about 8 seconds of the same looping gif
    //    break;
//      }
    }
}
```

Note: Haven't tested it...

UltrashRicco, 2023-04-10:
Thank you both! I believe the GIFs are designed to loop indefinitely (loop = 0).

@mrfaptastic your code change suggestion did work indeed:
```

void ShowGIF(char *name)
{
  start_tick = millis();
   
  if (gif.open(name, GIFOpenFile, GIFCloseFile, GIFReadFile, GIFSeekFile, GIFDraw))
  {
    x_offset = (MATRIX_WIDTH - gif.getCanvasWidth())/2;
    if (x_offset < 0) x_offset = 0;
    y_offset = (MATRIX_HEIGHT - gif.getCanvasHeight())/2;
    if (y_offset < 0) y_offset = 0;
    Serial.printf("Successfully opened GIF; Canvas size = %d x %d\n", gif.getCanvasWidth(), gif.getCanvasHeight());
    Serial.flush();
    /*while (gif.playFrame(true, NULL))
    {      
      if ( (millis() - start_tick) > 8000) { // we'll get bored after about 8 seconds of the same looping gif
        Serial.printf("reached 8000 milliseconds");
        Serial.flush();
        break;
      }
    }
    gif.close();
  }*/
  while(  (millis() - start_tick) < 8000 )
  {
      while (gif.playFrame(true, NULL))
      {      
        if ( (millis() - start_tick) > 8000) { // we'll get bored after about 8 seconds of the same looping gif
        Serial.printf("reached 8000 milliseconds, playing next GIF\n");
        Serial.flush();
          break;
        }
      }
  }
}
} /* ShowGIF() */
```

The GIFs are now looping for 8 seconds and my test serial output also works:

[image]

There is a micro-stutter with some GIFs (the larger ones, I guess) when looping, mostly noticeable with looping animations of course.

Thank you so much! I would suggest modifying the code in the example accordingly. :)
I'll try to add more options from the AnimatedGIFs library (play GIFs for x loops before moving to the next one, etc.), but this is really hard for me. :)

closed 2023-04-10 by mrfaptastic