mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #437: 128x64 panel - is this supported?

tavalin, 2023-04-13:
I've got two 128x64 panels which claim to be HUB75E however I'm not so sure they are using the 'e' pin like previous panels.

Is anyone able to offer any insight on these?

[image]
[image]
[image]
[image]

Thanks

tavalin, 2023-04-15:
Might be worth noting I managed to get this panel working with https://github.com/hzeller/rpi-rgb-led-matrix/ by @hzeller using the --led-row-addr-type=3 setting. Is there a comparable setting?

mrfaptastic, 2023-04-15:
There isn't, but you know what. I am might just try and take a clone of this library and see if I can port it to the esp32.