mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #438: Question on the how the library works

LeonardoCT, 2023-04-15:
Hello, i was wondering how the 

```c++
dma_display->println("Test"); 
```
works.
 
What i'm interested is where acctualy the drawpixel() commands are, i want to make some modifications to the println(), but i can't find it, all i foud leads me back to the print.h (default arudino library), and i can't understand how it calls drawpixel().

Thanks

mrfaptastic, 2023-04-15:
You need to dig through the Adafruit_GFX library. It must be in there somewhere.

LeonardoCT, 2023-04-15:
> You need to dig through the Adafruit_GFX library. It must be in there somewhere.

i've looked and all i could find are reference back to the print.h of the arduino default library no drawpixel(), in  Adafruit_GFX i can't even find the println() prototype... it must be somewhere else and i still have no clue.

mrfaptastic, 2023-04-24:
I took a bit of a look, and I think it works like this.

DMA Library inherits AdaFruit_GFX which inherits Print

* Calls to print("XXXX") go all the way down to the Base 'Print' class
* This then triggers a call upwards to the parent Adafruit_GFX class, specifically Adafruit_GFX::write(uint8_t c) for each character to be printed
* Adafruit_GFX::write then called 'drawChar(cursor_x, cursor_y, c, textcolor, textbgcolor, textsize_x,
               textsize_y)' for each character
* drawChar then calls writePixel when then calls drawPixel - which then calls the drawPixel of this library.

closed 2024-07-28 by mrcodetastic