mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #441: ESP32-S3: Can't use SPIRAM framebuffer and Wifi at the same time

rigorka, 2023-04-17:
Testing my sketch on ESP32-S3-N16R8 I've bumped into a problem with wifi:
- With SPIRAM enabled, the display dies immediately upon successful connection to the wifi AP (display works fine whilst we're connecting but before the connection is completed);
- With SPIRAM disabled everything works fine

Would appreciate some clues as to which direction should I look into. Can this be related to the clock speed manipulation related to the SPIRAM usage?

mrfaptastic, 2023-04-18:
Refer to #442

Probably something to do with available memory bandwidth. 

Try my suggestion on the clock rate to see if that helps.

rigorka, 2023-04-18:
Thanks again @mrfaptastic!
I've tried changing `LCD_CAM.lcd_clock.lcd_clkm_div_num` as per your suggestions and my observations are as follows:

- 10: screen dies when wifi connects
- 11 to 14: everything works perfectly
- 15 to 20: everything works with an increasing flicker. At 20 flicker is pretty much unbearable.

I'm leaving it at 12 and will observe the stability long term.

mrfaptastic, 2023-04-19:
I have adjusted the library down to 13mhz or a clock divisor of 12 based on this. Thank you.

closed 2023-04-19 by mrfaptastic