mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #442: ESP32-S3: No output on display

acvigue, 2023-04-18:
Using this library on an ESP32-S3 (TinyS3) w/ IDF 5.0.1 (also tried 4.4.4) causes nothing to display on the screen. I've reduced the `main.cpp` file to the below contents:

I've tried both 3.0.8 and master.

```cpp
#include <ESP32-HUB75-MatrixPanel-I2S-DMA.h>
#include <stdio.h>
MatrixPanel_I2S_DMA *matrix;

extern "C" void app_main(void) {
HUB75_I2S_CFG::i2s_pins _pins = {35, 37, 36, 34, 9, 8, 7, 6, 21, 5, -1, 4, 2, 1};
HUB75_I2S_CFG mxconfig(64, 32, 1, _pins);
matrix = new MatrixPanel_I2S_DMA(mxconfig);

matrix->begin();
matrix->drawPixelRGB888(1,1,255,0,0); //red pixel test

while(1) {}
}
```

Logic analyzer output: (all lines stay low after the initial burst of clock pulses)
[image]

Log output:
```
I (43) boot: ESP-IDF v5.0.1-dirty 2nd stage bootloader
I (43) boot: chip revision: v0.1
I (43) boot.esp32s3: Boot SPI Speed : 80MHz
I (46) boot.esp32s3: SPI Mode       : DIO
I (51) boot.esp32s3: SPI Flash Size : 8MB
I (56) boot: Enabling RNG early entropy source...
I (61) boot: Partition Table:
I (64) boot: ## Label            Usage          Type ST Offset   Length
I (72) boot:  0 nvs              WiFi data        01 02 00009000 00024000
I (79) boot:  1 otadata          OTA data         01 00 0002d000 00002000
I (87) boot:  2 phy_init         RF data          01 01 0002f000 00001000
I (94) boot:  3 factory          factory app      00 00 00030000 001db000
I (102) boot:  4 ota_0            OTA app          00 10 00210000 001db000
I (109) boot:  5 ota_1            OTA app          00 11 003f0000 001db000
I (117) boot:  6 littlefs         Unknown data     01 82 005cb000 001f4000
I (124) boot: End of partition table
I (129) boot: Defaulting to factory image
I (133) esp_image: segment 0: paddr=00030020 vaddr=3c0e0020 size=4e820h (321568) map
I (178) esp_image: segment 1: paddr=0007e848 vaddr=3fc9f100 size=017d0h (  6096) load
I (179) esp_image: segment 2: paddr=00080020 vaddr=42000020 size=dfdech (916972) map
I (286) esp_image: segment 3: paddr=0015fe14 vaddr=3fca08d0 size=03e98h ( 16024) load
I (289) esp_image: segment 4: paddr=00163cb4 vaddr=40374000 size=1b000h (110592) load
I (320) boot: Loaded app from partition at offset 0x30000
I (320) boot: Disabling RNG early entropy source...
D (331) flash HPM: HPM with dummy, status is 3
I (332) esp_psram: Found 8MB PSRAM device
I (332) esp_psram: Speed: 80MHz
I (333) cpu_start: Pro cpu up.
I (337) cpu_start: Starting app cpu, entry point is 0x40375a5c
0x40375a5c: call_start_cpu1 at /COMPONENT_ESP_SYSTEM_DIR/port/cpu_start.c:142

I (0) cpu_start: App cpu up.
I (1261) esp_psram: SPI SRAM memory test OK
D (1269) clk: RTC_SLOW_CLK calibration value: 3622016
I (1275) cpu_start: Pro cpu start user code
I (1275) cpu_start: cpu freq: 160000000 Hz
I (1276) cpu_start: Application information:
I (1279) cpu_start: Project name:     smartmatrix-idf
I (1284) cpu_start: App version:      926975f-dirty
I (1290) cpu_start: Compile time:      
I (1294) cpu_start: ELF file SHA256:  29f994a6cd8556f6...
I (1300) cpu_start: ESP-IDF:          v5.0.1-dirty
I (1306) cpu_start: Min chip rev:     v0.0
I (1311) cpu_start: Max chip rev:     v0.99 
[... 87 lines cut ...]
I (00:17:59.098) begin(): Using GPIO 9 for OE_PIN
I (00:17:59.104) begin(): Using GPIO 8 for CLK_PIN
I (00:17:59.109) I2S-DMA: Free heap: 253135
I (00:17:59.114) I2S-DMA: Free SPIRAM: 8386308
I (00:17:59.119) I2S-DMA: allocating rowBitStructs with pixel_color_depth_bits of 8
I (00:17:59.128) I2S-DMA: Allocating 16384 bytes memory for DMA BCM framebuffer(s).
I (00:17:59.136) I2S-DMA: Minimum visual refresh rate (scan rate from panel top to bottom) requested: 60 Hz
W (00:17:59.146) I2S-DMA: lsbMsbTransitionBit of 0 gives 30 Hz refresh rate.
W (00:17:59.154) I2S-DMA: lsbMsbTransitionBit of 1 gives 61 Hz refresh rate.
W (00:17:59.162) I2S-DMA: lsbMsbTransitionBit of 1 used to achieve refresh rate of 60 Hz. Percieved colour depth to the eye may be reduced.
I (00:17:59.175) I2S-DMA: DMA has pixel_color_depth_bits of 7
I (00:17:59.181) I2S-DMA: Recalculated number of DMA descriptors per row: 64
D (00:17:59.189) S3: Allocating 12288 bytes memory for DMA descriptors.
I (00:17:59.201) I2S-DMA: 1024 DMA descriptors linked to buffer data.
I (00:17:59.203) S3: DMA buffer is on PSRAM. Limiting clockspeed....
I (00:17:59.210) S3: Clock divider is 10
D (00:17:59.214) S3: Resulting output clock frequency: 16000000 Mhz
D (00:17:59.221) gdma: new group (0) at 0x3c134444
D (00:17:59.226) gdma: new pair (0,0) at 0x3c134004
D (00:17:59.230) gdma: new tx channel (0,0) at 0x3c134884
D (00:17:59.236) gdma: tx channel (0,0), (32:64) bytes aligned, burst enabled
D (1876) intr_alloc: Connected src 71 to int 9 (cpu 1)
D (00:17:59.248) gdma: install interrupt service for tx channel (0,0)
I (00:17:59.255) I2S-DMA: DMA setup completed
```

mrfaptastic, 2023-04-18:
Try latest git master and see if there's any fix.

mrfaptastic, 2023-04-18:
Try the latest git master version and see if that fixes it.

mrfaptastic, 2023-04-18:
Also, see if NOT using PSRAM solves the issue.

mrfaptastic, 2023-04-18:
And if not, then use the library without enabling PSRAM support. If that fixes the issue then there's a likely issue with your PSRAM being too slow for the DMA buffer.... You will need to edit this line:

https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-DMA/blob/8b77e3c793abc4e26bb608351a3e6f56126bbce9/src/platforms/esp32s3/gdma_lcd_parallel16.cpp#L115

```
     LCD_CAM.lcd_clock.lcd_clkm_div_num = 10; //16mhz is the fasted the Octal PSRAM can support it seems 
```

Change the value to '20' or even higher value. Or not use PSRAM (recommended).

acvigue, 2023-04-18:
Changing the clock divisor to 20 worked, could this be added as a constructor argument so I can keep the underlying code the same?

mrfaptastic, 2023-04-18:
Had you enabled psram support yourself or was it enabled already by the library. I need to check that it doesn't auto-enable.

mrfaptastic, 2023-04-18:
And I agree with your pre edited comment.. I need to make the clock divisor selectable for psram as there appears to be some memory bandwidth issue with DMA from psram that causes the LCD module to just bomb out.

What was the lowest clock divider you could use before it failed? Are you able to test for me? 

20, 18, 16, 12 etc. With 10 = blank

acvigue, 2023-04-18:
> Had you enabled psram support yourself or was it enabled already by the library. I need to check that it doesn't auto-enable.

Enabled automatically, the CMakeLists (line 32) has -Dsomething_spiram_related

acvigue, 2023-04-18:
> And I agree with your pre edited comment.. I need to make the clock divisor selectable for psram as there appears to be some memory bandwidth issue with DMA from psram that causes the LCD module to [...]

Tried 20, 18, 15. 18 was spotty, 15 refused to work. I had the qspi ram clock set at 80 Mhz.

acvigue, 2023-04-18:
and if it helps, here's the SDKCONFIG file that I am using,
[https://gist.github.com/acvigue/085fcced47d7008d3e8aab2ad7cfd0ec](https://gist.github.com/acvigue/085fcced47d7008d3e8aab2ad7cfd0ec)

mrfaptastic, 2023-04-18:
> Tried 20, 18, 15. 18 was spotty, 15 refused to work. I had the qspi ram clock set at 80 Mhz.
> edit: forgot to say that 20 did work

Excellent information, thanks for this. Proves this issue is triggered by DMA starvation/memory bandwidth.

My S3 that I test wish has Octal SPI (twice the bandwidth) hence why I can use a divisor of 10.

Is the output flickery with a divisor of 20? I would think so.

acvigue, 2023-04-19:
> Excellent information, thanks for this. Proves this issue is triggered by DMA starvation/memory bandwidth.
>
> My S3 that I test wish has Octal SPI (twice the bandwidth) hence why I can use a divisor [...]

I didn't notice too much of a difference w/ 20

closed 2023-04-19 by mrfaptastic