mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #445: No brightness control on ESP32-S3

acvigue, 2023-04-25:
Using the same example code as my previous issue relating to the S3, with a single line added `matrix->setBrightness(16)` does not change the panel brightness. SPIRAM is disabled, I have tried multiple brightness levels, using setPanelBrightness, setBrightness8, and setBrightness.

LeonardoCT, 2023-04-26:
Hello, i think the setBrightness does some pwm magic on the OE pin, I would check if it's configured correctly

mrfaptastic, 2023-04-26:
Brightness control is agnostic of the ESP32 variant, perhaps the panels you have do brightness differently. I just tested brightness control with my ESP-S3 and a1/16 scan panel, no issues.

mrfaptastic, 2023-04-26:
This is awesome by the way: https://vigue.me/posts/esp32-internet-connected-smart-matrix

Nice work!

mrfaptastic, 2023-04-26:
Brightness control is agnostic of the ESP32 variant and is a construct of a data sent via. DMA to the panel. Essentially toggling the OE pin to show or hide a row is how brightness is managed. Perhaps the panels you have do brightness differently? I just tested brightness control with my ESP-S3 and a 1/16 scan panel, no issues.

I just noticed your library here, this is great: https://vigue.me/posts/esp32-internet-connected-smart-matrix

I was looking to do something similar to you (generate bitmaps on a server and send down to an ESP), and leverag what this person wrote for a Raspberry Pi (where the PI does the graphics and sends it to the panel directly). 

https://github.com/feram18/led-stock-ticker

acvigue, 2023-04-27:
Thanks! I probably wired something up wrong, seeing as how it was working fine on a similar project.

acvigue, 2023-04-27:
> Brightness control is agnostic of the ESP32 variant and is a construct of a data sent via. DMA to the panel. Essentially toggling the OE pin to show or hide a row is how brightness is managed. Perhaps the panels you have do brightness differently? I just tested brightness control with my ESP-S3 and a 1/16 scan panel, no issues.
> 
> I just noticed your library here, this is great: https://vigue.me/posts/esp32-internet-connected-smart-matrix
> 
> I was looking to do something similar to you (generate bitmaps on a server and send down to an ESP), and leverage (re-code) what this person wrote for a Raspberry Pi (where the PI does the graphics and sends it to the panel directly) to do as such... might see if what you created could be used on the client/esp32 side.
> 
> https://github.com/feram18/led-stock-ticker

ESP32-S3 code: https://github.com/acvigue/smartmatrix-idf

closed 2023-04-27 by acvigue