mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #446: How to make sure the LED screen is blank during reboots/firmware updates/ESP downtime?

rigorka, 2023-04-26:
Every time I'm updating my ESP32S3 firmware my LED panels are flashing wildly for the entire duration of the update process. When ESP32 reboots I have about 2..4 seconds of wild flicker. When ESP is powered off the panels typically get stuck with some random very bright noise pattern.

Bringing the OE pin down to the ground does not change the situation. Of course I can add a MOSFET or a relay to the panel's power line and make sure the panels are powered off externally - but maybe there is a more clever way to ensure that the panels stay blank during the firmware updates or ESP downtimes?

board707, 2023-04-26:
> Bringing the OE pin down to the ground does not change the situation.

Did you try this by software or by external mosfet?

rigorka, 2023-04-26:
> > Bringing the OE pin down to the ground does not change the situation.
> 
> Did you try this by software or by external pulldown resistor?

I've tried it with my fingers :) Disconnecting OE from the ESP32-S3 pin 2 and connecting it to the GND pin results in a wild set of patterns displayed on the LED panels.

board707, 2023-04-26:
> Disconnecting OE from the ESP32-S3 pin 2 and connecting it to the GND pin

As I just remembered,, the polarity of OE is inverted on RGB panels, so you need to try pulup OE signal instead of pulling it down.

raulzanardo, 2024-06-26:
I'm using a "ESP32 NodeMcu – DevKit v1" connected directly to a 64x64 led matrix with ICN2038S chip. 

I tried the pullup resistor (10k and 1k) on the OE pin and it didn't work for my setup. 
What can it be?

BTW, the pull up solution worked on a ESP32-S3 (HD-WF2) board that has a 74hc245 buffer between the ESP and the matrix. I put the resistor in the ESP side and it works fine on reboots/firmware updates. Thanks @board707 !