mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #448: Chipset SM16208SJ Y-Index Error

Jiwno, 2023-05-08:
While I'm testing index with led Matrix using chipset SM16208SJ, I found an error with Y index.

X indexes are correct but y indexes are wrong.

Is there anyone who knows how to solve this problem?

[image]

https://user-images.githubusercontent.com/57893252/236830142-6a701b9d-3da2-412e-99ca-30d80b94d7e3.mp4

board707, 2023-05-11:
> While I'm testing index with led Matrix using chipset SM16208SJ, I found an error with Y index.

Every LED panel uses two types of IC to control the leds - the driver to manipulate individual pixels and the multiplexer to switch led rows on and off. You have a problem not with driver SM16208 (it is a standard driver, compatible with the library).  According to your photo, your panel uses TC7558 as multiplexer. This is a SHIFT-Register type multiplexer. You have to set up the library accordingly.
I have no idea, is it compatible with ESP32-HUB75-MatrixPanel-DMA code, ask @mrfaptastic

mrfaptastic, 2023-05-11:
What's another library this panel works with? PxMatrix? 

If so, what configuration. I'll look at how the code works there for insight then.

board707, 2023-05-11:
If I remember correctly, PxMatrix supports panels with a shift register multiplex.
I also have this option in my library.

The principle there is simple - a HUB pins A B C are connected to shift register, usually A is CLK, B = LAT and C = DATA. 
When you need to switch a row, you push a byte into the register, where the bit corresponding to the row is 1, and the rest are zero

Example code:
```
void set_mux(uint8_t curr_row) override {
		// Just shift the row mux by one for incremental access
		digitalWrite(pin_DMD_B, HIGH);                    // Latch open
		digitalWrite(pin_DMD_C, (curr_row == 0));   // Shift out 1 for line 0, 0 otherwise
		digitalWrite(pin_DMD_A, HIGH);                  // Clock out this bit
		digitalWrite(pin_DMD_A, LOW);
		digitalWrite(pin_DMD_B, LOW);                   // Latch close
	}
```

board707, 2023-05-11:
Addition
As I see, the shift type multiplexing has already been added to the library by DarrylStrong
https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-DMA/issues/164

closed 2024-01-09 by mrfaptastic