mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #449: Half of the matrix is red

DeltaW0x, 2023-05-08:
Hi, I've been trying to get this 32x64 FM6126A matrix to work with my original ESP32 for the past few days, but the lower half is red no matter what I do. I've tried the 1/8 scan mode, I've tried changing chip type, shortening the wires as much as possible, 
I even got a 3.3v -> 5v level shifter but the result is always the same. I tested the matrix on a RPI with the adafruit hub75 HAT and it does work perfectly with every example, so I'm really at a loss here. Maybe there's something wrong with my esp32? 
[image]

mrfaptastic, 2023-05-09:
> Maybe there's something wrong with my esp32?

Flip the R1, G1, B1 pins around (i.e. physically connect what should be R2, G2, B2) - If the problem then shows on the top half, then it's an issue with wiring.

Else, it could be a defective panel.

DeltaW0x, 2023-05-09:
Flipped the color pins,  the problem does show up in the top half, so I checked the pinout again and it's still correct. I know for a fact that the matrix does work, so maybe there's something wrong with the esp32?

mrfaptastic, 2023-05-10:
Change your red pin to another spare pin on your esp. Something wrong with the wiring or ESP.

DeltaW0x, 2023-05-10:
The issues was indeed that my esp32 is faulty. I transfered all the secondary color pins to other spots on the board and  it got fixed. Thank you for helping! I'll replace the esp32 asap but at least I can work now.
[image]

closed 2023-05-10 by DeltaW0x