mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #453: 80x40 1/20 SM5166P

sherlockedBBC, 2023-05-11:
Hey you all. I am trying to drive a 80x40 P4 Panel with a 1/20 Scanmode. The IC has SM5166P written on it, which looks similar to the SM5266P to me (but i might be wrong). Is there the possibility to drive a 80x40 panel? I only found one past issue to this topic.

Best

board707, 2023-05-11:
Every LED panel contains at least two types IC - LED driver itself and a line decoder (or multiplexor) - to switch the input data between lines of leds.
SM5166 is not a driver, it is a multiplexor with 138 type decoding ( compatible with the library).
However in order to understand if your panel will work, it is more important to know the type of LED driver chip

sherlockedBBC, 2023-05-11:
The LED driver chip are of the type ICN2037.

sherlockedBBC, 2023-05-11:
Currently i am getting these patterns when running the "[PIO_TestPatterns](https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-DMA/tree/master/examples/PIO_TestPatterns)" example.

[image]

board707, 2023-05-11:
> The LED driver chip are of the type ICN2037.

It should be supported by the library.
Are you edited panel configuration in the code ?
```
// Configure for your panel(s) as appropriate!
#define PIN_E 32
#define PANEL_WIDTH 64
#define PANEL_HEIGHT 64    
```

sherlockedBBC, 2023-05-11:
As the panel is 80x40 I changed the panel width to 80 and the panel height to 40, yes.

board707, 2023-05-11:
Do you define and connect pin E ?

sherlockedBBC, 2023-05-11:
Yes. Defined E as 32 and connected it to IO32

sherlockedBBC, 2023-05-12:
I should have changed the driver ... With this line it works now:

mxconfig.driver = HUB75_I2S_CFG::ICN2038S;

Thanks for the help!

board707, 2023-05-12:
It is a very strange, since 2037 and 2038 drivers requires a completely different initialisation

sherlockedBBC, 2023-05-12:
What should i be using idealy? As far as i can see this library offers these options:

FM6124,
FM6126A,
ICN2038S,
MBI5124,
SM5266P

board707, 2023-05-12:
If your panel works with ICN2038S setting - so use it.
I'm just surprised it works...

mohsin-rauma, 2024-07-26:
What type of controller you used for this 80*40 display, it is work with **### esp8266 or esp32 .........?** 
here are some picture of panel what I have 
[image]
[image]
[image]

board707, 2024-07-26:
@mohsin-rauma 
Your panel is 80x40 s20 with standard drivers, it should work with the library without any specific settings.

mohsin-rauma, 2024-07-26:
which library is available for this for ESP32

board707, 2024-07-26:
You are in that library github now...

mohsin-rauma, 2024-07-26:
https://github.com/mrcodetastic/ESP32-HUB75-MatrixPanel-DMA 
this is ok....?

rosid1, 2024-10-01:
how to fix HD WF2 using Panel 80*40 ICN2037BP?

board707, 2024-10-01:
@rosid1 
Did you read discussion #667 ?

rosid1, 2024-10-01:
> @rosid1 Did you read discussion #667 ?

i try this example its work [PIO_TestPatterns](https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-DMA/tree/master/examples/PIO_TestPatterns)" 
, but i set for drawtext on LED P4 80*40 its not work.

rosid1, 2024-10-01:
pls send me code drawtext example for panel 80*40 chip ICN2037BP on HD WF2

board707, 2024-10-01:
> pls send me example code drawtext for panel 80*40 chip ICN2037BP on HD WF2

There is no specific code for drawtext example using 80*40 panel.
If the code of [PIO_TestPatterns](https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-DMA/tree/master/examples/PIO_TestPatterns) example works for you, the drawtext also should works as well

rosid1, 2024-10-01:
> > pls send me example code drawtext for panel 80*40 chip ICN2037BP on HD WF2
> 
> There is no specific code for drawtext example using 80*40 panel. If the code of [PIO_TestPatterns](https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-DMA/tree/master/examples/PIO_TestPatterns) example works for you, the drawtext also should works as well.
> 
> Did you change the panel settings in drawtext example so that they match 80х40 panel size?

yes i change this 

// WF2 GPIO Configuration
#define WF2_X1_R1_PIN 2
#define WF2_X1_R2_PIN 3
#define WF2_X1_G1_PIN 6
#define WF2_X1_G2_PIN 7
#define WF2_X1_B1_PIN 10
#define WF2_X1_B2_PIN 11
#define WF2_X1_E_PIN 21

#define WF2_X2_R1_PIN 4
#define WF2_X2_R2_PIN 5
#define WF2_X2_G1_PIN 8
#define WF2_X2_G2_PIN 9
#define WF2_X2_B1_PIN 12
#define WF2_X2_B2_PIN 13
#define WF2_X2_E_PIN -1        // Currently unknown, so X2 port will not work (yet) with 1/32 scan panels

#define WF2_A_PIN 39
#define WF2_B_PIN 38
#define WF2_C_PIN 37
#define WF2_D_PIN 36
#define WF2_OE_PIN 35
#define WF2_CLK_PIN 34
#define WF2_LAT_PIN 33

#define WF2_BUTTON_TEST     17  // Test key button on PCB, 1=normal, 0=pressed
#define WF2_LED_RUN_PIN     40  // Status LED on PCB
#define WF2_BM8563_I2C_SDA  41  // RTC BM8563 I2C port
#define WF2_BM8563_I2C_SCL  42
#define WF2_USB_DM_PIN 19
#define WF2_USB_DP_PIN 20

#define PANEL_RES_X 80    // Number of pixels wide of each INDIVIDUAL panel module. 
#define PANEL_RES_Y 40    // Number of pixels tall of each INDIVIDUAL panel module.
#define PANEL_CHAIN 1      // Total number of panels chained one to another

board707, 2024-10-01:
Are these settings the same as you used when run the PIO_TestPatterns example?

rosid1, 2024-10-01:
> Are these settings the same as you used when run the PIO_TestPatterns example? What WF2 GPIO port do you use for the HUB panel - X1 or X2?

yes and im using pins X1

board707, 2024-10-01:
Could you show your edited drawtext example?

rosid1, 2024-10-01:
> Could you show the code of your edited drawtext example?

#include <Arduino.h>
#include <ESP32-HUB75-MatrixPanel-I2S-DMA.h>
#include <Adafruit_GFX.h>

// WF2 GPIO Configuration
#define WF2_X1_R1_PIN 2
#define WF2_X1_R2_PIN 3
#define WF2_X1_G1_PIN 6
#define WF2_X1_G2_PIN 7
#define WF2_X1_B1_PIN 10
#define WF2_X1_B2_PIN 11
#define WF2_X1_E_PIN 21

#define WF2_X2_R1_PIN 4
#define WF2_X2_R2_PIN 5
#define WF2_X2_G1_PIN 8
#define WF2_X2_G2_PIN 9
#define WF2_X2_B1_PIN 12
#define WF2_X2_B2_PIN 13
#define WF2_X2_E_PIN -1        // Currently unknown, so X2 port will not work (yet) with 1/32 scan panels

#define WF2_A_PIN 39
#define WF2_B_PIN 38
#define WF2_C_PIN 37
#define WF2_D_PIN 36
#define WF2_OE_PIN 35
#define WF2_CLK_PIN 34
#define WF2_LAT_PIN 33

#define WF2_BUTTON_TEST     17  // Test key button on PCB, 1=normal, 0=pressed
#define WF2_LED_RUN_PIN     40  // Status LED on PCB
#define WF2_BM8563_I2C_SDA  41  // RTC BM8563 I2C port
#define WF2_BM8563_I2C_SCL  42
#define WF2_USB_DM_PIN 19
#define WF2_USB_DP_PIN 20

#define PANEL_RES_X 80    // Number of pixels wide of each INDIVIDUAL panel module. 
#define PANEL_RES_Y 40    // Number of pixels tall of each INDIVIDUAL panel module.
#define PANEL_CHAIN 1      // Total number of panels chained one to another

MatrixPanel_I2S_DMA *matrix = nullptr;

void setup () {
  Serial.begin(115200);
  Serial.println("Starting test text...");
  HUB75_I2S_CFG::i2s_pins _pins_x1 = {WF2_X1_R1_PIN, WF2_X1_G1_PIN, WF2_X1_B1_PIN, WF2_X1_R2_PIN, WF2_X1_G2_PIN, WF2_X1_B2_PIN, WF2_A_PIN, WF2_B_PIN, WF2_C_PIN, WF2_D_PIN, WF2_X1_E_PIN, WF2_LAT_PIN, WF2_OE_PIN, WF2_CLK_PIN};
  HUB75_I2S_CFG::i2s_pins _pins_x2 = {WF2_X2_R1_PIN, WF2_X2_G1_PIN, WF2_X2_B1_PIN, WF2_X2_R2_PIN, WF2_X2_G2_PIN, WF2_X2_B2_PIN, WF2_A_PIN, WF2_B_PIN, WF2_C_PIN, WF2_D_PIN, WF2_X2_E_PIN, WF2_LAT_PIN, WF2_OE_PIN, WF2_CLK_PIN};

  HUB75_I2S_CFG mxconfig(
  PANEL_RES_X,   // module width
  PANEL_RES_Y,   // module height
  PANEL_CHAIN,   // Chain length
  _pins_x1    // pin mapping for port X1
  );
  matrix = new MatrixPanel_I2S_DMA(mxconfig);
  matrix->begin();
  
  
}

void loop (){
  const char *str = "HELLO";
  matrix->setTextColor(matrix->color565(255, 0, 0));
  matrix->setBrightness8(255);
  matrix->setTextSize(2);     // size 1 == 8 pixels high
  matrix->setTextWrap(false);
  matrix->setCursor(1, 1);  // Don't wrap at end of line - will do ourselves
  matrix->print(str);
}

rosid1, 2024-10-01:
[image]

rosid1, 2024-10-01:
[image]

board707, 2024-10-01:
What is the scan of the panel? 
Could you show its rear side?

rosid1, 2024-10-01:
i dont know how it choose to scan the panel

board707, 2024-10-01:
Please show a clear photo of the panel rear side

rosid1, 2024-10-02:
> Please show a clear photo of the panel rear side

this is panel rear side : 
[image]
[image]
[image]

rosid1, 2024-10-02:
> Please show a clear photo of the panel rear side

pls help me @mrcodetastic @board707

board707, 2024-10-02:
Hi
Your panel is 10 scan - so i.e "Quarter scan panel", which requires a specific settings to work.

If you want to set it up, I can help you. You'll need to run a few tests, the code for which I'll give you, and record a video of these tests running on your panel.

rosid1, 2024-10-02:
> Hi Your panel is 10 scan - so i.e "Quarter scan panel", which requires a specific settings to work.
> 
> If you want to set it up, I can help you. You'll need to run a few tests, the code for which I'll give you, and record a video of these tests running on your panel.

okay give me the code, i will test and video the result for you

board707, 2024-10-02:
Try this code. 
(Sorry for any typos, I can't check the code right now)
```
#include <Arduino.h>
#include <ESP32-HUB75-MatrixPanel-I2S-DMA.h>
#include <Adafruit_GFX.h>
#include "ESP32-VirtualMatrixPanel-I2S-DMA.h"

/* ================================================== */
// Define custom class derived from VirtualMatrixPanel
class EightPxBasePanel : public VirtualMatrixPanel
{
  public:
    using VirtualMatrixPanel::VirtualMatrixPanel; // inherit VirtualMatrixPanel's constructor(s)

  protected:

    VirtualCoords getCoords(int16_t x, int16_t y);  // custom getCoords() method for specific pixel mapping

};

// define custom getCoords() method for specific pixel mapping
inline VirtualCoords EightPxBasePanel ::getCoords(int16_t x, int16_t y) {

  coords = VirtualMatrixPanel::getCoords(x, y); // first call base class method to update coords for chaining approach

  if ( coords.x == -1 || coords.y == -1 ) { // Co-ordinates go from 0 to X-1 remember! width() and height() are out of range!
    return coords;
  }

 if (((coords.y / 10)%2) == 0)
        {
            coords.x += ((coords.x / panelResX) + 1) * panelResX; // 1st, 3rd 'block' of 8 rows of pixels, offset by panel width in DMA buffer
        }
        else
        {
            coords.x += (coords.x / panelResX) * panelResX; // 2nd, 4th 'block' of 8 rows of pixels, offset by panel width in DMA buffer
        }

       coords.y = (coords.y / 20) * 10 + (coords.y % 10);
  

  return coords;
}
/* ================================================== */

// WF2 GPIO Configuration
#define WF2_X1_R1_PIN 2
#define WF2_X1_R2_PIN 3
#define WF2_X1_G1_PIN 6
#define WF2_X1_G2_PIN 7
#define WF2_X1_B1_PIN 10
#define WF2_X1_B2_PIN 11
#define WF2_X1_E_PIN 21

#define WF2_X2_R1_PIN 4
#define WF2_X2_R2_PIN 5
#define WF2_X2_G1_PIN 8
#define WF2_X2_G2_PIN 9
#define WF2_X2_B1_PIN 12
#define WF2_X2_B2_PIN 13
#define WF2_X2_E_PIN -1 // Currently unknown, so X2 port will not work (yet) with 1/32 scan panels

#define WF2_A_PIN 39
#define WF2_B_PIN 38
#define WF2_C_PIN 37
#define WF2_D_PIN 36
#define WF2_OE_PIN 35
#define WF2_CLK_PIN 34
#define WF2_LAT_PIN 33

#define WF2_BUTTON_TEST 17 // Test key button on PCB, 1=normal, 0=pressed
#define WF2_LED_RUN_PIN 40 // Status LED on PCB
#define WF2_BM8563_I2C_SDA 41 // RTC BM8563 I2C port
#define WF2_BM8563_I2C_SCL 42
#define WF2_USB_DM_PIN 19
#define WF2_USB_DP_PIN 20

#define PANEL_RES_X 80 // Number of pixels wide of each INDIVIDUAL panel module.
#define PANEL_RES_Y 40 // Number of pixels tall of each INDIVIDUAL panel module.
#define PANEL_CHAIN 1 // Total number of panels chained one to another

// placeholder for the matrix object
MatrixPanel_I2S_DMA *dma_display = nullptr;

// placeholder for the virtual display object
EightPxBasePanel   *FourScanPanel = nullptr;

void setup () {
Serial.begin(115200);
HUB75_I2S_CFG::i2s_pins _pins_x1 = {WF2_X1_R1_PIN, WF2_X1_G1_PIN, WF2_X1_B1_PIN, WF2_X1_R2_PIN, WF2_X1_G2_PIN, WF2_X1_B2_PIN, WF2_A_PIN, WF2_B_PIN, WF2_C_PIN, WF2_D_PIN, WF2_X1_E_PIN, WF2_LAT_PIN, WF2_OE_PIN, WF2_CLK_PIN};

HUB75_I2S_CFG mxconfig(
PANEL_RES_X, // module width
PANEL_RES_Y, // module height
PANEL_CHAIN, // Chain length
_pins_x1 // pin mapping for port X1
);
// OK, now we can create our matrix object
  dma_display = new MatrixPanel_I2S_DMA(mxconfig);

  // let's adjust default brightness to about 75%
  dma_display->setBrightness8(40);    // range is 0-255, 0 - 0%, 255 - 100%

  // Allocate memory and start DMA display
  if ( not dma_display->begin() )
    Serial.println("****** !KABOOM! I2S memory allocation failed ***********");

  dma_display->clearScreen();
  

  // create FourScanPanellay object based on our newly created dma_display object
  FourScanPanel = new EightPxBasePanel ((*dma_display), NUM_ROWS, NUM_COLS, PANEL_RES_X, PANEL_RES_Y,    VIRTUAL_MATRIX_CHAIN_TYPE);

}

void loop (){
 for (int i = 0; i < FourScanPanel->height(); i++)
  {
    for (int j = 0; j < FourScanPanel->width(); j++)
    {

      FourScanPanel->drawPixel(j, i, FourScanPanel->color565(40, 0, 0));
      delay(30);
    }
  }

  delay(2000);
  dma_display->clearScreen();
}
```

rosid1, 2024-10-02:
> Try this code. it is not a solution, just a test. run it on your panel and show the video. (Sorry for any typos, I can't check the code right now)
> 
> ```
> #include <Arduino.h>
> #include <ESP32-HUB75-MatrixPanel-I2S-DMA.h>
> #include <Adafruit_GFX.h>
> #include "ESP32-VirtualMatrixPanel-I2S-DMA.h"
> 
> /* ================================================== */
> // Define custom class derived from VirtualMatrixPanel
> class EightPxBasePanel : public VirtualMatrixPanel
> {
>   public:
>     using VirtualMatrixPanel::VirtualMatrixPanel; // inherit VirtualMatrixPanel's constructor(s)
> 
>   protected:
> 
>     VirtualCoords getCoords(int16_t x, int16_t y);  // custom getCoords() method for specific pixel mapping
> 
> };
> 
> // define custom getCoords() method for specific pixel mapping
> inline VirtualCoords EightPxBasePanel ::getCoords(int16_t x, int16_t y) {
> 
>   coords = VirtualMatrixPanel::getCoords(x, y); // first call base class method to update coords for chaining approach
> 
>   if ( coords.x == -1 || coords.y == -1 ) { // Co-ordinates go from 0 to X-1 remember! width() and height() are out of range!
>     return coords;
>   }
> 
>  if (((coords.y / 10)%2) == 0)
>         {
>             coords.x += ((coords.x / panelResX) + 1) * panelResX; // 1st, 3rd 'block' of 8 rows of pixels, offset by panel width in DMA buffer
>         }
>         else
>         {
>             coords.x += (coords.x / panelResX) * panelResX; // 2nd, 4th 'block' of 8 rows of pixels, offset by panel width in DMA buffer
>         }
> 
>        coords.y = (coords.y / 20) * 10 + (coords.y % 10);
>   
> 
>   return coords;
> }
> /* ================================================== */
> 
> // WF2 GPIO Configuration
> #define WF2_X1_R1_PIN 2
> #define WF2_X1_R2_PIN 3
> #define WF2_X1_G1_PIN 6
> #define WF2_X1_G2_PIN 7
> #define WF2_X1_B1_PIN 10
> #define WF2_X1_B2_PIN 11
> #define WF2_X1_E_PIN 21
> 
> #define WF2_X2_R1_PIN 4
> #define WF2_X2_R2_PIN 5
> #define WF2_X2_G1_PIN 8
> #define WF2_X2_G2_PIN 9
> #define WF2_X2_B1_PIN 12
> #define WF2_X2_B2_PIN 13
> #define WF2_X2_E_PIN -1 // Currently unknown, so X2 port will not work (yet) with 1/32 scan panels
> 
> #define WF2_A_PIN 39
> #define WF2_B_PIN 38
> #define WF2_C_PIN 37
> #define WF2_D_PIN 36
> #define WF2_OE_PIN 35
> #define WF2_CLK_PIN 34
> #define WF2_LAT_PIN 33
> 
> #define WF2_BUTTON_TEST 17 // Test key button on PCB, 1=normal, 0=pressed
> #define WF2_LED_RUN_PIN 40 // Status LED on PCB
> #define WF2_BM8563_I2C_SDA 41 // RTC BM8563 I2C port
> #define WF2_BM8563_I2C_SCL 42
> #define WF2_USB_DM_PIN 19
> #define WF2_USB_DP_PIN 20
> 
> #define PANEL_RES_X 80 // Number of pixels wide of each INDIVIDUAL panel module.
> #define PANEL_RES_Y 40 // Number of pixels tall of each INDIVIDUAL panel module.
> #define PANEL_CHAIN 1 // Total number of panels chained one to another
> 
> // placeholder for the matrix object
> MatrixPanel_I2S_DMA *dma_display = nullptr;
> 
> // placeholder for the virtual display object
> EightPxBasePanel   *FourScanPanel = nullptr;
> 
> void setup () {
> Serial.begin(115200);
> HUB75_I2S_CFG::i2s_pins _pins_x1 = {WF2_X1_R1_PIN, WF2_X1_G1_PIN, WF2_X1_B1_PIN, WF2_X1_R2_PIN, WF2_X1_G2_PIN, WF2_X1_B2_PIN, WF2_A_PIN, WF2_B_PIN, WF2_C_PIN, WF2_D_PIN, WF2_X1_E_PIN, WF2_LAT_PIN, WF2_OE_PIN, WF2_CLK_PIN};
> 
> 
> HUB75_I2S_CFG mxconfig(
> PANEL_RES_X, // module width
> PANEL_RES_Y, // module height
> PANEL_CHAIN, // Chain length
> _pins_x1 // pin mapping for port X1
> );
> // OK, now we can create our matrix object
>   dma_display = new MatrixPanel_I2S_DMA(mxconfig);
> 
>   // let's adjust default brightness to about 75%
>   dma_display->setBrightness8(40);    // range is 0-255, 0 - 0%, 255 - 100%
> 
>   // Allocate memory and start DMA display
>   if ( not dma_display->begin() )
>     Serial.println("****** !KABOOM! I2S memory allocation failed ***********");
> 
> 
>   dma_display->clearScreen();
>   
> 
>   // create FourScanPanellay object based on our newly created dma_display object
>   FourScanPanel = new EightPxBasePanel ((*dma_display), NUM_ROWS, NUM_COLS, PANEL_RES_X, PANEL_RES_Y,    VIRTUAL_MATRIX_CHAIN_TYPE);
> 
> }
> 
> void loop (){
>  for (int i = 0; i < FourScanPanel->height(); i++)
>   {
>     for (int j = 0; j < FourScanPanel->width(); j++)
>     {
> 
>       FourScanPanel->drawPixel(j, i, FourScanPanel->color565(40, 0, 0));
>       delay(30);
>     }
>   }
> 
>   delay(2000);
>   dma_display->clearScreen();
> }
> ```

okay thank you, i have added this code
 #define NUM_ROWS 1 // Number of rows of chained INDIVIDUAL PANELS
 #define NUM_COLS 1 // Number of INDIVIDUAL PANELS per ROW
 #define PANEL_CHAIN NUM_ROWS*NUM_COLS  
 
  what is mean VIRTUAL_MATRIX_CHAIN_TYPE?

board707, 2024-10-02:
> what is mean VIRTUAL_MATRIX_CHAIN_TYPE?

Sorry, you don't need it. Edit the line where it occured such way:
```
FourScanPanel = new EightPxBasePanel ((*dma_display), NUM_ROWS, NUM_COLS, PANEL_RES_X, PANEL_RES_Y);
```

rosid1, 2024-10-02:
> > what is mean VIRTUAL_MATRIX_CHAIN_TYPE?
> 
> Sorry, you don't need it. Edit the line where it occured such way:
> 
> ```
> FourScanPanel = new EightPxBasePanel ((*dma_display), NUM_ROWS, NUM_COLS, PANEL_RES_X, PANEL_RES_Y);
> ```

This is result your code 

https://github.com/user-attachments/assets/8cd84eaa-3841-487d-9c0a-ebae473b9f50

board707, 2024-10-02:
It is not result of my code. My code has nothing to do with text.
Please use my code exactly as is, including the loop() function.

rosid1, 2024-10-02:
your code not add  NUM_ROWS, NUM_COLS ?

board707, 2024-10-02:
Yes, you need to include  NUM_ROWS, NUM_COLS and I said thank you for including it.

But do not edit the loop() code adding a text routine to it. This code is a test of pixel mappings and you text output interfere with it.

rosid1, 2024-10-02:
this is your code that i just added NUM_ROWS, NUM_COLS,

#include <Arduino.h>
#include <ESP32-HUB75-MatrixPanel-I2S-DMA.h>
#include <Adafruit_GFX.h>
#include "ESP32-VirtualMatrixPanel-I2S-DMA.h"

/* ================================================== */
// Define custom class derived from VirtualMatrixPanel
class EightPxBasePanel : public VirtualMatrixPanel
{
  public:
    using VirtualMatrixPanel::VirtualMatrixPanel; // inherit VirtualMatrixPanel's constructor(s)

  protected:

    VirtualCoords getCoords(int16_t x, int16_t y);  // custom getCoords() method for specific pixel mapping

};

// define custom getCoords() method for specific pixel mapping
inline VirtualCoords EightPxBasePanel ::getCoords(int16_t x, int16_t y) {

  coords = VirtualMatrixPanel::getCoords(x, y); // first call base class method to update coords for chaining approach

  if ( coords.x == -1 || coords.y == -1 ) { // Co-ordinates go from 0 to X-1 remember! width() and height() are out of range!
    return coords;
  }

 if (((coords.y / 10)%2) == 0)
        {
            coords.x += ((coords.x / panelResX) + 1) * panelResX; // 1st, 3rd 'block' of 8 rows of pixels, offset by panel width in DMA buffer
        }
        else
        {
            coords.x += (coords.x / panelResX) * panelResX; // 2nd, 4th 'block' of 8 rows of pixels, offset by panel width in DMA buffer
        }

       coords.y = (coords.y / 20) * 10 + (coords.y % 10);
  

  return coords;
}
/* ================================================== */

// WF2 GPIO Configuration
#define WF2_X1_R1_PIN 2
#define WF2_X1_R2_PIN 3
#define WF2_X1_G1_PIN 6
#define WF2_X1_G2_PIN 7
#define WF2_X1_B1_PIN 10
#define WF2_X1_B2_PIN 11
#define WF2_X1_E_PIN 21

#define WF2_X2_R1_PIN 4
#define WF2_X2_R2_PIN 5
#define WF2_X2_G1_PIN 8
#define WF2_X2_G2_PIN 9
#define WF2_X2_B1_PIN 12
#define WF2_X2_B2_PIN 13
#define WF2_X2_E_PIN -1 // Currently unknown, so X2 port will not work (yet) with 1/32 scan panels

#define WF2_A_PIN 39
#define WF2_B_PIN 38
#define WF2_C_PIN 37
#define WF2_D_PIN 36
#define WF2_OE_PIN 35
#define WF2_CLK_PIN 34
#define WF2_LAT_PIN 33

#define WF2_BUTTON_TEST 17 // Test key button on PCB, 1=normal, 0=pressed
#define WF2_LED_RUN_PIN 40 // Status LED on PCB
#define WF2_BM8563_I2C_SDA 41 // RTC BM8563 I2C port
#define WF2_BM8563_I2C_SCL 42
#define WF2_USB_DM_PIN 19
#define WF2_USB_DP_PIN 20

#define PANEL_RES_X 80 // Number of pixels wide of each INDIVIDUAL panel module.
#define PANEL_RES_Y 40 // Number of pixels tall of each INDIVIDUAL panel module.
#define PANEL_CHAIN 1 // Total number of panels chained one to another

 #define NUM_ROWS 1 // Number of rows of chained INDIVIDUAL PANELS
 #define NUM_COLS 1 // Number of INDIVIDUAL PANELS per ROW

// placeholder for the matrix object
MatrixPanel_I2S_DMA *dma_display = nullptr;

// placeholder for the virtual display object
EightPxBasePanel   *FourScanPanel = nullptr;

void setup () {
Serial.begin(115200);
HUB75_I2S_CFG::i2s_pins _pins_x1 = {WF2_X1_R1_PIN, WF2_X1_G1_PIN, WF2_X1_B1_PIN, WF2_X1_R2_PIN, WF2_X1_G2_PIN, WF2_X1_B2_PIN, WF2_A_PIN, WF2_B_PIN, WF2_C_PIN, WF2_D_PIN, WF2_X1_E_PIN, WF2_LAT_PIN, WF2_OE_PIN, WF2_CLK_PIN};

HUB75_I2S_CFG mxconfig(
PANEL_RES_X, // module width
PANEL_RES_Y, // module height
PANEL_CHAIN, // Chain length
_pins_x1 // pin mapping for port X1
);
// OK, now we can create our matrix object
  dma_display = new MatrixPanel_I2S_DMA(mxconfig);

  // let's adjust default brightness to about 75%
  dma_display->setBrightness8(40);    // range is 0-255, 0 - 0%, 255 - 100%

  // Allocate memory and start DMA display
  if ( not dma_display->begin() )
    Serial.println("****** !KABOOM! I2S memory allocation failed ***********");

  dma_display->clearScreen();
  

  // create FourScanPanellay object based on our newly created dma_display object
  FourScanPanel = new EightPxBasePanel ((*dma_display), NUM_ROWS, NUM_COLS, PANEL_RES_X, PANEL_RES_Y);

}

void loop (){
 for (int i = 0; i < FourScanPanel->height(); i++)
  {
    for (int j = 0; j < FourScanPanel->width(); j++)
    {

      FourScanPanel->drawPixel(j, i, FourScanPanel->color565(40, 0, 0));
      delay(30);
    }
  }

  delay(2000);
  dma_display->clearScreen();
}

here is the result of the video : 

https://github.com/user-attachments/assets/02f611e8-694c-4690-b1b4-8ad6acca39ed

board707, 2024-10-02:
Please change config to this
```
HUB75_I2S_CFG mxconfig(
PANEL_RES_X * 2,            // DO NOT CHANGE THIS
PANEL_RES_Y / 2,   
PANEL_CHAIN, // Chain length
_pins_x1 // pin mapping for port X1
);
```
and run again

board707, 2024-10-02:
Thank you very much
Then change the `getCoords(int16_t x, int16_t y) ` function this way:
```
inline VirtualCoords EightPxBasePanel ::getCoords(int16_t x, int16_t y) {

  coords = VirtualMatrixPanel::getCoords(x, y); // first call base class method to update coords for chaining approach

  if ( coords.x == -1 || coords.y == -1 ) { // Co-ordinates go from 0 to X-1 remember! width() and height() are out of range!
    return coords;
  }
const uint8_t pixBase =8;
 if (((coords.y / 10)%2) == 1)
        {
            coords.x = (coords.x / pxbase)*2*pxbase   + 7 - (coords.x & 0x7);
        }
        else
        {
           coords.x += ((coords.x / pxbase) + 1) * pxbase; 
        }

       coords.y = (coords.y / 20) * 10 + (coords.y % 10);
  

  return coords;
}
```

rosid1, 2024-10-02:
pxbase means pixBase? because pxbase is undefined

board707, 2024-10-02:
Yes, sure

rosid1, 2024-10-03:
this is the result of the video after i changed function getCoords(int16_t x, int16_t y) 

https://github.com/user-attachments/assets/190552d4-9260-4fbc-a137-27f603abdca2

board707, 2024-10-03:
@rosid1 
Your video is too dim, I can't see anything. Changing the code couldn't cause such a change in brightness, check your connections.
Show your whole last code please.

rosid1, 2024-10-03:
this is the last code that i flashed
`#include <Arduino.h>
#include <ESP32-HUB75-MatrixPanel-I2S-DMA.h>
#include <Adafruit_GFX.h>
#include "ESP32-VirtualMatrixPanel-I2S-DMA.h"

/* ================================================== */
// Define custom class derived from VirtualMatrixPanel
class EightPxBasePanel : public VirtualMatrixPanel
{
  public:
    using VirtualMatrixPanel::VirtualMatrixPanel; // inherit VirtualMatrixPanel's constructor(s)

  protected:

    VirtualCoords getCoords(int16_t x, int16_t y);  // custom getCoords() method for specific pixel mapping

};

// define custom getCoords() method for specific pixel mapping
inline VirtualCoords EightPxBasePanel ::getCoords(int16_t x, int16_t y) {

  coords = VirtualMatrixPanel::getCoords(x, y); // first call base class method to update coords for chaining approach

  if ( coords.x == -1 || coords.y == -1 ) { // Co-ordinates go from 0 to X-1 remember! width() and height() are out of range!
    return coords;
  }
const uint8_t pixBase =8;
 if (((coords.y / 10)%2) == 1)
        {
            coords.x = (coords.x / pixBase)*2*pixBase   + 7 - (coords.x & 0x7);
        }
        else
        {
           coords.x += ((coords.x / pixBase) + 1) * pixBase; 
        }

       coords.y = (coords.y / 20) * 10 + (coords.y % 10);
  

  return coords;
}
/* ================================================== */

// WF2 GPIO Configuration
#define WF2_X1_R1_PIN 2
#define WF2_X1_R2_PIN 3
#define WF2_X1_G1_PIN 6
#define WF2_X1_G2_PIN 7
#define WF2_X1_B1_PIN 10
#define WF2_X1_B2_PIN 11
#define WF2_X1_E_PIN 21

#define WF2_X2_R1_PIN 4
#define WF2_X2_R2_PIN 5
#define WF2_X2_G1_PIN 8
#define WF2_X2_G2_PIN 9
#define WF2_X2_B1_PIN 12
#define WF2_X2_B2_PIN 13
#define WF2_X2_E_PIN -1 // Currently unknown, so X2 port will not work (yet) with 1/32 scan panels

#define WF2_A_PIN 39
#define WF2_B_PIN 38
#define WF2_C_PIN 37
#define WF2_D_PIN 36
#define WF2_OE_PIN 35
#define WF2_CLK_PIN 34
#define WF2_LAT_PIN 33

#define WF2_BUTTON_TEST 17 // Test key button on PCB, 1=normal, 0=pressed
#define WF2_LED_RUN_PIN 40 // Status LED on PCB
#define WF2_BM8563_I2C_SDA 41 // RTC BM8563 I2C port
#define WF2_BM8563_I2C_SCL 42
#define WF2_USB_DM_PIN 19
#define WF2_USB_DP_PIN 20

#define PANEL_RES_X 80 // Number of pixels wide of each INDIVIDUAL panel module.
#define PANEL_RES_Y 40 // Number of pixels tall of each INDIVIDUAL panel module.
#define PANEL_CHAIN 1 // Total number of panels chained one to another

 #define NUM_ROWS 1 // Number of rows of chained INDIVIDUAL PANELS
 #define NUM_COLS 1 // Number of INDIVIDUAL PANELS per ROW

// placeholder for the matrix object
MatrixPanel_I2S_DMA *dma_display = nullptr;

// placeholder for the virtual display object
EightPxBasePanel   *FourScanPanel = nullptr;

void setup () {
Serial.begin(115200);
HUB75_I2S_CFG::i2s_pins _pins_x1 = {WF2_X1_R1_PIN, WF2_X1_G1_PIN, WF2_X1_B1_PIN, WF2_X1_R2_PIN, WF2_X1_G2_PIN, WF2_X1_B2_PIN, WF2_A_PIN, WF2_B_PIN, WF2_C_PIN, WF2_D_PIN, WF2_X1_E_PIN, WF2_LAT_PIN, WF2_OE_PIN, WF2_CLK_PIN};

HUB75_I2S_CFG mxconfig(
PANEL_RES_X*2, // module width
PANEL_RES_Y/2, // module height
PANEL_CHAIN, // Chain length
_pins_x1 // pin mapping for port X1
);
// OK, now we can create our matrix object
  dma_display = new MatrixPanel_I2S_DMA(mxconfig);

  // let's adjust default brightness to about 75%
  dma_display->setBrightness8(40);    // range is 0-255, 0 - 0%, 255 - 100%

  // Allocate memory and start DMA display
  if ( not dma_display->begin() )
    Serial.println("****** !KABOOM! I2S memory allocation failed ***********");

  dma_display->clearScreen();
  

  // create FourScanPanellay object based on our newly created dma_display object
  FourScanPanel = new EightPxBasePanel ((*dma_display), NUM_ROWS, NUM_COLS, PANEL_RES_X, PANEL_RES_Y);

}

void loop (){
 for (int i = 0; i < FourScanPanel->height(); i++)
  {
    for (int j = 0; j < FourScanPanel->width(); j++)
    {

      FourScanPanel->drawPixel(j, i, FourScanPanel->color565(40, 0, 0));
      delay(30);
    }
  }

  delay(40);
  dma_display->clearScreen();
}`

here is the result of the video :

https://github.com/user-attachments/assets/1180d292-0751-4919-a4bd-9d773e2e0c95

board707, 2024-10-03:
Thank you very much.
The last edition I hope
```
inline VirtualCoords EightPxBasePanel ::getCoords(int16_t x, int16_t y) {

  coords = VirtualMatrixPanel::getCoords(x, y); // first call base class method to update coords for chaining approach

  if ( coords.x == -1 || coords.y == -1 ) { // Co-ordinates go from 0 to X-1 remember! width() and height() are out of range!
    return coords;
  }
const uint8_t pixbase =8;
 if (((coords.y / 10)%2) == 0)
        {
            coords.x = (coords.x / pxbase)*2*pxbase   + 7 - (coords.x & 0x7);
        }
        else
        {
           coords.x += ((coords.x / pxbase) + 1) * pxbase; 
        }

       coords.y = (coords.y / 20) * 10 + (coords.y % 10);
  

  return coords;
}
```
Please show the video for the more than quarter of panel

rosid1, 2024-10-03:
@board707 Thank you for helping me get here, here is the video result:

https://github.com/user-attachments/assets/72a49476-7029-45b9-ba34-d27a06cf7c1d

I want to ask the last one, I have a panel with this type 80x16 20s, for which part of the code do I change or is there a calculation to scan this panel. 
[image]

board707, 2024-10-03:
> here is the video result:

Now you can test it with text.

> I have a panel with this type 80x16 20s

On the picture I see the label "80x16 16s". Could you please take a photo for general view of the panel - one for the front and the other from the rear side.

rosid1, 2024-10-03:
This is the results of the test text and panel photo "80x16 16s"
[image]

[image]
[image]
[image]

board707, 2024-10-03:
Unfortunately, I haven't worked with panels that have the same height as the scan.

You can try running it by configuring the library for a matrix of 80x32. Of course, in that case you will only be able to output data to the upper half of 32 lines.

rosid1, 2024-10-03:
do you mean this code is changed to 80x32?
#define PANEL_RES_X 80
#define PANEL_RES_Y 32

board707, 2024-10-03:
Get a library's standard example (not our code for your 80x40 panel) and set the panel dimensions 80x32. I think, you do not any specific settings for it.

board707, 2024-10-03:
Yes, the library we are currently in ...

mohsin-rauma, 2024-10-04:
@board707 i have this ( p4-8040-20s-v2.2 ) panel i want to make clock using esp32 or esp8266 board if you have code pls give me i never work with rgb led panel

rosid1, 2024-10-05:
Hello @board707 , how to change this font montserrat to library?
[Montserrat.zip](https://github.com/user-attachments/files/17265691/Montserrat.zip)

[35 more replies]