mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #455: 32x16 CHAIN_TOP_RIGHT_DOWN 3 Rows 1 Col rotated not showing in all displays

jfbvm, 2023-05-13:
Hi Mrfaptastic,

First of all, many many thanks for this library, really, it's a real piece of art!! 👏👏👏
And congrats for all the contributions from the community, I read almost all the issues and I saw all the work until now.

I saw in the readme file, if I understood it correctly, that your library will not work with 1/8 displays, but In fact it works like a charm, if I do not try to rotate...😅😅
So, I'm facing this problem, like I said in the title, I'm using "3 32x16 displays" "CHAIN_TOP_RIGHT_DOWN", and when I use the function "virtualDisp->setRotate(true);", the print function doesn't display in all displays, please see my code below.
Another point, when I try to use only the function "virtualDisp->setRotation(1)", nothing happens. It's the same for the values 2 and 3 for the rotation value.
I also tried to use the setRotation function from the "Adafruit_GFX library", but the result is a mess, so I abandoned this idea.

```cpp
// 1) Include key virtual display library
// VERSION 3.0.8

#include <ESP32-VirtualMatrixPanel-I2S-DMA.h>
#include <Fonts/FreeMono9pt7b.h>

// 2) Set configuration
#define PANEL_RES_X 32 // Number of pixels wide of each INDIVIDUAL panel module.
#define PANEL_RES_Y 16 // Number of pixels tall of each INDIVIDUAL panel module.

#define NUM_ROWS 3                     // Number of rows of chained INDIVIDUAL PANELS
#define NUM_COLS 1                     // Number of INDIVIDUAL PANELS per ROW
#define PANEL_CHAIN NUM_ROWS *NUM_COLS // total number of panels chained one to another

/* Configure the serpetine chaining approach. Options are:
CHAIN_NONE,
CHAIN_TOP_LEFT_DOWN,
CHAIN_TOP_RIGHT_DOWN,
CHAIN_BOTTOM_LEFT_UP,
CHAIN_BOTTOM_RIGHT_UP,
CHAIN_TOP_LEFT_DOWN_ZZ, /// ZigZag chaining. Might need a big ass cable to do this, all panels right way up.
CHAIN_TOP_RIGHT_DOWN_ZZ,
CHAIN_BOTTOM_RIGHT_UP_ZZ,
CHAIN_BOTTOM_LEFT_UP_ZZ

The location (i.e. 'TOP_LEFT', 'BOTTOM_RIGHT') etc. refers to the starting point where
the ESP32 is located, and how the chain of panels will 'roll out' from there.

In this example we're using 'CHAIN_BOTTOM_LEFT_UP' which would look like this in the real world:

Chain of 4 x 64x32 panels with the ESP at the CHAIN_TOP_LEFT_DOWN:

+---------+---------+
|    4    |    3    |
|         |         |
+---------+---------+
|    1    |    2    |
|  (ESP)  |         |
+---------+---------+

*/

#define VIRTUAL_MATRIX_CHAIN_TYPE CHAIN_TOP_RIGHT_DOWN

// 3) Create the runtime objects to use

// placeholder for the matrix object
MatrixPanel_I2S_DMA *dma_display = nullptr;

// placeholder for the virtual display object
VirtualMatrixPanel *virtualDisp = nullptr;

/******************************************************************************
 * Setup!
 ******************************************************************************/
void setup()
{

    delay(2000);
    Serial.begin(115200);

    /******************************************************************************
     * Create physical DMA panel class AND virtual (chained) display class.
     ******************************************************************************/

    HUB75_I2S_CFG mxconfig(
        PANEL_RES_X, // module width
        PANEL_RES_Y, // module height
        PANEL_CHAIN  // chain length
    );

    // mxconfig.driver = HUB75_I2S_CFG::FM6126A;     // in case that we use panels based on FM6126A chip, we can set it here before creating MatrixPanel_I2S_DMA object

    // Sanity checks
    if (NUM_ROWS <= 1){
        Serial.println(F("There is no reason to use the VirtualDisplay class for a single horizontal chain and row!"));
    }

    // OK, now we can create our matrix object
    dma_display = new MatrixPanel_I2S_DMA(mxconfig);

    // dma_display->setRotation(1);

    // let's adjust default brightness to about 75%
    dma_display->setBrightness8(192); // range is 0-255, 0 - 0%, 255 - 100%

    // Allocate memory and start DMA display
    if (not dma_display->begin()) Serial.println("****** !KABOOM! I2S memory allocation failed ***********");

    // create VirtualDisplay object based on our newly created dma_display object
    virtualDisp = new VirtualMatrixPanel((*dma_display), NUM_ROWS, NUM_COLS, PANEL_RES_X, PANEL_RES_Y, VIRTUAL_MATRIX_CHAIN_TYPE);

    // virtualDisp->setPhysicalPanelScanRate(NORMAL_ONE_SIXTEEN);
    // virtualDisp->setPhysicalPanelScanRate(FOUR_SCAN_16PX_HIGH);

    // So far so good, so continue
    virtualDisp->fillScreen(virtualDisp->color444(0, 0, 0));
    // virtualDisp->drawDisplayTest(); // draw text numbering on each screen to check connectivity

    // delay(1000);

    virtualDisp->setTextColor(virtualDisp->color565(0, 0, 255));

    // Red text inside red rect (2 pix in from edge)
    virtualDisp->drawRect(0, 0, virtualDisp->width(), virtualDisp->height(), virtualDisp->color565(255, 0, 0));

    // White line from top left to bottom right
    // virtualDisp->drawLine(0, 0, virtualDisp->width() - 1, virtualDisp->height() - 1, virtualDisp->color565(255, 255, 255));

    // virtualDisp->drawDisplayTest(); // re draw text numbering on each screen to check connectivity

    virtualDisp->setRotate(true);

    virtualDisp->setTextWrap(false);

    virtualDisp->setCursor(1, 1);
    virtualDisp->print("ABCDEFGHIJ");
    virtualDisp->setCursor(1, 8);
    virtualDisp->print("1234567890");
    virtualDisp->setCursor(1, 16);
    virtualDisp->print("ABCDEFGHIJ");
    virtualDisp->setCursor(1, 24);
    virtualDisp->print("1234567890");
    virtualDisp->setCursor(1, 32);
    virtualDisp->print("ABCDEFGHIJ");
    virtualDisp->setCursor(1, 40);
    virtualDisp->print("1234567890");
    virtualDisp->setCursor(1, 48);
    virtualDisp->print("ABCDEFGHIJ");
    
}

void loop(){

}
```

The result when not using rotate function:

[image]

The result when I **use the rotate function**:

[image]

Please see below the wire connections:

[image]

Please see below the display model that I'm using:

[image]

mrfaptastic, 2023-05-13:
setRotate shouldn't even exist as a function anymore.

Can you try with the git master version?

Alternatively if you change the panel chaining type from TOP_LEFT to BOTTOM_RIGHT it essentially rotates for you as well.

jfbvm, 2023-05-13:
> setRotate shouldn't even exist as a function anymore.
> 
> Can you try with the git master version?
> 
> Alternatively if you change the panel chaining type from TOP_LEFT to BOTTOM_RIGHT it essentially rotates for you as well.

Hummm, many thanks for the reply, let me try it...

Sorry to ask, but isn't the last version 3.0.8 the master version?
I forgot to say, I installed your library from the Arduino 1.8.19 library manager.

jfbvm, 2023-05-13:
Yes, you are correct, I think some error occurred when I downloaded using Arduino 1.8.19 library manager because now I made a test downloading the git master version as a zip file, and the function setRotate doesn't exist...😅😅

I'll try installing your library using the "install from zip file" function in Arduino 1.8.19. In fact, I'm using VS code to compile and upload the code to Esp32. The purpose of Arduino software for me is only to manage the libraries.

jfbvm, 2023-05-13:
> 

WORKED MAN WORKED!!!!! MANY THANKS!!! 🙏🙏🙏🙏

I only need to install the library using "install from zip file" from Arduino IDE, now the correct version of the library was installed!!!

And now I can use the "setRotation" function normally!

Many thanks again for your help and support, and congrats again for this library!!!

[image]

closed 2023-05-13 by jfbvm

mrfaptastic, 2023-05-14:
Thank you for your kind words! Glad the library works for you 

I think I need to release a new version as there are some fixes (namely this rotation) that might not be in the latest release.

jfbvm, 2023-05-15:
Haa man, you deserve these words!!
I'm a huge fan of opensource, simple because of this: besides the fact that it solves some really big problems (like mine 😅😅), I know they are created in most cases for people that don't expect nothing in exchange (and I know that they deserves much for their work), but I also know that the universe, in your great kindness and justice, will not let them with empty hands!!
I always say it to my son, someone is always watching our acts, and "we always reap what we sow".
Congrats again for you and all the contributors for this great work again!!