mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #456: Certain 64x64 display not showing anything

lohant, 2023-05-13:
I have multiple files (including provided example files from here) that are working fine on one of my 64x64 led matrixes, but the new 64x64 matrixes I received yesterday does not work with that code. I am using the same unit to connect and run the code, namely the ESP32 Trinity. I received five identical matrices yesterday and none works so it is highly unlikely that the problem is with the hardware.

These are the matrices that does not work: https://www.alibaba.com/product-detail/Led-Video-Wall-Led-Advertising-Module_62222559088.html?spm=a2700.shop_plgr.41413.3.5d621d582GGYWX

I suspect that the error is with the driver of the IC. It says that it uses the ICN2038S chip on the specification but from what I can see on the ICs themselves they say "TC7262". I have tried the PatternPlasma example as well as tried all the different driver configurations (mxconfig.driver = HUB75_I2S_CFG::ICN2038S; etc) but with no success.

On the HUB75 contact there is a difference between the panels, the new ones got a pin "D" where the old one is marked "GND".

I am not sure what to try next and would be very grateful for any help or suggestions I could get!

mrfaptastic, 2023-05-14:
Can you take a close up photo of the back for each of the main chips?

mrfaptastic, 2023-05-14:
> I am using the same unit to connect and run the code, namely the ESP32 Trinity. I received five identical matrices yesterday and none works so it is highly unlikely that the problem is with the hardware.

> On the HUB75 contact there is a difference between the panels, the new ones got a pin "D" where the old one is marked "GND".

Thinking about this more. I think you're screwed because of the Trinity PCB wiring assumes that that new 'D' pin shouldn't exist, then it's always going to be connected to ground right?

You need to hook these new panels up to an esp32 yourself wire-by-wire. That new D pin most likely needs to connect to something.

board707, 2023-05-14:
> I suspect that the error is with the driver of the IC. It says that it uses the ICN2038S chip on the specification but from what I can see on the ICs themselves they say "TC7262".

Every LED panel contains at least two types IC - LED driver itself and a line decoder (or multiplexor) - to switch the input data between lines of leds.
TC7262 is not a driver, it is a multiplexor with 138 type decoding ( standard decoding, compatible with the library).

However in order to understand if your panel will work, it is more important to know the type of LED driver chip

lohant, 2023-05-15:
Thank you both for your valuable feedback. After investigating the backside once again, I found that it uses the ICND2153. After looking at some data sheets I came to the conclusion that this IC is not supported by the library. This turned out to be the case, however I got it working with this library: https://github.com/LAutour/ESP32-HUB75-MatrixPanel-I2S-DMA-icn2053.

Do you think these types of ICs are starting to become the new standard? Or will they both continue to be used in parallel? I'm quite good at coding but electronics is new to me so I'm not really able to asses the situation.

board707, 2023-05-15:
> Do you think these types of ICs are starting to become the new standard?

For big panels the answer is definitely yes.

Most companies producing chips for LED panels (Chipone, Depuw, Fuman ) now have approximately the same ratio in announcements - 1-2 traditional drivers and 5-6 or even more new chips similar to 2153.

lohant, 2023-05-15:
Ah I see, interesting.

I have a question related to this that came up, you don't happen to know how to solve this?

I'm using the albumArtMatrix example (from: https://github.com/witnessmenow/spotify-api-arduino/blob/899502c193374daf054c8a36dc96ac2746d8d0fc/examples/displayAlbumArt/albumArtMatrix/albumArtMatrix.ino#L59) with my old matrix. 

However, I would like to use it with the new matrix that uses the ICND21153 IC and that I control with this library: https://github.com/LAutour/ESP32-HUB75-MatrixPanel-DMA-ICN2053.

I swapped to the new library in the code but ran into some problems.

When I plugged it in, this did not compile: dma_display->fillScreen(dma_display->color565(0, 0, 0));

Furthermore, when I commented the problematic rows out, the code compiled but the album art does not show up. I have previously modified the code to work with this library (https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-DMA).
I suspect that there is something that needs to be changed in the following function now:

// This next function will be called during decoding of the jpeg file to
// render each block to the Matrix.  If you use a different display
// you will need to adapt this function to suit.

int JPEGDraw(JPEGDRAW *pDraw) {
  // Stop further decoding as image is running off bottom of screen
  if (  pDraw->y >= dma_display->height() ) return 1;

  dma_display->drawRGBBitmap(pDraw->x, pDraw->y, pDraw->pPixels, pDraw->iWidth, pDraw->iHeight);
 return 1;
}

I would be very thankful if you could point me in the right direction to make this work. I understand that this will need some specific work done but if I could get some help in knowing what needs to be done it would be great.

board707, 2023-05-15:
@lohant 
In my opinion - everything you write, has nothing to do with which driver is used on the panel. All of the above problems are related to your custom code, not to the library, and therefore you know better how to fix them.

lohant, 2023-05-16:
@board707 
You are right.

After playing around with it I was able to make it work, although with some problems. It is very slow to update the images now compared to previously. I think I'm doing something wrong.

I realized that the flipDMABuffer() is used in the library by LAutour but not this one. Do you know what that function does and when to use it? In the one example code it is used after every command that makes something appear on the matrix. I tried adding it but I think theres some error on my side. 

Furthermore, color565() does not work. Isn't that a part of the Adafruit library? And thus should be available in all libraries that builds on it? In this (mrfaptastic) library I've used print to show text on the matrix, this does not work with the LAutour library.

Lastly, when the matrix is showing what it should the colors seem to be off and the image looks faded.

I understand that these are software problems but I would be grateful if anyone could help point me in the right direction.