mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #461: How to use the library with freertos?

jfbvm, 2023-05-17:
I need to receive data from Serial, and don't want to put the serial code at the setup block, so I'm trying to use freertos to create an "xTaskCreatePinnedToCore" or "xTaskCreate" to "Listening" the serial on TX2/RX2 in another thread.

@board707
@danieldecesaro
@vortigont 
@mrRobot62

Everything is working fine, I'm using virtual matrix, but if I use the function "xTaskCreatePinnedToCore( handleSerialComm_Task, "serialComm", 1024,  NULL, 1, NULL, 1 );" to the core 1 or even if I let freertos bind to a core with the function "xTaskCreate", the board starts to crash and restart...

I decoded the exception generated by the board, but it only says that there is an error in the line inside the function handleSerialComm_Task, which only does a "Serial.println("TEST...");"

What could be the problem, memory? Or something else?

mrfaptastic, 2023-05-18:
If you don't use Serial... what happens?

jfbvm, 2023-05-18:
In fact the serial isn't the problem, the problem, I thought, is freertos. Maybe it's consuming to much memory, because in the first moment that I use it, I saw an error message about memory...

board707, 2023-05-18:
Why not to use Serial without a RTOS ?

jfbvm, 2023-05-18:
NOOOO, my bad 🤦‍♂️, I forgot to use the while(1){} inside the task. It acts like a "loop" inside the task.

```c++
void handleSerialComm_Task(void * parameter){
    while(1){
        if (SerialPort.available()){
            char number = SerialPort.read();
            if (number == '0'){
                // digitalWrite(LED, LOW);
            }
            if (number == '1'){
                // digitalWrite(LED, HIGH);
            }
        }
        delay(2000);
    }
}
```

jfbvm, 2023-05-18:
I updated the title to let it be like a piece of advice in case someone falls into the same mistake...😅😅

closed 2023-05-18 by jfbvm