mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #467: panel chain limit?(chained more than 7 doesn't work)

Jiwno, 2023-06-07:
Is there a limit of chained panel number?

I'm trying to chain 12 panel(96 * 48 pixels) with one esp32 but if I chain more than 7 panel at once, it doesn't work at all.

I checked the power and it wasn't power problem.

I'm using 96 * 48 pixels panel in a row.

Jiwno, 2023-06-07:
#include <ESP32-HUB75-MatrixPanel-I2S-DMA.h>

#define PANEL_RES_X 96      // Number of pixels wide of each INDIVIDUAL panel module. 
#define PANEL_RES_Y 48     // Number of pixels tall of each INDIVIDUAL panel module.
#define PANEL_CHAIN 7      // Total number of panels chained one to another
#define PIN_E 32 

 
//MatrixPanel_I2S_DMA dma_display;
MatrixPanel_I2S_DMA *dma_display = nullptr;
uint16_t myWHITE = dma_display->color565(255, 255, 255);

void setup() {
  Serial.begin(115200);

  // Module configuration
  HUB75_I2S_CFG mxconfig(
    PANEL_RES_X,   // module width
    PANEL_RES_Y,   // module height
    PANEL_CHAIN    // Chain length
  );
  mxconfig.gpio.e = PIN_E;   

  // Display Setup
  dma_display = new MatrixPanel_I2S_DMA(mxconfig);
  dma_display->begin();
  dma_display->setBrightness8(90); //0-255
  dma_display->clearScreen();
  dma_display->fillScreen(myWHITE);
}

void loop() {
//  dma_display->fillScreen(myWHITE);
  dma_display->clearScreen();

  for (int y=0; y<PANEL_RES_Y; y++){
    for (int x=0; x<PANEL_RES_X * PANEL_CHAIN; x++){
      dma_display->drawPixelRGB888(x, y, 255, 255, 255);
      delay(1);
    }
  }
}

This it the code I'm running

board707, 2023-06-07:
> Is there a limit of chained panel number?

There is no chain limit , but there is a limit of memory used.
Your panel size 96x48 is quite large, chaining 12 of them will create a screen with 580x96 dimensions.
I think this required too much memory for work. Start from decreasing of colour depth to, for example, 3-4 bits per color

Jiwno, 2023-06-08:
Oh, I got it.
Thanks