mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #469: Waveshare 64x32 panels intermittent "boot" issue

elan, 2023-06-20:
Using these panels https://www.waveshare.com/rgb-matrix-p2.5-64x32.htm with ESP32 S3. Powering them with a lab power supply so it should have plenty of power.

When the board boots, about 50% of the time I end up with garbage on the matrix like the photo below (sometimes monochromatic, sometimes not). No matter what I do subsequently, it doesn't change what's displayed. Via serial I can see the processing is booting, on the network, everything seems to be fine. But somehow the panel seems to have failed to "initialize" or something like that. I reboot a few times, and then it works and keeps working.

Any ideas?

[image]

board707, 2023-06-20:
Try to pulldown OE pin by external resistor, so it will be LOW while board boots.

elan, 2023-07-05:
Thanks, will give this a try!

elan, 2023-07-05:
I played with this a bit and connected a 10K pull-up resistor, but this didn't seem to solve the issue. Looking in [the data sheet](https://olympianled.com/wp-content/uploads/2021/05/ICN2037_datasheet_EN_2017_V2.0.pdf) for the chip, it looks like the chip already has an internal one?

I then tried the time-honored tradition of just waiting a bit after boot to initialize the display, and waiting 500ms seemed to do the trick.

closed 2023-07-05 by elan

elan, 2023-07-05:
Actually, that didn't solve it, still seeing sporadic issues on boot. Only other thing I can think of is a power issue (I'm powering it with a lab power supply which in theory should work fine), but maybe putting a large capacitor across the power pins per recommendation might help.

davemaster, 2023-07-05:
About this issue, does anyone knows what individual rgb led are used in this rgb led panel? Thanks in advance.