mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #470: trying to make a virtualDisp for 4 64x32 panels into Serail.read()

truckerreid, 2023-06-22:
Trying to make a serial read so this will work with VixenLights which sends out a serial signal to the board, 
It worked ok for the commands, but obviously I do not have enough experiance to get it to work. 
I tried finding tutorials with no luck. 
Any help would be appreciated.
Thanks

`/******************************************************************************
    -------------------------------------------------------------------------
    Steps to create a virtual display made up of a chain of panels in a grid
    -------------------------------------------------------------------------
    Read the documentation!
    https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-DMA/tree/master/examples/ChainedPanels
    tl/dr:
    
    - Set values for NUM_ROWS, NUM_COLS, PANEL_RES_X, PANEL_RES_Y, PANEL_CHAIN_TYPE. 
    - Other than where the matrix is defined and matrix.begin in the setup, you 
      should now be using the virtual display for everything (drawing pixels, writing text etc).
       You can do a find and replace of all calls if it's an existing sketch
      (just make sure you don't replace the definition and the matrix.begin)
    - If the sketch makes use of MATRIX_HEIGHT or MATRIX_WIDTH, these will need to be 
      replaced with the width and height of your virtual screen. 
      Either make new defines and use that, or you can use virtualDisp.width() or .height()
*****************************************************************************/
// 1) Include key virtual display library
    #include <ESP32-VirtualMatrixPanel-I2S-DMA.h>
    #include <FastLED.h>
    
// 2) Set configuration
    #define PANEL_RES_X 64 // Number of pixels wide of each INDIVIDUAL panel module. 
    #define PANEL_RES_Y 32 // Number of pixels tall of each INDIVIDUAL panel module.

    #define NUM_ROWS 2 // Number of rows of chained INDIVIDUAL PANELS
    #define NUM_COLS 2 // Number of INDIVIDUAL PANELS per ROW
    #define PANEL_CHAIN NUM_ROWS*NUM_COLS    // total number of panels chained one to another

    /* Configure the serpetine chaining approach. Options are:
        CHAIN_TOP_LEFT_DOWN
        CHAIN_TOP_RIGHT_DOWN
        CHAIN_BOTTOM_LEFT_UP
        CHAIN_BOTTOM_RIGHT_UP
        The location (i.e. 'TOP_LEFT', 'BOTTOM_RIGHT') etc. refers to the starting point where 
        the ESP32 is located, and how the chain of panels will 'roll out' from there.
        In this example we're using 'CHAIN_BOTTOM_LEFT_UP' which would look like this in the real world:
        Chain of 4 x 64x32 panels with the ESP at the BOTTOM_LEFT:
        +---------+---------+
        |    4    |    3    |
        |         |         |
        +---------+---------+
        |    1    |    2    |
        |  (ESP)  |         |
        +---------+---------+      
    */
    #define VIRTUAL_MATRIX_CHAIN_TYPE CHAIN_BOTTOM_LEFT_UP 

// 3) Create the runtime objects to use

    // placeholder for the matrix object
    MatrixPanel_I2S_DMA *dma_display = nullptr;

    // placeholder for the virtual display object
    VirtualMatrixPanel  *virtualDisp = nullptr;
// placeholder for the matrix object
// MatrixPanel_I2S_DMA *dma_display = nullptr;

/********************************
It did not like this section

const uint16_t colors[] =
{
  virtualDisp->Color(255, 0, 0), virtualDisp->Color(0, 255, 0), virtualDisp->Color(0, 0, 255)
[... 186 lines cut ...]
    // print FPS rate every 5 seconds
    // Note: this is NOT a matrix refresh rate, it's the number of data frames being drawn to the DMA buffer per second
    if (fps_timer + 5000 < millis()){
      Serial.printf_P(PSTR("Effect fps: %d\n"), fps/5);
      fps_timer = millis();
      fps = 0;
    }
*/
}//END LOOP
/*****************************************************************************
    Thanks to:
    * Brian Lough for the original example as raised in this issue:
      https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-I2S-DMA/issues/26
      YouTube: https://www.youtube.com/brianlough
      Tindie: https://www.tindie.com/stores/brianlough/
      Twitter: https://twitter.com/witnessmenow
    * Galaxy-Man for the kind donation of panels make/test that this is possible:
      https://github.com/Galaxy-Man
`*****************************************************************************/`

it works fine for the normal commands just not the plasma effect.
[image]
[image]
[image]
[image]

board707, 2023-06-23:
Hi
Please explain the section below:
```
void chartoscreen()
{
  if (readString.length() > 0)  //if the number of bytes in readString are greater than 0
  {
    virtualDisp->println(readString);  //print the data in readString
    pass = -abs(readString.length());  //set pass to the negative value of the number of characters

    if(--x < pass * 6)  //reserve 6 spaces for each character - continue looping until x = pass * 6
    {
      x = virtualDisp->width();  //set x to the number of colums in the matrix
      readString = "";  //set the data to blank
    }
    virtualDisp->show();  //show the data stored
    delay(70);  //wait a bit - sets scrolling speed
  }
}}
```
Why `x` is decrementing while print and what purpose to make `pass` negative? In general, what do you try to achieve with this code?

truckerreid, 2023-06-23:
Every where that it said virtualDisp it used to say matrix. and that is what I had to do to get it display on the 2812 led display matrix. so I presume that it was the whole matrix to display. Old one was 22*44 pixels. again still trying to learn this thing. old dog new tricks.

truckerreid, 2023-06-23:
ok way to new at this. 

Basically vixen lights and nutcracker are programs that communicate with these boards to control different rgb and lights and set times based on a sequence that you put in. this is done with the com ports, COM6 for example. 
you can see a example of one i did here. [https://studio.youtube.com/video/yf6MAWIuTT4/edit](url)

So the grid where the face comes up to talk and the words and sayings that come up at the appropriate time need to be displayed in the DMD matrix. 

lightbulb. that is where my problem is i have been searching for wrong code. if you could help me with this i would be very helpful as i do not have time to do a ton of research on this.

board707, 2023-06-23:
> if you could help me with this i would be very helpful as i do not have time to do a ton of research on this.

I probably can't help because you don't even have time to explain what exactly doesn't work for you. Good luck.

truckerreid, 2023-06-23:
guess not explaining it right. but thank you for your time.

truckerreid, 2023-06-24:
trying to get it to use a program to send it information via usb cable and display the information that on the grid.

board707, 2023-06-25:
What kind of help are you looking for?
Do you expect someone to assemble the same screen as you and test your code?
In order for someone to help you, you must clearly explain what exactly you are trying to achieve and what does not work for you. Until you do this, you have zero chances.
Why didn't you answer the code questions? Maybe it's not your code and you don't understand it?

truckerreid, 2023-06-25:
I am still newer to this and have not studied every thing enough. It is a compilation of few codes to make it work I thought.  Can't find what I am looking for to learn this aspect. Think it archive the incoming bytes so that it can display them on the panels becausethe bytes are over 64 usually. Every part of the dmd display except the serial read section is by others. Who I mentioned in the end of the code, have not figured out how to dm them, to see if they know how to serial.read the code.

truckerreid, 2023-06-25:
The serial read is code that I was able to price together from one's I found for my 2812 grids.