mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #472: How to decode base64 bmp image string and draw

zhangtai, 2023-06-28:
I have created a simple http server with ESP32S3 board compiled with arduino-cli, now it can receive post request with text data and draw on the LED panel. I want to post encoded base64 string of bmp image to the server and draw accordingly.

Any advise on how to decode the string to PROGMEM(or other format can be draw by the library) and draw? Thanks in advance.

P.S. I already using the this approach to send image to a Divoom Pixoo64 LED board.

board707, 2023-06-28:
> Any advise on how to decode the string to PROGMEM or other format 

PROGMEM is not a data format, it is a memory location specificator  , indicating (mostly for AVR chips) that the data is located in the area of memory allocated for the program. It has nothing to do with the format of the data.

As for the question - in any library for base64 there are examples of both encoding and decoding data

zhangtai, 2023-06-28:
Thanks @board707 I have checked some libs and they provided very easy examples to how to decode base64 to normal string. But my encoded data is not string, and I wanted to decode to something can be draw with the function, since the example provided by this lib is using PROGMEM, that's why I wanted convert to it.

Maybe I need to make my requirement more simple as: Decode base64 string and draw it.

board707, 2023-06-28:
> since the example provided by this lib is using PROGMEM, that's why I wanted convert to it.

The BitmapIcons example is used a char array for storing image data, it is the same as that you called "normal string". If you have a library, that can decode base64 to string - you can use it examples for your code.

Don't pay attention to the "PROGMEM", you don't need it

zhangtai, 2023-06-28:
Oh got it, thanks for your help. Will try and update accordingly.

closed 2023-07-04 by zhangtai