mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #473: 3.0.8 -> 3.0.9 breaks ICN2037BP

sschueller, 2023-07-02:
This is more of an FYI for someone else having issues. If you see the following issue set your clkpase to false ```mxConfig.clkphase = false;``` 

true:

[image]

false:

[image]

closed 2023-07-02 by sschueller

mrfaptastic, 2023-07-05:
Yeah there was a bug in versions < 3.0.9 in that the clkphase value was always false and completely ignored.

elan, 2023-08-12:
OMG, thank you so much, I was going crazy trying to figure out what broke.