mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #477: use extended gpio by PCF8574 to control LED Matrix

Dubu0312, 2023-07-12:
hi, i want to thank you for making this project so useful and easy to use, i have been using esp32 to control 4 panels chained LED and it's working perfectly fine. But now i want to use wt32-eth01 instead of esp32 (still use esp32 in core but include ethernet) and it only have 10 GPIO available to use. I use PCF8574 to extend the number of GPIO but it seems to define R1_DEFAULT_PIN, the value must be an integer and accessible via the board's pin. Can you help me please? Thank you so much
[image]

board707, 2023-07-12:
> I use PCF8574 to extend the number of GPIO

It is absolutely impossible to run this library via PCF8574 extender.

Dubu0312, 2023-07-31:
since my problem is lacking of GPIO to connect to HUB75, my solution is buy another WT32-ETH01 to "share" the GPIO by rewrite the esp32-default-pins.hpp like this, and uploaded to each microcontroller. 

![image](https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-DMA/assets/136299162/

But the result is not as i expected, here's the result.

https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-DMA/assets/136299162/044923c3-9bdb-469f-85a7-5a2538ac248e

Still the text i want to display is in there, but moving constantly and fast, i think the way i edit the esp32-default-pins.hpp file is completely wrong. Can you help me please, thank you so much

Dubu0312, 2023-07-31:
since my problem is lacking of GPIO to connect to HUB75, my solution is buy another WT32-ETH01 to "share" the GPIO by rewrite the esp32-default-pins.hpp like this, and uploaded to each microcontroller.

[image]

But the result is not as i expected, here's the result.

https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-DMA/assets/136299162/5bc7695e-7ad7-47e9-9351-a00cb3f1f14b

Still the text i want to display is in there, but moving constantly and fast, i think the way i edit the esp32-default-pins.hpp file is completely wrong. Can you help me please, thank you so much

rigorka, 2023-07-31:
Hi! If you definitely need to control your panels via the Ethernet connection, you might want to get one eth01 module and use it as Ethernet-to-ttl gateway, and one Huidu HD-WF2 (or any other suitable ESP32 module) to drive the panels using the ttl data received from eth01. This way you wouldn't need to do any rewrites to this lib, you'll just need to implement the eth-2-ttl gateway, which is really straightforward.

board707, 2023-07-31:
> Can you help me please, thank you so much

@Dubu0312 
 I still think that there is no way to run this library by distributing pins across multiplexers or different boards.

mrfaptastic, 2023-07-31:
You can't share pins or use a multiplexer or anything as these panels need a clear 10-20Mhz signal and data. Otherwise you're going to get noise or other garbage.

The only option you have if you only have 10 pins I think is to run three wires from the R1,B1 and G1 output pins from the LAST panel in your chain - and connect to the R2, B2 and G2 pins of the FIRST panel in the chain.

Then you'll need to do some pixel mapping logic as now you've essentially made a panel of twice the length, but half the height.

closed 2024-01-09 by mrfaptastic