mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #480: There are no d and e pins. panel works differently.

PTCesur, 2023-07-20:
There are no d and e pins. panel works differently. https://imgur.com/0rL6SeR

board707, 2023-07-20:
Please insert clear photo panel's rear side .
There should be at least two types chips on the panel, please copy its partnumbers to the forum.

PTCesur, 2023-07-23:
[image]
[image]

PTCesur, 2023-07-23:
Below is the panel I tried(The 32x64 panel is the one I shared with you. ). When it didn't work, I tried the other panel, but it didn't work either.

board707, 2023-07-23:
It is normal for these panels that they have A B C pins only.The both panels are 8 scan, so they doesnt requires D and E pins for multiplexing.
Please find a chips on the panel and copy it partnumbers  to the forum.

PTCesur, 2023-07-30:
[image]

PTCesur, 2023-07-30:
[image]

PTCesur, 2023-07-30:
https://ibb.co/3yKdk8w%0A
https://ibb.co/ydtCNF3

board707, 2023-07-31:
Hi
These drivers are the most common types. The panel should work with the current library. The only problem is that a some pixel mapping could required.

PTCesur, 2023-08-01:
how can I do that?

board707, 2023-08-02:
Honesty, I don't know how to remapping pixels in this library. You should ask the author.