mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #481: how to use this library with ADAFruitGFX

kejuqeju, 2023-07-21:
Hello...
i have successfully used this library on my 16x32 RGB panel. Now i want to use Fonts from ADAFruitsGFX library.
Can anyone help me with this?

kejuqeju, 2023-07-21:
how to change font type? should i include ADAFruitGFX library? can you share me some simple code how to do it? thank you in advance

GrimorJander, 2023-07-23:
It took me several days to figure out.

First, install Adafruit GFX font library on Arduino IDE.
Make sure there´s a "Font" folder with those "fonts.h" files inside.
Then, at the begining of the skecht, call for library and each font you want to use with "#include":

#include <GFX.h>
#include <gfxfont.h>

#include <Arduino.h>

#include "Fonts/FreeSansBold9pt7b.h"
#include "Fonts/FreeSansBold12pt7b.h"
#include "Fonts/FreeSansBold18pt7b.h"
#include "Fonts/Picopixel.h"
#include "Fonts/FreeSerif9pt7b.h"
#include "Fonts/FreeSerif12pt7b.h"
#include "Fonts/FreeSerif18pt7b.h"

#include <ESP32-HUB75-MatrixPanel-I2S-DMA.h>

Then define your panel setting, and your colors

const int panelResX = 64;      // Number of pixels wide of each INDIVIDUAL panel module.
const int panelResY = 32;     // Number of pixels tall of each INDIVIDUAL panel module.
const int panel_chain = 1;      // Total number of panels chained one to another

uint16_t myBLACK = dma_display->color565(0, 0, 0);
uint16_t myWHITE = dma_display->color565(255, 255, 255);
uint16_t myRED = dma_display->color565(255, 0, 0);
uint16_t myGREEN = dma_display->color565(0, 255, 0);
uint16_t myBLUE = dma_display->color565(0, 0, 255);
uint16_t myPINK = dma_display->color565(227, 28, 121);
uint16_t myYELLOW = dma_display->color565(255, 255, 0);
uint16_t myPURPLE = dma_display->color565(199, 36, 177);

void displaySetup() {
  HUB75_I2S_CFG mxconfig(
    panelResX,   // module width
    panelResY,   // module height
    panel_chain    // Chain length
  );

On void Setup call for the setFont name with "&" symbol first:

 displaySetup();

  dma_display->fillScreen(myBLACK);

  dma_display->setFont(&FreeSansBold12pt7b);

  dma_display->setTextSize(1);     // size 1 == 8 pixels high
  dma_display->setTextWrap(false); // N.B!! Don't wrap at end of line
  dma_display->setTextColor(myRED); 

Then just fill screen black color, set cursor where you want and write text you want:

 dma_display->fillScreen(myBLACK);
dma_display->setCursor(textXPosition, textYPosition);
dma_display->print(text);

kejuqeju, 2023-07-26:
hi Grimmo, thank you for your answer. I tried to do
  dma_display->setFont(&FreeSansBold12pt7b);

but the result is not as expected
the first pic is after i do setfont
the second pic is if i dont do setfont

[image]
[image]

GrimorJander, 2023-07-27:
Well, your panel seens to be 16 Led tall, so try a short font like picopixel and set the cursor properly. If I get some free time this weekend I'll send you I simple skecht with different fonts

beta-tester, 2023-08-02:
> do you experience same problem?

i experience a similar issue as soon i set a custom font other that the default one.
for some reason the fonts text coordinate are offset or aligned in a different way.
so (0,0) in the default font is top left of the text.
when using a custom font, then (0,0) is at the bottom left of the text.
maybe this is the same thing at your issue.

board707, 2023-08-02:
There is a detail article about fonts on Adafruit site.

kejuqeju, 2023-08-03:
> > do you experience same problem?
> 
> i experience a similar issue as soon i set a custom font other that the default one. for some reason the fonts text coordinate are offset or aligned in a different way. so `setCursor(0,0)` in the default font is top left of the text. when using a custom font, then `setCursor(0,0)` is at the bottom left of the text (of first text row in case of multiline text). maybe this is the same thing at your issue.

i finally solved this problem. In order to use custom fonts i create a new display buffer using VirtualMatrixPanel which linked to MatrixPanel_I2S_DMA.
i tried with 9 daisy chained p10 LED module and 9 daisy chained p5 LED module, both were success.

//MatrixPanel_I2S_DMA dma_display;
MatrixPanel_I2S_DMA *dma_display = nullptr;
VirtualMatrixPanel  *virtualDisp = nullptr;

// Module configuration
HUB75_I2S_CFG mxconfig
 (
 PANEL_RES_X,   // module width
 PANEL_RES_Y,   // module height
 PANEL_CHAIN    // Chain length
 );

mxconfig.clkphase = false;
dma_display->begin();
dma_display->setBrightness8(BRIGHTNESS_M); //0-255
dma_display->setLatBlanking(1);

virtualDisp = new VirtualMatrixPanel((*dma_display), NUM_ROWS, NUM_COLS, PANEL_RES_X, PANEL_RES_Y, CHAIN_NONE);

but i dont know if i use VirtualMatrixPanel, will it cost extra RAM?

mrfaptastic, 2023-08-05:
No, virtualmatrixpanel just does pixel remapping on every call to drawPixel