mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #483: selective brightness for different areas

jdeltoft, 2023-07-22:
Is there a way to change brightness for a specific area?  

I have 3 panels chained together, and would like to change each 64x64 panel brightness independently

board707, 2023-07-22:
> Is there a way to change brightness for a specific area?

You can do it by changing the color brightness of the individual picture pixels.

jdeltoft, 2023-07-24:
thanks!!! Yea, I thought I tried this but also I used say (225,0,0) vs (5,0,0) and now I'm using (2,0,0) for off.  This is non linear but just thought it wouldn't be that far from linear.  I also set the global brightness to max 255, not sure how much that plays into it but happy now.

[image]

closed 2023-07-24 by jdeltoft