mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #484: how to prevent vertical ghosting/bleeding

beta-tester, 2023-08-02:
hi, i am driving two 32x32 panels in a chain directly by my ESP32 dev board (without 3.3V -> 5V buffers).

i noticed clearly vertical ghosting or is it bleeding. what can be the cause of that ghosting?
[image]

this is a view of very dark pixel colors for at night.
it is most obvious in the dark.
on the photo it is even more visible than with the naked eye.

here the schematic and pcb i use:
[image]
[image]

the pcb is plugged directly to the first panel.
nothing else is connected yet to the ESP32.
pin E of HUB75 is not in use it is GND on the panel.

here the initialization settings:
```C
/*--------------------- MATRIX GPIO CONFIG  -------------------------*/
// my JOY-iT NodeMCU ESP32 ->
#define R1_PIN   32
#define B1_PIN   33
#define R2_PIN   25
#define B2_PIN   26
#define A_PIN    27
#define C_PIN    14
#define CLK_PIN  12
#define OE_PIN   13

#define G1_PIN   19
// GND
#define G2_PIN    5
#define E_PIN    -1 //GND //17
#define B_PIN    16
#define D_PIN     4
#define LAT_PIN  15
// GND

...

  // Module configuration
  HUB75_I2S_CFG mxconfig(
    /*mx_width*/                32,
    /*mx_height*/               32,
    /*chain_length*/            2,
    /*gpio*/                    {
                                  R1_PIN, G1_PIN, B1_PIN, R2_PIN, G2_PIN, B2_PIN,
                                  A_PIN, B_PIN, C_PIN, D_PIN, E_PIN,
                                  LAT_PIN, OE_PIN, CLK_PIN
                                },
    /*driver*/                  HUB75_I2S_CFG::SHIFTREG, //SM5266P, definitiv wrong //MBI5124, artifacts like latch_blanking //ICN2038S, FM6126A, FM6124, SHIFTREG, working, no effect
    /*double_buff*/             true, //false,
    /*i2sspeed*/                HUB75_I2S_CFG::HZ_8M, //HZ_20M .. HZ_8M, no effect
    /*latch_blanking*/          4, //1 & 2, may cause artifacts (horizontal pixle bleeding at red pattern 0010100 shows as 0011100) after OTA reboot
    /*clkphase*/                true, //false, no effect
    /*min_refresh_rate*/        100, //120 .. 60, no effect
    /*pixel_color_depth_bits*/  12 //12 even better, specially dark colors with low panel brightness //8 good colors //6 missing dark pixels
  );

  // Display Setup
  display = new MatrixPanel_I2S_DMA(mxconfig);
  display->begin();
  display->setBrightness(95); //255); //127);
...
```

i tried many different settings (see the comment behind the settings), but none of them have an effect to that ghosting/bleeding.
did i used an unfortunate pin arrangement of my ESP32 board? i used those pins to minimize the pcb complexity and be easily able to build it on a _strip grid_ pcb.
do i need 3.3V->5V buffers? i think i don't need, because the panel itself has 74HCT245 Octal Bus Transceivers and i don't use long wires.
GND is also connected properly to each others GND.

i am out of ideas.

mrfaptastic, 2023-08-05:
Common causes:

*Dial back brightness a tiny bit
*Clkphase - invert value
*Grounding! So important this is, need to ensure the hub75 gnd pins are also connected to esp32s ground and the power supply ground.

mrfaptastic, 2023-08-05:
Common causes:

*Dial back brightness a tiny bit
*Clkphase - invert value
*Grounding! So important this is, need to ensure the hub75 gnd pins are also connected to esp32s ground and the power supply ground. (Edit, looks like you have done this though)
*Latch blanking =4 
*Reduce clkspeed 

I have never used this library with a subpixel colour depth > 8, so not sure if that's bringing out some weird bug.

beta-tester, 2023-08-05:
unfortunately all that i tried already without solving the issue i have.

> *Dial back brightness a tiny bit

i used the brightness initially to dim the panel at night to a lower value to not blind me when trying to sleep.
specially at low brightness and dark colors bleeding is very visible.
at high panel brightness and bright colors it is more or less not noticeable, because the eye is blinded 8-)

> *Clkphase - invert value

has no effect. the panel works at both the same. (only at 8MHz clk speed i see the bahavior like at low blanking. see below. but no effect to the bleeding issue)

> *Grounding! So important this is, need to ensure the hub75 gnd pins are also connected to esp32s ground and the power supply ground. (Edit, looks like you have done this though)

the ESP is connected to +5V & GND of the same power supply ar the both panels.
and the ESP is also connected to the GND at the HUB75 connector.
+5, GND at the power supply is in "star" topology. a pair of wires to the ESP, one pair to LED panel1 and one pair to LED panel2. all have the same wire length.
the only loop is GND power supply -> GND ESP -> GND HUB75 -> GND LED Panel -> GND power supply.

> *Latch blanking =4

i use 4, because lower values may show some strange pattern artifacts after a reboot. not always but when it happens, the pattern artifacts stays all the time.

> *Reduce clkspeed

i tried the full ranges from 8MHz to 20MHz. at lowesr speed i could see a behaviot like i using the a too low latch blanking when toggling clk phase. 

> I have never used this library with a colour depth > 8, so not sure if that's bringing out some weird bug.

i tried 12 bit only to see if it makes a different.
and later on i tried to be able to dim the panel in a wide range in controlling the brightness of the colors while drawing but that is another story.

the strange thing is that the bleeding is vertical. i expect horizontal bleeding artifacts in case of wrong or unsharp timing signals.
but vertical bleeding i can not explain where is might coming from. i guess it can only come from the LED panel itself like cross talking.

the LED panles are quite old. some of the very early ones from adafruit.

mrfaptastic, 2023-08-06:
Can you try an older version of the library? 2.0.7 and see if there is the same issue?

Also, do you get bleeding with PxMatrix library?

beta-tester, 2023-08-07:
> Can you try an older version of the library? 2.0.7 and see if there is the same issue?

i comment out the `pixel_color_depth_bits` and `setBrightness();` for 2.0.7 but it still does not compile anymore:
```C
Arduino: 1.8.19 (Linux), Board: "ESP32 Dev Module, Disabled, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, Core 1, Core 1, None, Disabled"

/home/pi/Arduino/libraries/ESP32_HUB75_LED_MATRIX_PANEL_DMA_Display/ESP32-HUB75-MatrixPanel-I2S-DMA.cpp:10:21: note: #pragma message: Compiling for original (released 2016) 520kB SRAM ESP32.
     #pragma message "Compiling for original (released 2016) 520kB SRAM ESP32."
                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
sketch/ESP32_I2S_RGB_Matrix_Clock.ino.cpp.o: in function `MatrixPanel_I2S_DMA::flipDMABuffer()':
Mehrere Bibliotheken wurden für "WiFi.h" gefunden
 Benutzt: /home/pi/.arduino15/packages/esp32/hardware/esp32/2.0.11/libraries/WiFi
 Nicht benutzt: /home/pi/bin/arduino-1.8.19/libraries/WiFi
/home/pi/Arduino/libraries/ESP32_HUB75_LED_MATRIX_PANEL_DMA_Display/ESP32-HUB75-MatrixPanel-I2S-DMA.h:528:(.iram1.29[MatrixPanel_I2S_DMA::flipDMABuffer()]+0x8): dangerous relocation: l32r: literal placed after use: .literal._ZN19MatrixPanel_I2S_DMA13flipDMABufferEv
/home/pi/Arduino/libraries/ESP32_HUB75_LED_MATRIX_PANEL_DMA_Display/ESP32-HUB75-MatrixPanel-I2S-DMA.h:530:(.iram1.29[MatrixPanel_I2S_DMA::flipDMABuffer()]+0xe): dangerous relocation: l32r: literal placed after use: (.literal._ZN19MatrixPanel_I2S_DMA13flipDMABufferEv+0x4)
/home/pi/Arduino/libraries/ESP32_HUB75_LED_MATRIX_PANEL_DMA_Display/ESP32-HUB75-MatrixPanel-I2S-DMA.h:532:(.iram1.29[MatrixPanel_I2S_DMA::flipDMABuffer()]+0x1c): dangerous relocation: l32r: literal placed after use: (.literal._ZN19MatrixPanel_I2S_DMA13flipDMABufferEv+0x8)
/home/pi/Arduino/libraries/ESP32_HUB75_LED_MATRIX_PANEL_DMA_Display/ESP32-HUB75-MatrixPanel-I2S-DMA.h:537:(.iram1.29[MatrixPanel_I2S_DMA::flipDMABuffer()]+0x2e): dangerous relocation: l32r: literal placed after use: (.literal._ZN19MatrixPanel_I2S_DMA13flipDMABufferEv+0xc)
/home/pi/Arduino/libraries/ESP32_HUB75_LED_MATRIX_PANEL_DMA_Display/ESP32-HUB75-MatrixPanel-I2S-DMA.h:539:(.iram1.29[MatrixPanel_I2S_DMA::flipDMABuffer()]+0x34): dangerous relocation: l32r: literal placed after use: (.literal._ZN19MatrixPanel_I2S_DMA13flipDMABufferEv+0x10)
collect2: error: ld returned 1 exit status
exit status 1
Fehler beim Kompilieren für das Board ESP32 Dev Module.
```

mrfaptastic, 2023-08-07:
You won't be able to use your existing sketch when testing with this version as there are too many compatibility breaking changes.

But run one of the examples like simple test shapes

beta-tester, 2023-08-08:
> does not compile anymore

i just also comment out `flipDMABuffer()` and disabled `double_buff`. now is is compiling.

there is still bleeding.

my HD-WF1 it is now at import customs clearance and will arrive soon (with some new 64x64 LED panels).
in case that controller is based on an ESP32-S, i will try out the stock firmware and see if this will show bleeding on the old LED panels.
if the controller is an ESP32-S based one, then i try to use your library and see what happens.

to see if it is my old LED panel, my ESP32 dev board, the used library, ...
i will report how it went...

i wish i owned a 16bit logic analyzer to see what is going on and compare signals from stock firmware and non-stock firmware.

rwahyur, 2023-08-16:
How to make dual hub75..
I have made, but horizontal ghosting

board707, 2023-08-16:
> How to make dual hub75..

please do not hijack others threads, make your own one

beta-tester, 2023-10-23:
> my HD-WF1 it is now at import customs clearance and will arrive soon (with some new 64x64 LED panels). in case that controller is based on an ESP32-S, i will try out the stock firmware and see if [...]

sorry for my very late reply.

i used the original stock HD-WF1 firmware and tried several settings with the old 32x32 panels. with that original stock firmware and hardware the old panels showed always the ghosting/bleeding.

the same firmware and hardware in combination of  new 64x64 HUB75 panels there is no such ghosting/bleeding visible at all.

so your ESP32-HUB75-MatrixPanel-DMA can not be the issue.
also the cabeling, powersupply, controller can not be the cause of the issue...

i will close this issue.

closed 2023-10-23 by beta-tester