mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #485: 128X64 MatrixPanel Problem

dowandev, 2023-08-04:
Hello, when I apply the code attached below, some spaces are displayed as blank. How should the code fix or the wiring?

`// 1) Include key virtual display library
// VERSION 3.0.7

#include <ESP32-VirtualMatrixPanel-I2S-DMA.h>
#include <Fonts/FreeMono9pt7b.h>

// 2) Set configuration
#define PANEL_RES_X 128 // Number of pixels wide of each INDIVIDUAL panel module.
#define PANEL_RES_Y 64 // Number of pixels tall of each INDIVIDUAL panel module.

#define NUM_ROWS 1                     // Number of rows of chained INDIVIDUAL PANELS
#define NUM_COLS 1                     // Number of INDIVIDUAL PANELS per ROW
#define PANEL_CHAIN NUM_ROWS *NUM_COLS // total number of panels chained one to another

/* Configure the serpetine chaining approach. Options are:
CHAIN_NONE,
CHAIN_TOP_LEFT_DOWN,
CHAIN_TOP_RIGHT_DOWN,
CHAIN_BOTTOM_LEFT_UP,
CHAIN_BOTTOM_RIGHT_UP,
CHAIN_TOP_LEFT_DOWN_ZZ, /// ZigZag chaining. Might need a big ass cable to do this, all panels right way up.
CHAIN_TOP_RIGHT_DOWN_ZZ,
CHAIN_BOTTOM_RIGHT_UP_ZZ,
CHAIN_BOTTOM_LEFT_UP_ZZ

The location (i.e. 'TOP_LEFT', 'BOTTOM_RIGHT') etc. refers to the starting point where
the ESP32 is located, and how the chain of panels will 'roll out' from there.

In this example we're using 'CHAIN_BOTTOM_LEFT_UP' which would look like this in the real world:

Chain of 4 x 64x32 panels with the ESP at the CHAIN_TOP_LEFT_DOWN:

+---------+---------+
|    4    |    3    |
|         |         |
+---------+---------+
|    1    |    2    |
|  (ESP)  |         |
+---------+---------+

*/

#define VIRTUAL_MATRIX_CHAIN_TYPE CHAIN_TOP_RIGHT_DOWN

// 3) Create the runtime objects to use

// placeholder for the matrix object
MatrixPanel_I2S_DMA *dma_display = nullptr;

// placeholder for the virtual display object
VirtualMatrixPanel *virtualDisp = nullptr;

/******************************************************************************
 * Setup!
 ******************************************************************************/
void setup()
{

    delay(2000);
    Serial.begin(115200);

    /******************************************************************************
     * Create physical DMA panel class AND virtual (chained) display class.
     ******************************************************************************/

    HUB75_I2S_CFG mxconfig(
        PANEL_RES_X, // module width
        PANEL_RES_Y, // module height
        PANEL_CHAIN  // chain length
    );

    // mxconfig.driver = HUB75_I2S_CFG::FM6126A;     // in case that we use panels based on FM6126A chip, we can set it here before creating MatrixPanel_I2S_DMA object

    // Sanity checks
    if (NUM_ROWS <= 1){
        Serial.println(F("There is no reason to use the VirtualDisplay class for a single horizontal chain and row!"));
    }

    // OK, now we can create our matrix object
    dma_display = new MatrixPanel_I2S_DMA(mxconfig);

    // dma_display->setRotation(1);

    // let's adjust default brightness to about 75%
    dma_display->setBrightness8(192); // range is 0-255, 0 - 0%, 255 - 100%

    // Allocate memory and start DMA display
    if (not dma_display->begin()) Serial.println("****** !KABOOM! I2S memory allocation failed ***********");

    // create VirtualDisplay object based on our newly created dma_display object
    virtualDisp = new VirtualMatrixPanel((*dma_display), NUM_ROWS, NUM_COLS, PANEL_RES_X, PANEL_RES_Y, VIRTUAL_MATRIX_CHAIN_TYPE);

    // virtualDisp->setPhysicalPanelScanRate(NORMAL_ONE_SIXTEEN);
    // virtualDisp->setPhysicalPanelScanRate(FOUR_SCAN_16PX_HIGH);

    // So far so good, so continue
    virtualDisp->fillScreen(virtualDisp->color444(0, 0, 0));
    // virtualDisp->drawDisplayTest(); // draw text numbering on each screen to check connectivity

    // delay(1000);

    virtualDisp->setTextColor(virtualDisp->color565(0, 0, 255));

    // Red text inside red rect (2 pix in from edge)
    virtualDisp->drawRect(0, 0, virtualDisp->width(), virtualDisp->height(), virtualDisp->color565(255, 0, 0));

    // White line from top left to bottom right
    // virtualDisp->drawLine(0, 0, virtualDisp->width() - 1, virtualDisp->height() - 1, virtualDisp->color565(255, 255, 255));

    // virtualDisp->drawDisplayTest(); // re draw text numbering on each screen to check connectivity

    virtualDisp->setTextWrap(false);

    virtualDisp->setCursor(1, 1);
    virtualDisp->print("ABCDEFGHIJ");
    virtualDisp->setCursor(1, 8);
    virtualDisp->print("1234567890");
    virtualDisp->setCursor(1, 16);
    virtualDisp->print("ABCDEFGHIJ");
    virtualDisp->setCursor(1, 24);
    virtualDisp->print("1234567890");
    virtualDisp->setCursor(1, 32);
    virtualDisp->print("ABCDEFGHIJ");
    virtualDisp->setCursor(1, 40);
    virtualDisp->print("1234567890");
    virtualDisp->setCursor(1, 48);
    virtualDisp->print("ABCDEFGHIJ");
    
}

void loop(){

}`

dowandev, 2023-08-06:
@mrfaptastic 
Sorry, Can you help me?

board707, 2023-08-06:
Single geometry of the matrix like "128x64" is not enough to manage it to work/ we need a more detail. Is it a scan32 matrix? Or scan 16 or scan8?
What  the chips are used as drivers?

dowandev, 2023-08-06:
@board707 
It is 1/32 scan and uses ICND2038S chip

mrfaptastic, 2023-08-06:
Can you please try with the simpletestshapes sketch?

dowandev, 2023-08-06:
@mrfaptastic 
Apply the code below to make it look like the picture.
I don't know if this is displayed well, but it is displayed normally.
But in BItMapIcons //mxconfig.gpio.e = -1; When I uncomment // Assign a pin if you have a 64x64 panel and apply, I get a Compilation error: 'mxconfig' does not name a type error.

mrfaptastic, 2023-08-06:
Looks like there's a bug with the VirtualMatrixPanel class when using it with a single panel (which there is no need to use this class for), which I'll need to investigate.

Now using the underlying library to draw directly to the panel - looks like it works for me. I can't understand your gpio.e compile error comment unless you provide the actualy sketch code.