mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #488: SM16106SC support (for CampZone 2019 Badge/PixelBadge)

agners, 2023-08-10:
The CampZone 2019 Badge/PixelBadge comes with 8x32 LED matrix. The chip used is the SM16106SC.

From what I can see from the original firmware the R0/G0/B0 as well as A/B/C GPIOs are provided, but no D pin for instance.

I do have a PixelBadge here, and would like to get this library going. From what I understand the CampZone 2019 badge is quite similar.

Do you know if the SM16106SC chip can be supported? What would need to be done to get this library going?

Curious Supply PixelBadge: https://pixel.curious.supplies/
CampZone 2019 Badge: https://hackaday.com/2019/07/25/campzone-2019-badge-is-begging-to-become-a-huge-billboard/

board707, 2023-08-11:
According to datasheet  SM16106 a standard driver, it should be supported.
 https://www.waveshare.com/w/upload/3/30/SM16106SC.pdf

agners, 2023-08-11:
@board707 oh ok, cool. Thoughts wrt missing D-pin? Can I just set that to -1, will the library handle that?

board707, 2023-08-11:
There is no needs of D pin, because the matrix has only 8 rows

agners, 2023-08-11:
FWIW, this is the HUB75 configuration of the original firmware: https://github.com/badgeteam/ESP32-platform-firmware/blob/master/firmware/configs/pixel_defconfig#L204-L221

agners, 2023-08-11:
@board707 I think the other thing I am a bit worried is the "regulate VLED" logic the original firmware does. Not sure about its implications exactly :sweat_smile:  https://github.com/badgeteam/ESP32-platform-firmware/blob/ce4dd3ac64a9742ea348f65ce075b5fe7ae041f4/firmware/components/driver_display_hub75/driver_hub75.c#L181-L223

board707, 2023-08-11:
What are you going to do with this matrix?
As far as I can see, it can be run on this library as half of a 32x16 scan8 panel. But the code will work with only one matrix, chaining will not work

closed 2024-01-09 by mrfaptastic