mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #491: Ask

rwahyur, 2023-08-15:
Hello Mr Faptastic.. how to make dual hub75 without ghosting ? Thanks

mrfaptastic, 2023-08-18:
I don't understand the question

board707, 2023-08-24:
You will need an additional 6 data pins for RGB channels of the second HUB75 instance. All other HUB75 pins will be common for both rows. 
I am not sure, is it possible to add another six pins to ESP32 I2s parallel interface. if there are not enough free I2s lines, you can transfer work with A B C D E pins to an external multiplexer, as is done in the smartmatrix library

rwahyur, 2023-08-25:
I have changed the library.

The first and second Hub pin are common accept Lat pin, but the display is ghosting and other panel normal.

I will send the code

tsctrl, 2023-10-14:
hi @mrfaptastic, possible to revisit this issue? we could have the panels as separate screen by using 2 hubs if i sane

@rwahyur can share the changes done?