mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #493: 32x16 full color 1/4 scan, is this library available?

Xeyeong, 2023-08-21:
If 1/4 scan is not possible, can 1/8 be used?

board707, 2023-08-21:
> Is there a scan problem?

Most likely so. Outputting the text doesn't depend on type of matrix.
But you will more chances to answer if you show your code and pin assignments.

Xeyeong, 2023-08-21:
> > Is there a scan problem?
> 
> Most likely so. Outputting the text doesn't depend on type of matrix. But you will more chances to answer if you show your code and pin assignments.

#include <Adafruit_GFX.h>
#include <ESP32-VirtualMatrixPanel-I2S-DMA.h >
#include <Fonts/GodoM6pt8b.h>

#define PANEL_RES_X 32                  //가로
#define PANEL_RES_Y 16                  //세로
#define NUM_ROWS 1                      // 열
#define NUM_COLS 1                      // 행
#define PANEL_CHAIN NUM_ROWS *NUM_COLS  //열 * 행
#define PIN_E 32

MatrixPanel_I2S_DMA *dma_display = nullptr;
VirtualMatrixPanel *virtualDisp = nullptr;

#define R1 25
#define G1 26
#define BL1 27
#define R2 14
#define G2 12
#define BL2 13
#define CH_A 22
#define CH_B 21
#define CH_C 19
#define CH_D 18
#define CH_E -1  // assign to any available pin if using two panels or 64x64 panels with 1/32 scan
#define CLK 5
#define LAT 4
#define OE 32

void setup() {
  Serial.begin(115200);
  HUB75_I2S_CFG::i2s_pins _pins = { R1, G1, BL1, R2, G2, BL2, CH_A, CH_B, CH_C, CH_D, CH_E, LAT, OE, CLK };

  HUB75_I2S_CFG mxconfig(  //HUB75_I2S_CFG의 객체, 원하는 패널 해상도 및 체인 구성으로 HUB75 I2S 디스플레이 드라이버를 초기화하거나 설정하는 데 사용
    PANEL_RES_X,
    PANEL_RES_Y,
    PANEL_CHAIN,
    _pins);

  dma_display = new MatrixPanel_I2S_DMA(mxconfig);
  //mxconfig.gpio.e = PIN_E;
  // matrix와 폰트 설정
  dma_display->setBrightness8(70);  //밝기
  dma_display->begin();
  virtualDisp = new VirtualMatrixPanel((*dma_display), NUM_ROWS, NUM_COLS, PANEL_RES_X, PANEL_RES_Y);  //가상 매트릭스 패널
  virtualDisp->setFont(&GodoM6pt8b);                                                                   //폰트
  virtualDisp->setTextSize(1);                                                                         //폰트크기
  virtualDisp->setAttribute(UTF8_ENABLE, true);                                                        //utf8 인코딩
  virtualDisp->clearScreen();
}

void loop() {
  virtualDisp->setCursor(0, 0);  //위치할 커서
  virtualDisp->setTextColor(virtualDisp->color565(100, 32, 255));
  virtualDisp->print("가나다");
  delay(5000);
  virtualDisp->clearScreen();
}
This is the code I used
The 64x32 matrix outputs well without problems.
32x16 outputs as if it were broken. What is the problem?

xuniuer, 2023-09-12:
Just have a look an examples routine - `Four_Scan_Panel` for more details.