mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #500: P5 RVB 64x32 1/8 scan with  ICN2038S controller... and mess!

veryslow, 2023-09-13:
Hi, 
I'm fabrice and trying to deal with panels that I received from eagerled.
They are P5 RVB 64x32 1/8 scan with  ICN2038S controller, and seem to act as double 64x16 panels...

I did not notice that they were 1/8 scan and I've been digging for 2 weeks with an ESP32 WROOM trying every combinations in the library, changing pins, adjusting variables without any success yet.
The panels are functionnal, I have a HD-D16 asynchronous controller from HUIDU that works well and gives me technical datas on the panel structure (see screen captures below). Nevertheless, I cannot manage to show any line without messy blocks...

I've been following  this feed, with no success...
I'm an amateur programmer with some skills but the library is a bit to complex for me to manage and modify it for my needs.

I also wonder if my pin configuration is ok, considering some pin limits in the ESP32 (I tried different ones based on default pins configurations from library), the actuel one is :
#define A_PIN  23
  #define B_PIN  19
  #define C_PIN  5
  #define D_PIN  -1
  #define E_PIN  -1
  
  #define R1_PIN  25
  #define R2_PIN  14
  #define G1_PIN  26
  #define G2_PIN  12
  #define B1_PIN  27
  #define B2_PIN  13
  
  #define CLK_PIN 16
  #define LAT_PIN 4
  #define OE_PIN  15

here are pictures frome the huidu configuration software that may help and 4 pictures of the panel itself.
I plan to create A 6 panels linear display to make scrolling texts updated by wifi for a theatre company (begat.org).

Thanks by advance of any help, I've been almost pulling all my hair!

[image]
[image]
[image]
[image]
[image]
[image]

board707, 2023-09-13:
> They are P5 RVB 64x32 1/8 scan with ICN2038S controller,

Every panel contains at least two types of chips, the ICN2038 is only one from it. Knowing the second chip type is required to handle the matrix properly. Please try to read the headings in the other chips on the panel.

> here are pictures from the huidu configuration software that may help

Is the panel works with this config?

veryslow, 2023-09-13:
Hi, thanks for your answer.
Yes, the panel works well with this config, It comes with a config file from the manufacturer (here is the xml file also).
[P5-8S-64X32-SM16208-512X32-4KAI.ssx.zip](https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-DMA/files/12600873/P5-8S-64X32-SM16208-512X32-4KAI.ssx.zip)

and checking the chips, I see that the ICN2038 is in fact a SM16208S and the other chip is a MW245B.

thkx!

fabrice

board707, 2023-09-14:
> I see that the ICN2038 is in fact a SM16208S and the other chip is a MW245B.

No, the MW245B is not "the other chip" of the interest, it is just a buffer. Your panel should contains yet another driver.

But it is not so important already. According with screenshot of the huidu software configuration screen, the panel has a standard drivers, supported by the library.  All what you need - correct transform of the pixel coordinates.

mrfaptastic, 2023-09-14:
The best bet is to simply make a script that draws a pixel to y coordinate zero, x coordinate 0, 1, 2 etc, and then repeat the same for Y coordinate one, X coordinate 0, one, 2 etc and etc.

From this you might be able to reverse engineer the pixel mapping approach to the panel, there's so many permutations of these panels now I don't bother even coding for them anymore, as long as it works electrically, it's a waste of time.

veryslow, 2023-09-14:
Thank you very much for your answers, 

If I manage to understand the panel permutations, where in the code could I implement that specificity? 
I was thinking of make a correspondance map or algorithm but I don't even know where to insert it in the library...
COUld you give me some guidelines if it's not too tricky?

thank you, 
fabrice

board707, 2023-09-15:
> there's so many permutations of these panels now I don't bother even coding for them anymore, as long as it works electrically, it's a waste of time.

In most cases the mapping is not more than two-three lines of code. 
So far, I have been able to find transformations for all the matrices that came to me - as long as their drivers were compatible with the code. Although sometimes this process may be quite tricky :)

board707, 2023-09-15:
@veryslow 
Do as @mrfaptastic advises - run an example with filling the panel with one color pixel by pixel along horizontal lines and post a video of this process.  Perhaps I can advise you something

veryslow, 2023-09-16:
Thank you very much for your answers, 

I run this code (you know it!) with a pause 1000ms tu understand what happens with the pixels inrelation with the coordinates...
It seems that pixels are writing one on each others, adding their brightness...

`#include "ESP32-HUB75-MatrixPanel-I2S-DMA.h"

 // default pins: 
  #define A_PIN  23
  #define B_PIN  19
  #define C_PIN  5
  #define D_PIN  -1
  #define E_PIN  -1
  
  #define R1_PIN  25
  #define R2_PIN  14
  #define G1_PIN  26
  #define G2_PIN  12
  #define B1_PIN  27
  #define B2_PIN  13
  
  #define LAT_PIN 4
  #define OE_PIN  15
  #define CLK_PIN 16

  // Panel configuration
  #define PANEL_RES_X 64 // Number of pixels wide of each INDIVIDUAL panel module. 
  #define PANEL_RES_Y 32 // Number of pixels tall of each INDIVIDUAL panel module.
  
  #define NUM_ROWS 1// Number of rows of chained INDIVIDUAL PANELS
  #define NUM_COLS 1 // Number of INDIVIDUAL PANELS per ROW
  
  // ^^^ NOTE: DEFAULT EXAMPLE SETUP IS FOR A CHAIN OF TWO x 1/8 SCAN PANELS
    
  // Change this to your needs, for details on VirtualPanel pls read the PDF!
  #define SERPENT false
  #define TOPDOWN false
  
  // placeholder for the matrix object
  MatrixPanel_I2S_DMA *dma_display = nullptr;
  
  /******************************************************************************
   * Setup!
   ******************************************************************************/
  void setup()
  {
    delay(250);
   
    Serial.begin(115200);
    Serial.println(""); Serial.println(""); Serial.println("");
    Serial.println("*****************************************************");
    Serial.println("*         1/4 Scan Panel                            *");
    Serial.println("*****************************************************");
  
  HUB75_I2S_CFG::i2s_pins _pins =
  { R1_PIN, G1_PIN, B1_PIN, R2_PIN, G2_PIN, B2_PIN, A_PIN, B_PIN, C_PIN, D_PIN, E_PIN, LAT_PIN, OE_PIN, CLK_PIN };

    HUB75_I2S_CFG mxconfig(
                PANEL_RES_X,              // DO NOT CHANGE THIS
                PANEL_RES_Y,              // DO NOT CHANGE THIS
                NUM_ROWS*NUM_COLS           // DO NOT CHANGE THIS
                //,_pins            // Uncomment to enable custom pins
    );
    mxconfig.driver = HUB75_I2S_CFG::ICN2038S;  
    mxconfig.latch_blanking=1;
    mxconfig.clkphase = false; // Change this if you see pixels showing up shifted wrongly by one column the left or right.
    // OK, now we can create our matrix object
    mxconfig.i2sspeed =HUB75_I2S_CFG::HZ_20M;

    dma_display = new MatrixPanel_I2S_DMA(mxconfig);
  
    // let's adjust default brightness to about 75%
    dma_display->setBrightness8(10);    // range is 0-255, 0 - 0%, 255 - 100%
  
    // Allocate memory and start DMA display
    if( not dma_display->begin())
      Serial.println("****** !KABOOM! I2S memory allocation failed ***********");

    dma_display->clearScreen();
    delay(500);

   }

  void loop() 
  {
      uint16_t color = dma_display->color565(50, 50,50); //blue
      
      for (int i = 0; i < PANEL_RES_X; i++)
     // for (int i = 0; i <128; i=i+2)
      {
         for (int f = 0; f < PANEL_RES_Y; f++)
       //for (int f = 0; f < 64; f=f+1)
        {
            dma_display->drawPixel(i,f, color);
            Serial.print("drawPixel: ");
            Serial.print("(");
            Serial.print(i, DEC);
            Serial.print(",");
            Serial.print(f, DEC);
            Serial.println(")");
            delay(1000);
          }
      }
      delay(5000);
      dma_display->clearScreen();

  } // end loop`

https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-DMA/assets/2689912/a81ac8e5-2bf8-47d6-b568-56d6d888207d

veryslow, 2023-09-16:
and here it is at normal speed (with a 10ms pause between each iteration).

https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-DMA/assets/2689912/df2c3db2-af39-47d8-a821-e07c4903a0b9

board707, 2023-09-16:
You do it in incorrect order, the first (outer) loop should be by Y direction and the inner loop by X.

According to the video your problem is not (or not only) with geometry. You have a problem with multiplexing - either with your ABC connections or with multiplexor chip. I have to ask you again to find the ID of all the chips on the matrix.

(Excuse my English, I don't have a translator on the phone)

veryslow, 2023-09-16:
Hi,
after cutting, sanding and varnish dilution, I've found the hidden ICN2012!
ANd here is the video with the panel running the good x,y loop!

[image]

https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-DMA/assets/2689912/7bbae6ff-09db-47a6-ac14-d9e9344b1ac4

board707, 2023-09-16:
Thank you for uploading the new video. But the conclusion still the same - you have a multiplexing problem. Look to the picture - the panel is filled in by two lines with two lines gaps between it. It looks like the C pin is not worked or not connected.

veryslow, 2023-09-16:
Thank you! I did this mistake while soldering and soldering wires again and again!
Now I can fill the complete surface, that's great, but still lost with messy display.

If I want a complete horizontal line, I have to define a 128 wide panel forr 16 high...

https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-DMA/assets/2689912/b672d048-26bb-451c-8dc6-802e8e200d85

board707, 2023-09-16:
Perfect !
Now you are ready to tune a pixel mapping to your panel.

veryslow, 2023-09-16:
Yes! Can yout tell me where int the library I can tune this mapping? 
I did a new test ans realised that the panel seems considered as a 128x16 ...

With an adaptated code, I get a full-filled screen, with an organisation of pixels like in the sreen capture I posted before...
The thing is that I don't know where to program a mapping translation...

https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-DMA/assets/2689912/46e44f0f-9f49-445c-b615-c7efd558236a

https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-DMA/assets/2689912/1b50e653-c58a-4537-b9f5-155d5bc591b1

``#include "ESP32-HUB75-MatrixPanel-I2S-DMA.h"

 // default pins: 
  #define A_PIN  23
  #define B_PIN  19
  #define C_PIN  5
  #define D_PIN  -1
  #define E_PIN  -1
  
  #define R1_PIN  25
  #define R2_PIN  14
  #define G1_PIN  26
  #define G2_PIN  12
  #define B1_PIN  27
  #define B2_PIN  13
  
  #define LAT_PIN 4
  #define OE_PIN  15
  #define CLK_PIN 16

  // Panel configuration
  #define PANEL_RES_X 128 // Number of pixels wide of each INDIVIDUAL panel module. 
  #define PANEL_RES_Y 16 // Number of pixels tall of each INDIVIDUAL panel module.
  
  #define NUM_ROWS 1// Number of rows of chained INDIVIDUAL PANELS
  #define NUM_COLS 1 // Number of INDIVIDUAL PANELS per ROW
  
  // ^^^ NOTE: DEFAULT EXAMPLE SETUP IS FOR A CHAIN OF TWO x 1/8 SCAN PANELS
    
  // Change this to your needs, for details on VirtualPanel pls read the PDF!
  #define SERPENT false
  #define TOPDOWN false
  
  // placeholder for the matrix object
  MatrixPanel_I2S_DMA *dma_display = nullptr;
  
  /******************************************************************************
   * Setup!
   ******************************************************************************/
  void setup()
  {
    delay(250);
   
    Serial.begin(115200);
    Serial.println(""); Serial.println(""); Serial.println("");
    Serial.println("*****************************************************");
    Serial.println("*         1/4 Scan Panel                            *");
    Serial.println("*****************************************************");
  
  HUB75_I2S_CFG::i2s_pins _pins =
  { R1_PIN, G1_PIN, B1_PIN, R2_PIN, G2_PIN, B2_PIN, A_PIN, B_PIN, C_PIN, D_PIN, E_PIN, LAT_PIN, OE_PIN, CLK_PIN };

    HUB75_I2S_CFG mxconfig(
                PANEL_RES_X,              // DO NOT CHANGE THIS
                PANEL_RES_Y,              // DO NOT CHANGE THIS
                NUM_ROWS*NUM_COLS           // DO NOT CHANGE THIS
                //,_pins            // Uncomment to enable custom pins
    );
    mxconfig.driver = HUB75_I2S_CFG::SHIFTREG;  
    mxconfig.latch_blanking=1;
    mxconfig.clkphase = false; // Change this if you see pixels showing up shifted wrongly by one column the left or right.
    // OK, now we can create our matrix object
    mxconfig.i2sspeed =HUB75_I2S_CFG::HZ_10M;

    dma_display = new MatrixPanel_I2S_DMA(mxconfig);
  
    // let's adjust default brightness to about 75%
    dma_display->setBrightness8(10);    // range is 0-255, 0 - 0%, 255 - 100%
  
    // Allocate memory and start DMA display
    if( not dma_display->begin())
      Serial.println("****** !KABOOM! I2S memory allocation failed ***********");

    dma_display->clearScreen();
    delay(500);

   }

  void loop() 
  {
      uint16_t color = dma_display->color565(50, 50,50); //blue
      
      for (int y = 0; y < PANEL_RES_Y; y++)
     // for (int i = 0; i <128; i=i+2)
      {
         for (int x = 0; x < PANEL_RES_X; x++)
       //for (int f = 0; f < 64; f=f+1)
        {
            dma_display->drawPixel(x,y, color);
            Serial.print("drawPixel: ");
            Serial.print("(");
            Serial.print(x, DEC);
            Serial.print(",");
            Serial.print(y, DEC);
            Serial.println(")");
            delay(100);
          }
      }
      delay(5000);
      dma_display->clearScreen();

  } // end loop``

board707, 2023-09-16:
> Can yout tell me where int the library I can tune this mapping?

This is a question to the author, @mrfaptastic

veryslow, 2023-09-18:
Hi, after a lot of tests, I finally switched to hzeller library for Raspberry pi...
I wrote a scrappy algorithm to map the pixels for this P5 panel. I understood that the relevent thing is that they ae OUTDOOR panels with strange mapping... I did not consider this caracteristic as pertinent but it was apparently central.

here is the code for mapping in hzeller's library (If it can help anyone with thi s kinf of panels)
Panels are adressed as 128x16 matrix, and the code converts 64x32 coordinates to this 128x16 esoteric structure!

```
void MapSinglePanel(int x, int y, int *matrix_x, int *matrix_y) const {
    int pix_nb = x+(y*64);
    *matrix_x = ((((pix_nb%64)/4)+1)*8)-((((pix_nb%1024)/512)+1)*4)+(pix_nb%4);
    *matrix_y = ((pix_nb/64)%8)+((pix_nb/1024)*8);
 
```
I'm still interested in the way I could implement this code in ESP32-HUB75-MatrixPanel-DMA library!

Thanks for all your answers, and I come back in this issue if I find a way to do it with ESP32.

board707, 2023-09-18:
> Panels are adressed as 128x16 matrix, and the code converts 64x32 coordinates to this 128x16 esoteric structure!

Does this transform supports a chaining for more than one panel?

board707, 2023-09-18:
Perfect

ericchehata, 2024-07-26:
Hello, @veryslow I am also using the HUIDU HD-D16 asynchronous controller to control a 64x32 LED panel. I tried using your code for mapping in hzeller's library:

```
void MapSinglePanel(int x, int y, int *matrix_x, int *matrix_y) const {
    int pix_nb = x+(y*64);
    *matrix_x = ((((pix_nb%64)/4)+1)*8)-((((pix_nb%1024)/512)+1)*4)+(pix_nb%4);
    *matrix_y = ((pix_nb/64)%8)+((pix_nb/1024)*8);
```

But I am getting `Bus error`. I think I got the GPIOS mapping wrong. This is what I am using and it matches what you put in your first comment:
```
 {
    .name          = "d16",
    .output_enable = GPIO_BIT(15),
    .clock         = GPIO_BIT(16),
    .strobe        = GPIO_BIT(4),

    .a             = GPIO_BIT(23),
    .b             = GPIO_BIT(19),
    .c             = GPIO_BIT(5),
    .d             = GPIO_BIT(-1),
    .e		   = GPIO_BIT(-1),

    .p0_r1         = GPIO_BIT(25),
    .p0_g1         = GPIO_BIT(26),
    .p0_b1         = GPIO_BIT(27),
    .p0_r2         = GPIO_BIT(14),
    .p0_g2         = GPIO_BIT(12),
    .p0_b2         = GPIO_BIT(13),
  },

```

Is there anything I am missing? Any help would be greatly appreciated.

board707, 2024-07-26:
@ericchehata 
First of all, please write what panel you have. Size, as I understand it, is 64x32. Also, please indicate the number of scans and driver types.

And second, not all 64x32 panels have the same mapping, so the option by @veryslow may not work for you.

ericchehata, 2024-07-26:
@board707 Thank you for your quick reply. 

Yes I have a P5 RGB 64x32 panel. To be honest I am not sure how I can find the number of scans and driver types. But from the looks of it, the panel looks exactly like the one @veryslow showed above.

This is a picture of the panel I have:

[image]

Just FYI, I am really new in the LED screen world.

ericchehata, 2024-07-26:
Also not sure if this is useful, but these are the configurations I am using:

[image]
[image]

board707, 2024-07-26:
> the panel looks exactly like the one @veryslow showed above.

No,  the panels have different chip arrangements. 

From your picture it clear that you have a 8scans panel (take the note of label "P5-1921-64x32-8S..."). 
Could you please try to look at the ID of the chips I have highlighted in blue and red?
[image]

ericchehata, 2024-07-26:
ID of the blue highlighted chips:
```
DP5125D
481802
```

ID of the red highlighted chips:
```
RUC7258D
0AF01833
```

board707, 2024-07-26:
Thank you
The good news is that these are the most common driver and multiplexer, and any library supports them. This doesn't bring you any closer to solving your problems, but it shows that at least the matrix is ​​compatible with the current code.

Now explain in detail what problems you have with multiplexing. Unfortunately, I don't know the hzeller's library well. Have you tried connecting the panel to the ESP32 under the control of this library?

ericchehata, 2024-07-26:
I believe the main problem I have is getting my pin configuration correct. I am not sure where I can find it as no documentation is provided with the HD-D16 controller.

I think the Bus Error I get is due to this misconfiguration.

ericchehata, 2024-07-26:
The controller has its own software to manage the display and it is working fine. But I prefer using an open source software to control the panel.

board707, 2024-07-26:
I can't suggest anything about Bus error - I've never worked with Raspberry
As for multiplexing - judging by the picture of the settings that you provided for the HDplayer (table with pixel numbers) - you have a very simple version of the panel, which can be easily launched on the current library

ericchehata, 2024-07-26:
Okay thank you very much for your help. Do you have any advice on how I find the correct pin configuration?

board707, 2024-07-26:
If you using hzeller's library , I think it would better to open issue on that github

ericchehata, 2024-07-26:
I will attempt to use your library. Which configuration files need modifications to satisfy my hardware specifications? Is it just the pin configuration [here](https://github.com/mrcodetastic/ESP32-HUB75-MatrixPanel-DMA/blob/master/src/platforms/esp32/esp32-default-pins.hpp) or is there something else as well?

board707, 2024-07-26:
It would be better to ask author - @mrcodetastic

mrcodetastic, 2024-07-27:
I have no idea what the HD-16 is and this library is written for people who use specific ESP32's.

ericchehata, 2024-07-28:
@mrcodetastic I am referring to the following [Huidu controller](https://www.aliexpress.us/item/3256805357649867.html?spm=a2g0o.productlist.main.1.2a7d9351frOK3g&algo_pvid=1ca4c630-37aa-4238-8102-4106aafbf469&algo_exp_id=1ca4c630-37aa-4238-8102-4106aafbf469-0&pdp_npi=4%40dis%21USD%2138.65%2138.65%21%21%21278.06%21278.06%21%402101c9ac17221302623216638e4e1b%2112000034185061874%21sea%21US%210%21ABX&curPageLogUid=9OH0J0na7Cgl&utparam-url=scene%3Asearch%7Cquery_from%3A). I see that you have previously worked with a Huidu controller: https://github.com/mrcodetastic/HD-WF1-LED-MatrixPanel-DMA/blob/main/README.md. 

Any pointers on how I can set it up to work with the ESP32-HUB75-MatrixPanel-DMA library?

mrcodetastic, 2024-07-28:
Is it an esp32 based card?

ericchehata, 2024-07-28:
No I don't think so...

mrcodetastic, 2024-07-28:
Then this library is of no use, I'm afraid.