mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #502: Absolute maximum Pixel Count for S3?

DerPicknicker, 2023-09-24:
Hi, 

I just wanted to know how high is the led count while using an esp32s3 with an big amount of psram I have a model lying around with 16mb of psram. I tried your spreadsheet but unfortunately it was not possible for me to calculate the highest possible amount for an s3. 

Is it possible for your to estimate the numbers of total pixels? Is there any plans to increase that? Like using an additional esp just for the second part of the display? The esp32s3 has two i2s components so it should be possible to transfer data in a small amount of time. Do you think this could be an workaround for increasing the pixel size? 

Thank you for your time!!

mrfaptastic, 2023-09-24:
Hello

The esp32s3's i2s can't do the LCD output mode anymore, it's a seperate LCD module of which there is only one.

Theoretically you can have as many pixels as your want if you use psrsm, but you can only clock out the data at a maximum rate of about 10Mhz and for that you MUST have octal psrsm.

10mhz bandwidth won't drive a lot of panels without headache inducing flicker.

mrfaptastic, 2023-09-24:
I can't give you any more specifics as other than coding up support for the esp32s3 and psram, I haven't done much real world testing.

DerPicknicker, 2023-09-25:
@mrfaptastic ...

Thank you for your information... Do you think with 10mhz it's possible to drive matrix up to 128x64 or even bigger like 256x128? 

Which chaining method should I use?

Would you recommend using an P1.5 panel or P2.0? Or even bigger than P2.0?

Thank you anyways.

board707, 2023-09-25:
> Would you recommend using an P1.5 panel or P2.0? Or even bigger than P2.0?

This is a panel pitch in millimeters. How it connected to your question of the maximum pixel count?

DerPicknicker, 2023-09-25:
> > Would you recommend using an P1.5 panel or P2.0? Or even bigger than P2.0?
> 
> This is a panel pitch in millimeters. How it connected to your question of the maximum pixel count?

I am thinking about the sweet spot in terms of resolution and size... I saw there are some nice 128x64 / 64x64 panels on aliexpress...  but with p2.0 they are getting large... What is the maximum which you would recommend on an esp32(s3) .. 256x128 , 128x128, 64x64 ...?

board707, 2023-09-25:
Look in the issues and discussions of the repo. There have been reports of launching screens significantly larger than 128x128

DerPicknicker, 2023-09-25:
Thank you. I will have a look..

mrfaptastic, 2023-09-25:
You don't need to have the physical screens to test the system. Just load the firmware onto the S3 with you desired configuration and check the serial output to see if it works.

I cannot give any guarantees that when you do get the physical panels however, that it won't look be a flickery mess.

mrfaptastic, 2023-09-25:
You don't need to have the physical panels just to test the system. Just load the firmware onto the S3 with your desired configuration and check the serial output to see if it works/runs first.

I cannot give any guarantees that when you do get the physical panels however, that it won't look be a flickery mess.

You call also test with only one panel connected even if you have 10 'configured' to get a sense check for the colour depth and flicker you'll get across all the panels.

DerPicknicker, 2023-09-25:
Hi, i will try that later one. I also saw that the P2 panels are MUCH cheaper. I have to check it.. I will update this Issue if I tested different settings... 

One Last question... This Picture show the Module I plan to buy. 
[image]

If I want to build a 2x2 Matrix with them. So in total resolution 128x128 should fit also an normal esp32 or am I wrong: 
[image]

mrfaptastic, 2023-09-25:
Try a 128x128 sketch and see if it crashes from the serial output.

If it doesn't, then you might be ok to then order the panels. Pretty sure others have managed such resolutions, but unlikely at 24bpp.