mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #507: question about sm16208SJ chips

burgessa23, 2023-10-03:
I have a 64x64 panel that has sm16208SJ chips. Are these supported? Or same issue as sm1620B?

Thanks

board707, 2023-10-15:
The driver sm16208SJ is supported.
However, to confirm that your panel will work, we need to know the type of the second driver as well as such a parameter as the “number of scans”

orrious, 2023-11-29:
I can confirm that 64x64 panel with SM16208S work as well

orrious, 2023-11-30:
> I can confirm that 64x64 panel with SM16208S work as well

I'm going to amend my prior comment.  There must be a breaking change or PR that needs to be merged as the current version of ESP32-HUB75-MatrixPanel-DMA libraries do not work with the SM16208S.  There is a version of the library that is being distributed in https://www.waveshare.com/wiki/RGB-Matrix-P4-64x32.  Specifically in this zip https://files.waveshare.com/upload/6/6d/RGB-Matrix-P4-64x32-Demo.zip

board707, 2023-11-30:
>  64x64 panel with SM16208S

Too little information. As I said earlier, in addition to the number of pixels and the driver type, you need to know, at least, the number of scans and the type of multiplexer.

> Specifically in this zip ...  demo that does work with this chip

In your first comment you wrote about 64x64 panel, but the demo is for 64x32. It is a different things. And again, you say nothing about scans and multiplexor. Without this, your message is of little value.

mrfaptastic, 2023-11-30:
Can't that library just be used then?

orrious, 2023-11-30:
Sure, but it won't receive any of your updates, folks that are attempting to use your library and these panels will continually run into the same issue.  Maybe they will find this ticket, but that's a needle in a haystack.

I don't know enough about your library, I haven't dug that deep into it.  What is the down side to adding the lower frequency to the `clk_speed` array?

mrfaptastic, 2023-11-30:
If that's the only change to get these panels to run with this library then all you need to do is add the 8mhz option into the enum.

Feel free to raise a pull request.

closed 2024-01-09 by mrfaptastic