mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #509: Is there a way to produce a desired refresh rate?

toine512, 2023-10-13:
Hi.

The lib produces an arbitrary refresh rate above `HUB75_I2S_CFG::min_refresh_rate`. Is there a way to manually tweak the computations in `MatrixPanel_I2S_DMA::allocateDMAmemory()` so that a defined refresh rate is produced?
I'm using the ESP32-S3, the LCD-CAM clock cannot be fine-tuned with the fractional divider when the source is the 160 MHz PLL.

board707, 2023-10-15:
Did you tried to modify the code yourself?

toine512, 2023-10-15:
> I'm using the ESP32-S3, the LCD-CAM clock cannot be fine-tuned with the fractional divider when the source is the 160 MHz PLL.

PLL_D2_CLK allows using full divider capability thus adjusting LCD-CAM frequency quite precisely. Then how can the refresh rate be computed from LCD_CAM frequency? Values printed in the log don't seem to match.

closed 2024-01-09 by mrfaptastic