mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #512: Suddenly, after hours, the panel goes black

jgaetev, 2023-10-22:
Hello mrfaptastic, first of all, thank you for the great work you have done with this library. I would like you to give me some clue on how to debug the problem I have. I have a 64x32 panel with an ESP n16r8v, with a couple of light tasks and a web server. Everything works fine for a few hours (up to a couple of days), but without being able to see any errors or problems on the serial monitor (I have the debug level at 3), the panel stops displaying (it remains black), the code continues running as if nothing, executing each task normally, except for the display on the panel, no signal is seen on the hub75 control pins either. What do you recommend I do to determine what is happening and resolve this? thank you so much.

mrfaptastic, 2023-10-23:
Hello . Thank you for your kind words. Reduce the clock rate and see if that helps avoid the issue.

jgaetev, 2023-10-24:
Hello, thank you very much for the response, I reduced the clock speed to the minimum to 8 MHz, it has operated for 24 hours, the maximum has been a little more than 48, is there any method to detect that the DMA engine is operating?
Thanks again.

board707, 2023-10-24:
> he maximum has been a little more than 48

48 Mhz? - it seems to be a completely meaningless value, since the most of HUB75 panels has a maximum clock frequency about 25-30 MHz

jgaetev, 2023-10-24:
I thought you were referring to the i2sspeed, which I left at HZ_8M, I have a refresh rate of 114 fps with 2 panels in a chain, the esp32 clock is 240 mhz by default.

tsctrl, 2023-10-24:
you could try to use esp32s2 or esp32 to check if the issue related to your hardware. try disable those additional function as well, 24 hours is not that long to wait and confirm.

jgaetev, 2023-10-24:
Yes, that's what I planned to do, in any case is there a way to know if the DMA engine is operating?, thus determining if it is necessary to reset the system.
Thank you a lot!

mrfaptastic, 2023-10-24:
> Yes, that's what I planned to do, in any case is there a way to know if the DMA engine is operating?, thus determining if it is necessary to reset the system.

There is, but it depends on what exactly is causing the issue. The "cleanest" possible cause of the issue is due to some random DMA starvation occurring (perhaps some  code you have might be triggering this) so the DMA engine terminates the endless loop. The library doesn't check for this, and if it did occur it would require one of the DMA or FIFO empty interrupts to be raised and handled by the library code to restart things.

Or, it could be some random silicon / app bug that can't be detected.

mrfaptastic, 2023-10-24:
Potential areas to hack to check to see if the esp 32s3 still thinks it's sending it DMA data. Perhaps in the callback function have an integer that keeps counting up and see if that's still happening even when the screen in black.

https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-DMA/blob/master/src/platforms/esp32s3/gdma_lcd_parallel16.cpp#L265

Is the example callback that is generated every time a full scan has completed - so potentially 100+ times a second.

I think the ESP32/ESP32S2 had an interrupt that could also be raised not only when the "a round of" DMA data had been 'sent out' once, but also when DMA had finished entirely - which should never happen without being explicitly requested.

Not sure how the interrupt stuff works on the S3 as I only use the currently configured interrupt to assist with back buffer switch timing.

jgaetev, 2023-10-25:
Thank you mrfaptastic, I implemented the counter in the callback function now to wait if it works as a flag to reset the system. 

Bye !

mrfaptastic, 2023-10-27:
Was it still incrementing even when the screen went black?

jgaetev, 2023-10-27:
Hello mrfaptastic, I have it in tests, for now it is 27 continuous hours, I am going to leave it for at least a week... anyway, if the counter stops increasing, it should reset the esp, which I can also detect with the continuous hours of operation counter.
Thank you!

closed 2024-01-09 by mrfaptastic