mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #516: 64x32 1/32 only works properly when PANEL_RES_Y set to 64

kucingpresto, 2023-10-26:
Hi. I have a problem, in my current configuration i have a 64x32 panel. i got it working properly, but it only shows fully when in the sketch i use PANEL_RES_Y to 64, if i set if i change it to 32 it only draws 16 pixels. Also the display shifted 1 pixel in row number 17 and below. Heres the attached photos 
[image]
[image]

bluesubway, 2023-10-28:
I have always had this confusion and don't know how to solve it

changathi123, 2025-12-17:
@board707 @mrcodetastic  i tried it myself ,i cant fix it, plzz help

board707, 2025-12-17:
@changathi123 
On your next problem please open a separate issue, do not hijack an old one...

board707, 2025-12-17:
> i need to show a jpg image in this display like portrait view

Sorry, I can't help with it, do not know this library very well.