mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #518: 128x64-32S Not working

HennieLP, 2023-11-05:
[image]
[image]
[image]
[image]
Screen is just black if I press the reset button on the ESP screen some times Flashes

HennieLP, 2023-11-06:
only flashing on reset if mxconfig.driver = HUB75_I2S_CFG::FM6126A is set

board707, 2023-11-06:
Your panel's driver is DP3246. It require additional configuration before work and is not fully compatible with the library.
By the way, you panel should contains a three types of chip, but you show us only two.

closed 2023-11-06 by HennieLP