mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #520: support for new arduino-esp32 core v3.0.0

beta-tester, 2023-11-12:
hi, i saw [Announcing the Arduino ESP32 Core version 3.0.0](https://blog.espressif.com/announcing-the-arduino-esp32-core-version-3-0-0-3bf9f24e20d4).
there will be breaking changes and new features. also I2S is listed there.
i guess `ESP32-HUB75-MatrixPanel-DMA` will be affected from the changes as well (or can benefit from the new I2S API).
is there already a working implementation of this project available for the new arduino-esp32 core?

mrfaptastic, 2023-11-26:
> is there already a working implementation of this project available for the new arduino-esp32 core?

Not that I am aware of. I installed the Arduino IDE and used the latest alpha2 version of the esp32-idf and I see it has failed to build with errors such as:

```
C:\Users\Anonymous\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-6b1f40b9bf/esp32/include/driver/deprecated/driver/i2s.h:27:2: warning: #warning "This set of I2S APIs has been deprecated, please include 'driver/i2s_std.h', 'driver/i2s_pdm.h' or 'driver/i2s_tdm.h' instead. if you want to keep using the old APIs and ignore this warning, you can enable 'Suppress leagcy driver deprecated warning' option under 'I2S Configuration' menu in Kconfig" [-Wcpp]
   27 | #warning "This set of I2S APIs has been deprecated, \
      |  ^~~~~~~
c:\Users\Anonymous\Documents\Arduino\libraries\ESP32_HUB75_LED_MATRIX_PANEL_DMA_Display\src/platforms/esp32/esp32_i2s_parallel_dma.hpp:65:1: error: 'i2s_dev_t' does not name a type; did you mean 'i2s_dir_t'?
   65 | i2s_dev_t* getDev();
      | ^~~~~~~~~
      | i2s_dir_t
c:\Users\Anonymous\Documents\Arduino\libraries\ESP32_HUB75_LED_MATRIX_PANEL_DMA_Display\src/platforms/esp32/esp32_i2s_parallel_dma.hpp:148:14: error: 'i2s_dev_t' does not name a type; did you mean 'i2s_dir_t'?
  148 |     volatile i2s_dev_t* _dev;
```

mrfaptastic, 2023-11-26:
I see that the Espressif migration guide provides no commentary on I2S LCD mode code changes that this library depends on. For now, this library won't support 3.0.0.

mrfaptastic, 2023-11-27:
Made some changed with commit 7a100a0a35d300876c44c81d9e818961495b9c4d - should work now across ESP32/S2/S3 on Arduino 3.0.0 alpha.

closed 2023-12-03 by mrfaptastic