mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #521: Using ESP32CAM w/ HUB75E

flewid, 2023-11-22:
Hello, 

I am wondering if it's possible to use the ESP32CAM module with this driver (it appears to work for me in testing) however I'd like to get 'input' from the camera to actually display on the matrix. Has anyone tried this before?

mrfaptastic, 2023-11-26:
I don't think there are enough spare GPIO's on the ESP32CAM to do this?

flewid, 2023-11-27:
Ah good point - I suppose not eh?

[image]

darn okay - will have to do some kind of server/client thing with the cam module + esp32 separate I guess eh?

mrfaptastic, 2023-11-27:
Yeah it would be much easier to just get another ESP32 to do the RGB panel display stuff. Perhaps setup a direct wifi link between the two and send the frame data across when captured.

Of course,  you're not going to get high speed video with this... But even if the esp32cam had enough pins, I doubt you'd be able to do either.

flewid, 2023-11-27:
Okay great thank you for the input, will see what i can do!

flewid, 2023-11-29:
OO okay I'm down to work on something - I actually found this guy on  [youtube](https://www.youtube.com/shorts/BsvCnc151FE) - tracked him down and got the code  - which I've spent the past week on, but not functional yet - I'll attach it here - [non_working_esp32cam.zip](https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-DMA/files/13494917/non_working_esp32cam.zip)

However, he's using wifi from esp32cam (server) -> client (esp32) to get the data - How would the DMA variety work? If you have something I can look into (as a newb) be happy to start playing around - I assume I should just fork your repo and go from there eh? (create a new example basically)

mrfaptastic, 2023-11-29:
Shouldn't need to fork the library at all. The library just pushes pixel to a panel. Your sketch, even his, should just use the library as-is. The trick is getting the pixels getting the camera and then "sending them across" to this library, that's all.

flewid, 2023-11-29:
Got it! Okay will post an update when I have time to mess around :)

flewid, 2023-12-12:
@mrfaptastic i've been playing with it for a week or so now - the server (esp32cam) seems to be functioning fine, i've got the client (esp32) working fine as well, it's just the 'decode the jpeg' stream i'm having difficulty with - I invited you to my repo so you can check things out (private for now)

closed 2024-01-28 by mrfaptastic