mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #522: 1/8 Scan Panel Chained problem

Cenatronics, 2023-11-25:
Hello,
I would like to thank everyone who helped create this beautiful library.
I'm having a problem, and from what I understand no one else has experienced this.
I have 4 64x32 panels labeled P5-1921-64*32-8S-S2. This panel operates at 1/8 scan. The FourScanPanel sample application works flawlessly, but I couldn't succeed with the ChainedPanel example despite trying many configurations. I ask for your help regarding this.

Cenatronics

board707, 2023-11-25:
Using the FourScanPanel mode for 8scan panel is a trick, based on representation of 8 scan panel as two 4 scan. But I doubt that the chaining is supported it that mode.

Cenatronics, 2023-11-25:
Thank you for your quick response. You are right, although I have tried many codes, I am sure that it is not supported. Do you have any suggestions?
Or should I buy other panels? I want to use DMA, so using another library would not be good for my project.

Cenatronics

board707, 2023-11-25:
To avoid problems it is recommended to choose panels with scan as a half of height. i.e 64x32 s16, 80x40 s20, 32x16 s8 and so on. These panels has a simplest internal design and most of them are supported by any HUB75 library.

Cenatronics, 2023-11-25:
thanks for  your comments.

closed 2023-11-25 by Cenatronics