mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #530: distorted and inaccurate image

K0rkunc, 2023-11-30:
Hello, I have created a display with 2 columns, 6 rows, and 32x16 p8 RGB panels, but I couldn't get meaningful images. I tried activating the top two rows to experiment, but I still got complex results on the screen.

When I fill the screen like this in my ICN2037 chip-based displays, I can see the correct colors:

However, when I try to draw text or lines, I get distorted and inaccurate images. Could you please assist me with this?

[image]
[image]

COde 👍 

`
    #include <ESP32-VirtualMatrixPanel-I2S-DMA.h>

// 2) Set configuration
    #define PANEL_RES_X 32 // Number of pixels wide of each INDIVIDUAL panel module. 
    #define PANEL_RES_Y 16 // Number of pixels tall of each INDIVIDUAL panel module.

    #define NUM_ROWS 1 // Number of rows of chained INDIVIDUAL PANELS
    #define NUM_COLS 2 // Number of INDIVIDUAL PANELS per ROW
    #define PANEL_CHAIN NUM_ROWS*NUM_COLS    // total number of panels chained one to another

    /* Configure the serpetine chaining approach. Options are:
        CHAIN_TOP_LEFT_DOWN
        CHAIN_TOP_RIGHT_DOWN
        CHAIN_BOTTOM_LEFT_UP
        CHAIN_BOTTOM_RIGHT_UP

        The location (i.e. 'TOP_LEFT', 'BOTTOM_RIGHT') etc. refers to the starting point where 
        the ESP32 is located, and how the chain of panels will 'roll out' from there.

        In this example we're using 'CHAIN_BOTTOM_LEFT_UP' which would look like this in the real world:

        Chain of 4 x 64x32 panels with the ESP at the BOTTOM_LEFT:

        +---------+---------+
        |    4    |    3    |
        |         |         |
        +---------+---------+
        |    1    |    2    |
        |  (ESP)  |         |
        +---------+---------+      

    */
    #define VIRTUAL_MATRIX_CHAIN_TYPE CHAIN_BOTTOM_LEFT_UP 

// 3) Create the runtime objects to use

    // placeholder for the matrix object
    MatrixPanel_I2S_DMA *dma_display = nullptr;

    // placeholder for the virtual display object
    VirtualMatrixPanel  *virtualDisp = nullptr;

/******************************************************************************
 * Setup!
 ******************************************************************************/
void setup() {
  
  delay(2000);
  Serial.begin(115200);
  Serial.println(""); Serial.println(""); Serial.println("");
  Serial.println("*****************************************************");
  Serial.println(" HELLO !");
  Serial.println("*****************************************************");

  /******************************************************************************
   * Create physical DMA panel class AND virtual (chained) display class.   
   ******************************************************************************/

  /*
    The configuration for MatrixPanel_I2S_DMA object is held in HUB75_I2S_CFG structure,
    All options has it's predefined default values. So we can create a new structure and redefine only the options we need

	Please refer to the '2_PatternPlasma.ino' example for detailed example of how to use the MatrixPanel_I2S_DMA configuration
  */

  HUB75_I2S_CFG mxconfig(
                PANEL_RES_X,   // module width
                PANEL_RES_Y,   // module height
                PANEL_CHAIN    // chain length
  );

  //mxconfig.driver = HUB75_I2S_CFG::FM6126A;     // in case that we use panels based on FM6126A chip, we can set it here before creating MatrixPanel_I2S_DMA object

  // Sanity checks
  if (NUM_ROWS <= 1) {
    Serial.println(F("There is no reason to use the VirtualDisplay class for a single horizontal chain and row!"));
  }

  // OK, now we can create our matrix object
  dma_display = new MatrixPanel_I2S_DMA(mxconfig);

  // let's adjust default brightness to about 75%
  dma_display->setBrightness8(192);    // range is 0-255, 0 - 0%, 255 - 100%

  // Allocate memory and start DMA display
  if( not dma_display->begin() )
      Serial.println("****** !KABOOM! I2S memory allocation failed ***********");

  // create VirtualDisplay object based on our newly created dma_display object
  virtualDisp = new VirtualMatrixPanel((*dma_display), NUM_ROWS, NUM_COLS, PANEL_RES_X, PANEL_RES_Y, VIRTUAL_MATRIX_CHAIN_TYPE);

  // So far so good, so continue
  virtualDisp->fillScreen(virtualDisp->color444(0, 0, 0));
  virtualDisp->drawDisplayTest(); // draw text numbering on each screen to check connectivity

 // delay(1000);

  Serial.println("Chain of 4x 64x32 panels for this example:");
  Serial.println("+---------+---------+");
  Serial.println("|    4    |    3    |");
  Serial.println("|         |         |");
  Serial.println("+---------+---------+");
  Serial.println("|    1    |    2    |");
  Serial.println("| (ESP32) |         |");
  Serial.println("+---------+---------+");

   // draw blue text
   virtualDisp->setFont(&FreeSansBold12pt7b);
   virtualDisp->setTextColor(virtualDisp->color565(0, 0, 255));
   virtualDisp->setTextSize(1); 
   virtualDisp->setCursor(1, 1);    
   virtualDisp->print("ABCD");

}

void loop() {
  

} `

K0rkunc, 2023-12-03:
@mrfaptastic can u help me ??  If it is not a problem for you, you can also get paid support from you

mrfaptastic, 2023-12-03:
Try four_scan_panel example

But try changing

    FourScanPanel->setPhysicalPanelScanRate(FOUR_SCAN_32PX_HIGH);

To

    FourScanPanel->setPhysicalPanelScanRate(FOUR_SCAN_16PX_HIGH);

K0rkunc, 2023-12-03:
Tried but again same

board707, 2023-12-03:
Is it works with one panel?

K0rkunc, 2023-12-03:
What I display with the code below is as in this image.
[image]

```
/*************************************************************************
 * Description: 
 * 
 * The underlying implementation of the ESP32-HUB75-MatrixPanel-I2S-DMA only
 * supports output to 1/16 or 1/32 scan panels - which means outputting 
 * two lines at the same time, 16 or 32 rows apart. This cannot be changed
 * at the DMA layer as it would require a messy and complex rebuild of the 
 * library's DMA internals.
 *
 * However, it is possible to connect 1/8 scan panels to this same library and
 * 'trick' the output to work correctly on these panels by way of adjusting the
 * pixel co-ordinates that are 'sent' to the ESP32-HUB75-MatrixPanel-I2S-DMA
 * library.
 * 
 **************************************************************************/
#include "ESP32-HUB75-MatrixPanel-I2S-DMA.h"

/* Use the Virtual Display class to re-map co-ordinates such that they draw 
 * correctly on a 32x16 1/8 Scan panel (or chain of such panels).
 */
#include "ESP32-VirtualMatrixPanel-I2S-DMA.h" 

  // Panel configuration
  #define PANEL_RES_X 32 // Number of pixels wide of each INDIVIDUAL panel module. 
  #define PANEL_RES_Y 32 // Number of pixels tall of each INDIVIDUAL panel module.
  
  
  #define NUM_ROWS 1 // Number of rows of chained INDIVIDUAL PANELS
  #define NUM_COLS 1 // Number of INDIVIDUAL PANELS per ROW
  
  // ^^^ NOTE: DEFAULT EXAMPLE SETUP IS FOR A CHAIN OF TWO x 1/8 SCAN PANELS
    
  // Change this to your needs, for details on VirtualPanel pls read the PDF!
  #define SERPENT true
  #define TOPDOWN false
  
  // placeholder for the matrix object
  MatrixPanel_I2S_DMA *dma_display = nullptr;

  // placeholder for the virtual display object
  VirtualMatrixPanel  *OneEightMatrixDisplay = nullptr;
  
  /******************************************************************************
   * Setup!
   ******************************************************************************/
  void setup()
  {
    delay(250);
   
    Serial.begin(115200);
    Serial.println(""); Serial.println(""); Serial.println("");
    Serial.println("*****************************************************");
    Serial.println("*         1/8 Scan Panel Demonstration              *");
    Serial.println("*****************************************************");
  
/* 
     // 62x32 1/8 Scan Panels don't have a D and E pin!
     
     HUB75_I2S_CFG::i2s_pins _pins = {
      R1_PIN, G1_PIN, B1_PIN, R2_PIN, G2_PIN, B2_PIN, 
      A_PIN, B_PIN, C_PIN, D_PIN, E_PIN, 
      LAT_PIN, OE_PIN, CLK_PIN
     };
*/
    HUB75_I2S_CFG mxconfig(
                PANEL_RES_X*2,              // DO NOT CHANGE THIS
                PANEL_RES_Y/2,              // DO NOT CHANGE THIS
                NUM_ROWS*NUM_COLS           // DO NOT CHANGE THIS
                //,_pins            // Uncomment to enable custom pins
    );
    
    mxconfig.clkphase = false; // Change this if you see pixels showing up shifted wrongly by one column the left or right.
    
    //mxconfig.driver   = HUB75_I2S_CFG::FM6126A;     // in case that we use panels based on FM6126A chip, we can set it here before creating MatrixPanel_I2S_DMA object
  
    // OK, now we can create our matrix object
    dma_display = new MatrixPanel_I2S_DMA(mxconfig);
  
    // let's adjust default brightness to about 75%
    dma_display->setBrightness8(96);    // range is 0-255, 0 - 0%, 255 - 100%
  
    // Allocate memory and start DMA display
    if( not dma_display->begin() )
      Serial.println("****** !KABOOM! I2S memory allocation failed ***********");

   
    dma_display->clearScreen();
    delay(500);
    
    // create OneEightMatrixDisplaylay object based on our newly created dma_display object
    OneEightMatrixDisplay = new VirtualMatrixPanel((*dma_display), NUM_ROWS, NUM_COLS, PANEL_RES_X, PANEL_RES_Y, SERPENT, TOPDOWN);
    
	// THE IMPORTANT BIT BELOW!
    OneEightMatrixDisplay->setPhysicalPanelScanRate(NORMAL_ONE_SIXTEEN);
  }

  
  void loop() {

      
        OneEightMatrixDisplay->setTextColor(OneEightMatrixDisplay->color565(255, 255, 255));
        OneEightMatrixDisplay->setCursor(1, 1); 
      
        OneEightMatrixDisplay->print("Panel " );
       

      delay(2000);
      dma_display->clearScreen();
  
  } // end loop
```

mrfaptastic, 2023-12-05:
You will need to write a sketch that lights up over pixel art a time starting to from top left row and working left, then the next row etc.

From this is might be possible to deduce the mapping of this panel.

Please refer to the pinned issue for example code.

Take a video.

closed 2024-01-09 by mrfaptastic