mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #531: Support for 64x32 SM1666PF RUL6024 or a 64x32 RUC7258D RUL2038S?

michaelnaman, 2023-11-30:
Hi,

I'm struggling to get either of my panels to work, doubled checked wiring, power, etc and all good.

Is anyone able to confirm whether either of the driver combinations are supported and if they are, what is the config required to get them to run?

Thanks

board707, 2023-12-01:
RUL6024 - unsupported driver.

 64x32 RUC7258D RUL2038S - should be fine if the panel has 16 scans

michaelnaman, 2023-12-01:
[image]
 I'veg ot this far with the 64x32 RUC7258D RUL2038S but the bottom half of the display is green. I rewired it twice now and tested different panels (same spec) plus there's a wire from the ground on the matrix power to the ESP32.

I'm using an ESP32 Wroom-32.

Any ideas on how to fix this?

closed 2023-12-01 by michaelnaman

mrfaptastic, 2023-12-03:
@michaelnaman - what was the solution in the end?