mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #533: New HUB75 driving concept

ScobbyDoo, 2023-12-03:
As of now to drive HUB75 port we need almost 14 GPIO  including E pin 

Instead of that by using DMA & High speed i2s it is possible to drive 74HC595 serial to parallel shift register ic.

By feeding serially highspeed data , parallel pins act as output bits.

https://github.com/hpwit/I2SClocklessVirtualLedDriver

Here is the example to drive timing controlled addressable led with virtual pins (74HC595).

here we've advantage that strict timings can't affect as addressable led required so that we can reduce timings as well as addressable led required 2 pulses High & Low timings ,but here is only one.

Additionally there is (8pins x 2 ic = 16pin -14 pins used) 2 pin spare which can be drive by calling simple function.

So with using 3 pins we get fully functional HUB75 & also 2 digital Output pins 

Also can Drive multiple HUB75 parallel to extend Height of display without chaining zigzag wiring

board707, 2023-12-03:
> Here is the example to drive timing controlled addressable led with virtual pins (74HC595).

Addressable leds drives by 800 kHz, the HUB75 signal is 10-20 MHz.
Moreover, HUB75 is parallel, but 74HC595 is serial. Changing 14 pins output from parallel to serial requires 14 times more frequency

mrfaptastic, 2023-12-03:
The point of this library was to enable users to be able to simply connect an esp32's gpio pins directly to a hub 75 panel and get it to work... there are other libraries on github which enable the use of such methods (such as the smart matrix library I believe) but this is not something that I intend to implement in this library.

As @board707 says anyway. Can't be done given the frequencies required.

Longer term if you look at the discussion forum you'll see a lot of chat about new styles of "PWM" panels that need far fewer pins now anyway as they are essentially becoming like LCD displays with their own onboard RAM and seemingly need fewer pins as well.

board707, 2023-12-08:
@ScobbyDoo  Try to implement your idea in practice, in real hardware. Based on the results, it will be clear which of us is right.

In my opinion, your idea is absolutely hopeless. The most critical point of working with hub75 matrices is the speed of data output. The drivers of most panels do not support clock frequencies of more than 20-25 MHz, which imposes a limit on the screen size at approximately 20-30 thousand pixels. That is why, to create large screens, they are trying to set up parallel output of the signal to several chains of matrices.
Your method moves the situation backwards. Instead of parallel output, you even propose to do output to one matrix sequentially. As I already wrote, this will reduce the output speed by 14 times, so even updating one 64x32 panel will be a problem.

closed 2024-01-09 by mrfaptastic