mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #536: help needed: stream video over WiFI

HarikrishnanBalagopal, 2023-12-10:
# Overview

Hi, is there a way to stream video over WiFi and display it using this library?
I have a 128x128 (4 64x64 panel) setup. 2 chains 2 panels each.
I can't allocate a 128x128x3 byte array because the ESP32 doesn't have enough memory.
So I have a 64x64x3 byte array that I fill and then use that to read the current frame over the network.
Then I display that image in one of the 4 panels and read the next panel of the current frame.
Frames are drawn using https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-DMA/blob/722358ad2d990d0aa36600da95e4b36b740ff5f7/src/ESP32-VirtualMatrixPanel-I2S-DMA.h#L463

Needless to say the frame rate is bad. Around 3-5 frames per second in the best case.

## Help needed

- Some example code if anyone has managed to stream video over the network and get decent FPS.
- Some way to access the internal arrays of the library so I can read the data directly into it. This avoids me having to allocate a separate array and run out of memory. The server/client providing the frames usually run on higher end hardware (Desktop PC) and can produce complicated bit patterns if required.

## Things I have tried

- Raw TCP socket reads.
- ESP32 as a client pulling frames from a server running on my PC.
- ESP32 as a server with my PC pushing frame to it.

## Things I have not tried

- Hacking the library to get access to internal data structures
- Compression. Using a JPEG decompression library like in this video to stream frames https://youtu.be/G6MROvlLeKE

mrfaptastic, 2023-12-11:
From the sounds of it ii can successfully stream pixel data but can't get the performance you need?

3-5fps seems about right to me. These LED panels have little resemblance to a LCD panel like in that video as these LED panels need more CPU because they're so "dumb".

You could try turning down the library colour depth. Send 16 bit 565 colour instead of 24 bits. Use UDP not TCP.

Probably need to write a scaled back custom library potentially I think. ESP32 (the original one) is pretty old as well.

Or, use a raspberry pi instead.

drvkmr, 2023-12-11:
I have been able to stream e131 (sacn) at 60fps on a 64x64 matrix. I don't have the entire code organised but can share a few snippets which can help you. From the 'sending' side, you need a way of sending sacn data (it's similar to artnet). Many softwares can help you do that, and you can write simple softwares yourself if needed as well.

[Gist to the specific code which gets and displays data.](https://gist.github.com/drvkmr/6ade8bcb5b8476f05910dea27ecd88f2)

[E131 library I am using (modified from original).](https://github.com/drvkmr/ESPAsyncE131)

Finally one thing which took me a long time to realise is that you need a decent router WiFi streaming. I have a 5g router which does a terrible job at this, but a cheap £20 somehow does exactly what I need.

Feel free to ask if you are struggling.

aokailin, 2023-12-29:
我已经使用此库做成功了一个80*40分辨率的WIFI流传输，并通过SIGNALRGB控制它！

mrfaptastic, 2023-12-29:
您愿意分享代码吗？

aokailin, 2023-12-29:
我的代码很乱，你愿意看的话。它还集成了MDNS、web服务，wifi配网等，仿WLED协议和SIGNALRGB通讯，但是我不知道怎么上传到这个网站，我也才开始登陆这个网站