mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #539: Flicking issue with 32x16 panel - Please help!

priya2212, 2023-12-14:
Dear @mrfaptastic , Thank you very much for making the resources public, they are helping us alot in further development. 

I am using 16x32 panels with 1 row and 2 column configuration. I am successfully able to print the static text on display. But when scrolling, the text seems to flicker or two adjacent column are getting on at same time (visibility like - flickering and bold)-> When increasing delay, I see no flickering. Would be really great if you can give me your valuable suggestions. I am stuck from past few days in this issue. 

Platform - esp-idf
Development board - esp32-s3-n16r8

Attached is the code snippet for your reference. 

```
void scroll_text_task(void *pvParameter)
{
  MatrixPanel_I2S_DMA *dma_display = nullptr;
  VirtualMatrixPanel *FourScanPanel = nullptr;

  HUB75_I2S_CFG mxconfig(
      PANEL_RES_X * 2,    // DO NOT CHANGE THIS
      PANEL_RES_Y / 2,    // DO NOT CHANGE THIS
      NUM_ROWS * NUM_COLS // DO NOT CHANGE THIS
                          //,_pins            // Uncomment to enable custom pins
  );
  mxconfig.clkphase = false; // Change this if you see pixels showing up shifted wrongly by one column the left or right.
  mxconfig.min_refresh_rate = 255;

  // mxconfig.driver   = HUB75_I2S_CFG::FM6126A;     // in case that we use panels based on FM6126A chip, we can set it here before creating MatrixPanel_I2S_DMA object

  // OK, now we can create our matrix object
  dma_display = new MatrixPanel_I2S_DMA(mxconfig);

  dma_display->begin();
  dma_display->setBrightness8(100);
  dma_display->clearScreen();

  // create FourScanPanellay object based on our newly created dma_display object
  FourScanPanel = new VirtualMatrixPanel((*dma_display), NUM_ROWS, NUM_COLS, PANEL_RES_X, PANEL_RES_Y);

  // // THE IMPORTANT BIT BELOW!
  FourScanPanel->setPhysicalPanelScanRate(FOUR_SCAN_16PX_HIGH);
  int16_t y = 2;
  int16_t speed = 10;
  String flight = "HELLO WORLD!";
  int w_text = 64; // screen width
  int len = flight.length() * 6;
  FourScanPanel->clearScreen();
  FourScanPanel->setTextSize(1);     
  FourScanPanel->setTextWrap(false); // Don't wrap at end of line - will do ourselves
  FourScanPanel->setTextColor(FourScanPanel->color444(15, 0, 0));

  while (1)
  {
    // display flight
    FourScanPanel->flipDMABuffer();
    FourScanPanel->fillScreen(0); // Now clear the back-buffer
    FourScanPanel->setCursor(w_text, y);

    FourScanPanel->print(flight);

    w_text--;
    if (w_text + len == 0)
    {
      w_text = 64;
    }
    delay(speed);
  }
}

extern "C" void app_main()
{
     xTaskCreatePinnedToCore(&scroll_text_task, "scroll_text_task", 1024 * 8, NULL, 1, NULL, 1);

}
```

With delay - 10ms

https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-DMA/assets/20925260/9f31b833-f3ea-460c-867a-70926f0cb8af

With delay - 200ms

https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-DMA/assets/20925260/a41121c1-6329-4567-896a-781993d4e741

mrfaptastic, 2023-12-14:
Thank you for your kind words. 

Can you take an ultra slow motion video to check that it isn't an optical illusion?

priya2212, 2023-12-14:
@mrfaptastic , thank you very much for prompt and positive response. 
Attaching here is the slow motion video. We can see the flickering. I see visually that for each pixel, 3 leds are getting ON. 

https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-DMA/assets/20925260/c4e1921c-9462-4cd6-a17f-b68a9397cf86

Looking forward for your esteemed response.

board707, 2023-12-14:
Looks like the OE and multiplexing are not fully synchronized.

priya2212, 2023-12-14:
@board707 , thank you for your kind attention to this issue and providing a quick response. 
So, when increasing the delay, it resolves the issue?? 
How can I debug it?

board707, 2023-12-14:
Sorry, perhaps my response was too quick ;)
The fact that the issue resolves on slow scrolling, making me think that my idea was incorrect. Or probably it is not a main reason of the effect

board707, 2023-12-14:
@mrfaptastic 
Is this line means using a dual buffering?
```
FourScanPanel->flipDMABuffer();
```

priya2212, 2023-12-14:
@board707 , yes, I have recorded slow-motion of 200ms case, doubling of dots is there but hardly visible comparing to 10ms case.

mrfaptastic, 2023-12-14:
Double buffering isn't perfect..

Best create your own memory pixel buffer, and when you've updated it, then draw to this library's buffer.

priya2212, 2023-12-19:
Hi @mrfaptastic , thank you for your time and valuable inputs n sorry for the the delayed response. If possible at your end, please share a near about example I'll try to contribute in creating a good one with that.

closed 2024-01-09 by mrfaptastic