mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #540: P4 LED Panel 1/32 s

mjmokhtar, 2023-12-15:
[image]

I have some problems using this panel
1. panel cut in middle at panel
2. color must be RED become green

https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-DMA/assets/108962994/53a9be78-c899-4656-82f2-2f72d2063595

here my code
`#include "ESP32-HUB75-MatrixPanel-I2S-DMA.h"

//************************************************************ Display Config Declaration
#define double_buffer

#define PANEL_RES_X 64  // Number of pixels wide of each INDIVIDUAL panel module.
#define PANEL_RES_Y 32  // Number of pixels tall of each INDIVIDUAL panel module.
#define PANEL_CHAIN 2   // Total number of panels chained one to another

MatrixPanel_I2S_DMA* dma_display = nullptr;

#define FONT_SIZE 2.85 // Text will be FONT_SIZE x 8 pixels tall.

int delayBetweeenAnimations = 35; // How fast it scrolls, Smaller == faster
int scrollXMove = -1; //If positive it would scroll right

int textXPosition = PANEL_RES_X * PANEL_CHAIN;    // Number of pixels wide of each INDIVIDUAL panel module.
// Will start one pixel off screen
int textYPosition = PANEL_RES_Y / 2 - (FONT_SIZE * 8 / 2); // This will center the text

// For scrolling Text
unsigned long isAnimationDue;

String text ="Restek";

uint16_t myBLACK = dma_display->color565(0, 0, 0);
uint16_t myWHITE = dma_display->color565(255, 255, 255);
uint16_t myRED = dma_display->color565(255, 0, 0);
uint16_t myGREEN = dma_display->color565(0, 255, 0);
uint16_t myBLUE = dma_display->color565(0, 0, 255);

// Will be used in getTextBounds.
int16_t xOne, yOne;
uint16_t w, h;

void setup() {

  Serial.begin(115200);  
  // Configurations for Display
  HUB75_I2S_CFG mxconfig(
    PANEL_RES_X,  // module width
    PANEL_RES_Y,  // module height
    PANEL_CHAIN   // Chain length
  );

  mxconfig.double_buff = true;
  mxconfig.gpio.e = 32;
//  mxconfig.clkphase = false;
//  mxconfig.driver = HUB75_I2S_CFG::FM6124;

  // Display Setup
  dma_display = new MatrixPanel_I2S_DMA(mxconfig);
  dma_display->begin();
  dma_display->setBrightness8(255);  //0-255
  dma_display->fillScreen(myBLACK);
  dma_display->setTextSize(FONT_SIZE); 
  dma_display->setTextWrap(false); // N.B!! Don't wrap at end of line 
  dma_display->setTextColor(myRED); // Can change the colour here
  
  
}

void loop(){
  unsigned long now = millis();
  if (now > isAnimationDue)
  {
    // This tells the code to update the second buffer
    dma_display->flipDMABuffer();

    // This sets the timer for when we should scroll again.
    isAnimationDue = now + delayBetweeenAnimations;

    textXPosition += scrollXMove;

    // Checking if the very right of the text off screen to the left
    dma_display->getTextBounds(text, textXPosition, textYPosition, &xOne, &yOne, &w, &h);
    if (textXPosition + w <= 0)
    {      
      textXPosition = dma_display->width();
    }

    dma_display->setCursor(textXPosition, textYPosition);

    // The display has to do less updating if you only black out the area
    // the text is
    //dma_display->fillScreen(myBLACK);
    dma_display->fillRect(0, textYPosition, dma_display->width(), FONT_SIZE * 8, myBLACK);
    dma_display->print(text);
  }
}`

as any one have suggest for fix this

mjmokhtar, 2023-12-16:
this panel using DP5125D 
please help me mr [mrfaptastic](https://github.com/mrfaptastic)

mrfaptastic, 2023-12-16:
Can you test with another library as well? Like PxMatrix library.

If you have same issues then you have defective panel.

You can easily workaround the red/green issue by just swapping the pins around in the config.

mjmokhtar, 2023-12-16:
actually, when I use 64x64 px, the panel can display not cutting like my video.
I've using PxMatrix and it's same 
and I've read documentation of PxMatrix, that's library just for one panel 

so i've don't know what I should I do for this think

mjmokhtar, 2023-12-16:
https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-DMA/issues/516

its same like this

board707, 2023-12-16:
If the label "32scan" is not a mistake, it is something completely new among the HUB75 panels.
All the panels I had to deal with had a scan factor of no more than half the height. That is, for a 64x32 matrix, the scan should be 16 or less.
Your matrix is some new type and I don’t know any libraries that support such panels.

mjmokhtar, 2023-12-16:
for your information mr
my panel have led driver IC DP5125D

i tried to find datasheet but I didn't found it

once again when I use 64x64 px, the panel can display not cutting like my video
https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-DMA/issues/516

its same like this

mjmokhtar, 2023-12-16:
when i change panel Y from 32 to 64 
display not cutting but it will be cutting when I define Panel Y become 32

mjmokhtar, 2024-03-27:
i've try scan cordinate using this code

`#include "ESP32-HUB75-MatrixPanel-I2S-DMA.h"

//************************************************************ Display Config Declaration
#define double_buffer

#define PANEL_RES_X 64  // Number of pixels wide of each INDIVIDUAL panel module.
#define PANEL_RES_Y 32  // Number of pixels tall of each INDIVIDUAL panel module.
#define PANEL_CHAIN 1   // Total number of panels chained one to another

MatrixPanel_I2S_DMA* dma_display = nullptr;

uint16_t myBLACK;
uint16_t myWHITE;
uint16_t myRED;
uint16_t myGREEN;
uint16_t myBLUE;

void setup() {
  Serial.begin(115200);
  
  // Configurations for Display
  HUB75_I2S_CFG mxconfig(
    PANEL_RES_X,  // module width
    PANEL_RES_Y,  // module height
    PANEL_CHAIN   // Chain length
  );

  mxconfig.double_buff = true;
  mxconfig.gpio.e = 32;
  mxconfig.clkphase = false;

  // Display Setup
  dma_display = new MatrixPanel_I2S_DMA(mxconfig);
  dma_display->begin();
  dma_display->setBrightness8(255);  //0-255
  
  // Define colors
  myBLACK = dma_display->color565(0, 0, 0);
  myWHITE = dma_display->color565(255, 255, 255);
  myRED = dma_display->color565(255, 0, 0);
  myGREEN = dma_display->color565(0, 255, 0);
  myBLUE = dma_display->color565(0, 0, 255);

  // Print dot at each coordinate
  for (int y = 0; y < PANEL_RES_Y; y++) {
    for (int x = 0; x < PANEL_RES_X * PANEL_CHAIN; x++) {
      dma_display->drawPixel(x, y, myRED);
      // Print coordinates
      Serial.print("(");
      Serial.print(x);
      Serial.print(",");
      Serial.print(y);
      Serial.print(") ");
      delay(100); // Delay 500ms before printing the next dot
    }
    Serial.println(); // Move to the next line after each row
  }
}

void loop() {
  // Nothing here, just printing dots in setup()
}
`
here's debug [image]

[image]

here's in visual led matrix

for red color I can fix it but for coordinate i need help 
thanks

mjmokhtar, 2024-03-27:
is sombody can help me 
i still stuck

board707, 2024-03-27:
I am in doubt that anybody can help without having the panel for tests.
How many such panels do you have? Perhaps buying a different model could be a solution?

mjmokhtar, 2024-03-27:
@board707 actualy i just wonder how to fix this code by my self but I still stuck , i can't buying for solution because it new feature led but still no code cant explain this what this happen

i need help Mr @board707 

if you don't mine i will send my email to you, and we can solve it together if you don't mind

board707, 2024-03-27:
Unfortunately, I do not use ESP32 with HUB75 RGB matrices.
If you don’t have to use ESP32, we can try your panel with boards based on Pico RP2040 or STM32 controllers.

mjmokhtar, 2024-03-28:
https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-DMA/assets/108962994/63b1c265-f553-4b4e-939c-0f776665b895

i found another condition 
my HUB75 have pin a,b,c,d and pin E
in another hub75 they don't have pin E

so I tried to measure the voltage in pin E there is no signal, different from the another pin a,b,c, and d have voltage 1.12v a data

I hope this data can we figure it out together

board707, 2024-03-28:
> my HUB75 have pin a,b,c,d and pin E
> in another hub75 they don't have pin E

There is no magic in it. It depends only on number of scan lines. Panels with 16 scans and less doesn't need a E line.

mjmokhtar, 2024-03-28:
i know but my p4 led panel matrix need pin e, cause 32 scans 
what I would do for this

board707, 2024-03-28:
> my p4 led panel

"P4" says nothing about panel type, please be more specific and use informative labels - for example "64x32 s32 panel". And "p4" is completely unnecessary for it :) 

> my hub is different

There is no difference in the fact that your HUB has E pin. This is common to all 32s panels

mjmokhtar, 2024-03-28:
yes i have it 
this panel will full https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-DMA/issues/516 same like this

> #516
> 
> its same like this

board707, 2024-03-28:
So the panel works? What is your problem?

mjmokhtar, 2024-03-28:
it cut my content display when it parallel that why I ask how can be to solve this for multi panel

board707, 2024-03-28:
> not work for 64x32

Please understand that technically your panel is not a 64x32, it is a half of 64x64 panel. So you have to setup and use it as 64x64. Do not try to setup it as 64x32 - nothing will come of this.
 If you need a several panels, chained horizontally, you can setup it as chain of 64x64 panels - it should work. If you need a several horizontal rows of panels - the setup is slightly different. You need to connect R0 G0 B0 lines to corresponding pins of upper row of panels, and R1 G1 B1 lines to R0 G0 B0 pins of lower row

mjmokhtar, 2024-03-28:
yes that why i ask how to try it 64x32 because when I want to print something that cut even I use 64x64 and been chain panel it cause set 64x64 my panel thing is one panel not chain panel 

you said about R0G0B0 
can you tell me how to set up that ?

board707, 2024-03-28:
> that why i ask how to try it 64x32

as 64x32 no way. use it as 64x64

> even I use 64x64 and been chain panel it cause set 64x64 my panel thing is one panel not chain panel

I don't understand... Do you use a google translate?

mjmokhtar, 2024-03-28:
yes i use translator app
i know its not helping but I just share my problem 
I still want use 64x32 code and I try figure it out to found it 

event you said 64x64 code, I think there is way to use 64x32 panel without problem and that solution is change library code
that's why I ask

thank you

board707, 2024-03-28:
> I think there is way to use 64x32 panel without problem and that solution is change library code
> that's why I ask

Do you expect that somebody will redesign the library for you? - as I said before, for this one at least need to have such a panel.
More realistic if you would try to edit the code yourself and then offer a PR with your changes.

board707, 2024-03-28:
However, the simplest thing is to use the panel as 64x64, as I described above. This does not require editing the library

board707, 2024-06-05:
Could you test this code with two 64x32 panels chained and show the video?
```
#include "ESP32-HUB75-MatrixPanel-I2S-DMA.h"

//************************************************************ Display Config Declaration

#define PANEL_RES_X 64  // Number of pixels wide of each INDIVIDUAL panel module.
#define PANEL_RES_Y 32  // Number of pixels tall of each INDIVIDUAL panel module.
#define PANEL_CHAIN 2   // Total number of panels chained one to another

MatrixPanel_I2S_DMA* dma_display = nullptr;

uint16_t myBLACK;
uint16_t myWHITE;
uint16_t myRED;
uint16_t myGREEN;
uint16_t myBLUE;

void setup() {
  Serial.begin(115200);
  
  // Configurations for Display
  HUB75_I2S_CFG mxconfig(
    PANEL_RES_X,  // module width
    PANEL_RES_Y,  // module height
    PANEL_CHAIN   // Chain length
  );

  mxconfig.double_buff = true;
  mxconfig.gpio.e = 32;
  mxconfig.clkphase = false;

  // Display Setup
  dma_display = new MatrixPanel_I2S_DMA(mxconfig);
  dma_display->begin();
  dma_display->setBrightness8(255);  //0-255
  
  // Define colors
  myBLACK = dma_display->color565(0, 0, 0);
  myWHITE = dma_display->color565(255, 255, 255);
  myRED = dma_display->color565(255, 0, 0);
  myGREEN = dma_display->color565(0, 255, 0);
  myBLUE = dma_display->color565(0, 0, 255);

  // Print dot at each coordinate
  for (int y = 0; y < PANEL_RES_Y; y++) {
    for (int x = 0; x < PANEL_RES_X * PANEL_CHAIN; x++) {
      dma_display->drawPixel(x, y, myRED);
     delay(30); // Delay 500ms before printing the next dot
    }
    Serial.println(); // Move to the next line after each row
  }
}

void loop() {
  // Nothing here, just printing dots in setup()
}
```

mjmokhtar, 2024-06-05:
here is video

https://github.com/mrcodetastic/ESP32-HUB75-MatrixPanel-DMA/assets/108962994/f11ce173-7d83-47e6-845d-750efda85f22

something happens when the left panel should display pixels but none appear instead the one on the right appears

i don't know, When we want to display red, what appears instead is a green pixel 
maybe need to set the initial value for red

board707, 2024-06-05:
Thank you

> When we want to display red, what appears instead is a green pixel

Perhaps the panel just gas a color channels mixed? Did you try to output a green, red, blue?

mjmokhtar, 2024-06-05:
I've tried changing colors
When I want green, blue appears instead
When I want blue, red appears instead

```
    #define R1_PIN_DEFAULT  25
    #define G1_PIN_DEFAULT  26
    #define B1_PIN_DEFAULT  27
    #define R2_PIN_DEFAULT  14
    #define G2_PIN_DEFAULT  12
    #define B2_PIN_DEFAULT  13

    #define A_PIN_DEFAULT   23
    #define B_PIN_DEFAULT   22
    #define C_PIN_DEFAULT   5
    #define D_PIN_DEFAULT   19
    #define E_PIN_DEFAULT   32 // IMPORTANT: Change to a valid pin if using a 64x64px panel. 32
              
    #define LAT_PIN_DEFAULT 21
    #define OE_PIN_DEFAULT  33
    #define CLK_PIN_DEFAULT 18
```

I used this pin and PIN E need request because when E still -1 that panel doesn't work 

my senior use stm32 and he give me some code like this

```
#if MATRIX_SCANROW==MATRIX_SCANROWS_32
		row = y;
//		lines = x;

		if (x < MATRIX_PANEL_WIDTH)
		{
			lines = x;
		}
		else if (x < (MATRIX_PANEL_WIDTH * 2))
		{
			lines = (x % MATRIX_PANEL_WIDTH) + MATRIX_PANEL_WIDTH;
			busNumber = 1;
		}
		else
		{
			lines = (x % MATRIX_PANEL_WIDTH) + MATRIX_PANEL_WIDTH;
		}
#endif	//if MATRIX_SCANROW==MATRIX_SCANROWS_32
```
he said we need virtual matrix for this condition
and color condition change from 0b001 to 0b100 (I still didn't get it from this initial)

board707, 2024-06-05:
> When I want green, blue appears instead
When I want blue, red appears instead

So your panels has a different color order, like GBR instead of RGB. It all.

> he said we need virtual matrix for this condition

Virtual matrix can't resolve the pin E problem.

mjmokhtar, 2024-06-05:
currently, i have 3 panels right now

https://github.com/mrcodetastic/ESP32-HUB75-MatrixPanel-DMA/assets/108962994/fa583019-5c19-4945-a32a-8f0b29f79371

I just add one panel and just change color become blue

board707, 2024-06-05:
Please chain a three panels and run the code from [this post ](https://github.com/mrcodetastic/ESP32-HUB75-MatrixPanel-DMA/issues/540#issuecomment-2149766952) again

mjmokhtar, 2024-06-05:
result same like last video

> currently, i have 3 panels right now
> video4.mp4
> 
> I just add one panel and just change color become blue

board707, 2024-06-05:
I think we need a virtual matrix with double height 
```
 HUB75_I2S_CFG mxconfig(
    PANEL_RES_X,            // DO NOT CHANGE THIS
    PANEL_RES_Y*2,            // DO NOT CHANGE THIS 
    NUM_ROWS * NUM_COLS         // DO NOT CHANGE THIS
    //,_pins            // Uncomment to enable custom pins
  );
```

and use coordinates transform something like this:
```
if ( coords.x & 64) {
  coords.x -= 64;
  coords.y +=32;
}
```

mjmokhtar, 2024-06-06:
I'm sorry, I've try 3 times, the code doesn't display anything just blank

here I write code for virtual matrix which you have suggested

```
/* Use a custom Virtual Display class to re-map co-ordinates such that they draw
   correctly on a 32x16 1/4 Scan panel (or chain of such panels).
*/
#include "ESP32-VirtualMatrixPanel-I2S-DMA.h"

/* ================================================== */
// Define custom class derived from VirtualMatrixPanel
class EightPxBasePanel : public VirtualMatrixPanel
{
  public:
    using VirtualMatrixPanel::VirtualMatrixPanel; // inherit VirtualMatrixPanel's constructor(s)

  protected:

    VirtualCoords getCoords(int16_t x, int16_t y);  // custom getCoords() method for specific pixel mapping

};

inline VirtualCoords EightPxBasePanel ::getCoords(int16_t x, int16_t y) {

  coords = VirtualMatrixPanel::getCoords(x, y); // first call base class method to update coords for chaining approach

  if ( coords.x == -1 || coords.y == -1 ) { // Co-ordinates go from 0 to X-1 remember! width() and height() are out of range!
    return coords;
  }

if ( coords.x & 64) {
  coords.x -= 64;
  coords.y +=32;
}
}
/* ================================================== */
// Panel configuration
#define PANEL_RES_X 64 // Number of pixels wide of each INDIVIDUAL panel module. 
#define PANEL_RES_Y 32 // Number of pixels tall of each INDIVIDUAL panel module.

#define NUM_ROWS 1 // Number of rows of chained INDIVIDUAL PANELS
#define NUM_COLS 1 // Number of INDIVIDUAL PANELS per ROW

// ^^^ NOTE: DEFAULT EXAMPLE SETUP IS FOR A CHAIN OF TWO x 1/8 SCAN PANELS

// Change this to your needs, for details on VirtualPanel pls read the PDF!
#define SERPENT true
#define TOPDOWN false
#define VIRTUAL_MATRIX_CHAIN_TYPE CHAIN_TOP_RIGHT_DOWN

// placeholder for the matrix object
MatrixPanel_I2S_DMA *dma_display = nullptr;

// placeholder for the virtual display object
EightPxBasePanel   *FourScanPanel = nullptr;

/******************************************************************************
   Setup!
 ******************************************************************************/
void setup()
{
  delay(250);

  Serial.begin(115200);
  Serial.println(""); Serial.println(""); Serial.println("");
  Serial.println("*****************************************************");
  Serial.println("*         1/32 Scan Panel Demonstration              *");
  Serial.println("*****************************************************");

  /*
       // 62x32 1/8 Scan Panels don't have a D and E pin!

       HUB75_I2S_CFG::i2s_pins _pins = {
        R1_PIN, G1_PIN, B1_PIN, R2_PIN, G2_PIN, B2_PIN,
        A_PIN, B_PIN, C_PIN, D_PIN, E_PIN,
        LAT_PIN, OE_PIN, CLK_PIN
       };
  */
  HUB75_I2S_CFG mxconfig(
    PANEL_RES_X / 2 ,            // DO NOT CHANGE THIS
    PANEL_RES_Y * 2,            // DO NOT CHANGE THIS
    NUM_ROWS * NUM_COLS         // DO NOT CHANGE THIS
    //,_pins            // Uncomment to enable custom pins
  );

  mxconfig.clkphase = false; // Change this if you see pixels showing up shifted wrongly by one column the left or right.

  //mxconfig.driver   = HUB75_I2S_CFG::FM6126A;     // in case that we use panels based on FM6126A chip, we can set it here before creating MatrixPanel_I2S_DMA object

  // OK, now we can create our matrix object
  dma_display = new MatrixPanel_I2S_DMA(mxconfig);

  // let's adjust default brightness to about 75%
  dma_display->setBrightness8(255);    // range is 0-255, 0 - 0%, 255 - 100%

  // Allocate memory and start DMA display
  if ( not dma_display->begin() )
    Serial.println("****** !KABOOM! I2S memory allocation failed ***********");

  dma_display->clearScreen();
  delay(500);

  // create FourScanPanellay object based on our newly created dma_display object
  FourScanPanel = new EightPxBasePanel ((*dma_display), NUM_ROWS, NUM_COLS, PANEL_RES_X, PANEL_RES_Y,    VIRTUAL_MATRIX_CHAIN_TYPE);

  // THE IMPORTANT BIT BELOW!
  // FOUR_SCAN_16PX_HIGH
  // FOUR_SCAN_32PX_HIGH
// **** You don't need to set PhysicalPanelScanRate when use your own virtual panel class
  //FourScanPanel->setPhysicalPanelScanRate(FOUR_SCAN_16PX_HIGH);
}

// Test the pixel mapping - fill the panel pixel by pixel
void loop() {

  for (int i = 0; i < FourScanPanel->height(); i++)
  {
    for (int j = 0; j < FourScanPanel->width(); j++)
    {

      FourScanPanel->drawPixel(j, i, FourScanPanel->color565(255, 0, 0));
      delay(30);
    }
  }

  delay(2000);
  dma_display->clearScreen();

} // end loop
```

mjmokhtar, 2024-06-06:
hi @board707 
I am so sorry to disturb you, any suggestions what should I do now?

board707, 2024-06-06:
I don't see anything incorrect in your code.
Try to change this:
```
 HUB75_I2S_CFG mxconfig(
    PANEL_RES_X ,            // DO NOT CHANGE THIS
    PANEL_RES_Y*2,            // DO NOT CHANGE THIS 
    NUM_ROWS * NUM_COLS         // DO NOT CHANGE THIS
    //,_pins            // Uncomment to enable custom pins
  );
```

mjmokhtar, 2024-06-06:
I've tried it and the dot matrix doesn't appear, even by changing the color it doesn't appear

board707, 2024-06-06:
Could you test again your code that produce a video2 from this message https://github.com/mrcodetastic/ESP32-HUB75-MatrixPanel-DMA/issues/540#issuecomment-2149701599 ?
If it still works, please show that code in full.

mjmokhtar, 2024-06-06:
actualy i never using that code 
i just using 

```
#include "ESP32-HUB75-MatrixPanel-I2S-DMA.h"

//************************************************************ Display Config Declaration

#define PANEL_RES_X 64  // Number of pixels wide of each INDIVIDUAL panel module.
#define PANEL_RES_Y 32  // Number of pixels tall of each INDIVIDUAL panel module.
#define PANEL_CHAIN 1   // Total number of panels chained one to another

MatrixPanel_I2S_DMA* dma_display = nullptr;

uint16_t myBLACK;
uint16_t myWHITE;
uint16_t myRED;
uint16_t myGREEN;
uint16_t myBLUE;

void setup() {
  Serial.begin(115200);
  
  // Configurations for Display
  HUB75_I2S_CFG mxconfig(
    PANEL_RES_X,  // module width
    PANEL_RES_Y,  // module height
    PANEL_CHAIN   // Chain length
  );

  mxconfig.double_buff = true;
  mxconfig.gpio.e = 32;
  mxconfig.clkphase = false;

  // Display Setup
  dma_display = new MatrixPanel_I2S_DMA(mxconfig);
  dma_display->begin();
  dma_display->setBrightness8(255);  //0-255
  
  // Define colors
  myBLACK = dma_display->color565(0, 0, 0);
  myWHITE = dma_display->color565(255, 255, 255);
  myRED = dma_display->color565(255, 0, 0);
  myGREEN = dma_display->color565(0, 255, 0);
  myBLUE = dma_display->color565(0, 0, 255);

  // Print dot at each coordinate
  for (int y = 0; y < PANEL_RES_Y; y++) {
    for (int x = 0; x < PANEL_RES_X * PANEL_CHAIN; x++) {
      dma_display->drawPixel(x, y, myRED);
      // Print coordinates
      Serial.print("(");
      Serial.print(x);
      Serial.print(",");
      Serial.print(y);
      Serial.print(") ");
      delay(100); // Delay 500ms before printing the next dot
    }
    Serial.println(); // Move to the next line after each row
  }
}

void loop() {
  // Nothing here, just printing dots in setup()
}
```

board707, 2024-06-06:
I meant  your comment below:

>  when i change panel y become times 2 (32*2 =64)
```
   PANEL_RES_Y*2,            // DO NOT CHANGE THIS
```
> when i use this he be come full dot in this display

And as illustration to it you shown a `video2.mp4` file where the first panel fulfilled completely.

mjmokhtar, 2024-06-06:
actually that is just my suggestion to change that but I haven't done it yet

mjmokhtar, 2024-06-06:
video2.mp4 is when panel_res_y = 64 

> //************************************************************ Display Config Declaration
>
> #define PANEL_RES_X 64  // Number of pixels wide of each INDIVIDUAL panel module.
> #define PANEL_RES_Y 32 [...]

board707, 2024-06-06:
I don't understand why it is nothing on the panel. Perhaps it might not work correctly - but at least a few leds should be on. 
Please test again tour connections.

If the problem is not with the connections, then I don't know how else to help you.

Where are you from?

mjmokhtar, 2024-06-06:
i from indonesia

```
#include "ESP32-HUB75-MatrixPanel-I2S-DMA.h"

/* Use a custom Virtual Display class to re-map co-ordinates such that they draw
   correctly on a 32x16 1/4 Scan panel (or chain of such panels).
*/
#include "ESP32-VirtualMatrixPanel-I2S-DMA.h"

/* ================================================== */
// Define custom class derived from VirtualMatrixPanel
class EightPxBasePanel : public VirtualMatrixPanel
{
  public:
    using VirtualMatrixPanel::VirtualMatrixPanel; // inherit VirtualMatrixPanel's constructor(s)

  protected:

    VirtualCoords getCoords(int16_t x, int16_t y);  // custom getCoords() method for specific pixel mapping

};

inline VirtualCoords EightPxBasePanel ::getCoords(int16_t x, int16_t y) {

  coords = VirtualMatrixPanel::getCoords(x, y); // first call base class method to update coords for chaining approach

  if ( coords.x == -1 || coords.y == -1 ) { // Co-ordinates go from 0 to X-1 remember! width() and height() are out of range!
    return coords;
  } 

  if ( coords.x & 64) {
    coords.x -= 64;
    coords.y +=32;
  }
}
/* ================================================== */
// Panel configuration
#define PANEL_RES_X 64 // Number of pixels wide of each INDIVIDUAL panel module. 
#define PANEL_RES_Y 32 // Number of pixels tall of each INDIVIDUAL panel module.

#define NUM_ROWS 1 // Number of rows of chained INDIVIDUAL PANELS
#define NUM_COLS 1 // Number of INDIVIDUAL PANELS per ROW

// ^^^ NOTE: DEFAULT EXAMPLE SETUP IS FOR A CHAIN OF TWO x 1/8 SCAN PANELS

// Change this to your needs, for details on VirtualPanel pls read the PDF!
#define SERPENT true
#define TOPDOWN false
#define VIRTUAL_MATRIX_CHAIN_TYPE CHAIN_TOP_RIGHT_DOWN

// placeholder for the matrix object
MatrixPanel_I2S_DMA *dma_display = nullptr;

// placeholder for the virtual display object
EightPxBasePanel   *FourScanPanel = nullptr;

/******************************************************************************
   Setup!
 ******************************************************************************/
void setup()
{
  delay(250);

  Serial.begin(115200);
  Serial.println(""); Serial.println(""); Serial.println("");
  Serial.println("*****************************************************");
  Serial.println("*         1/32 Scan Panel Demonstration              *");
  Serial.println("*****************************************************");

  /*
       // 62x32 1/8 Scan Panels don't have a D and E pin!

       HUB75_I2S_CFG::i2s_pins _pins = {
        R1_PIN, G1_PIN, B1_PIN, R2_PIN, G2_PIN, B2_PIN,
        A_PIN, B_PIN, C_PIN, D_PIN, E_PIN,
        LAT_PIN, OE_PIN, CLK_PIN
       };
  */
  HUB75_I2S_CFG mxconfig(
    PANEL_RES_X ,            // DO NOT CHANGE THIS
    PANEL_RES_Y * 2,            // DO NOT CHANGE THIS
    NUM_ROWS * NUM_COLS         // DO NOT CHANGE THIS
    //,_pins            // Uncomment to enable custom pins
  );

  mxconfig.clkphase = false; // Change this if you see pixels showing up shifted wrongly by one column the left or right.
  mxconfig.gpio.e = 32;
  //mxconfig.driver   = HUB75_I2S_CFG::FM6126A;     // in case that we use panels based on FM6126A chip, we can set it here before creating MatrixPanel_I2S_DMA object
  
  // OK, now we can create our matrix object
  dma_display = new MatrixPanel_I2S_DMA(mxconfig);

  // let's adjust default brightness to about 75%
  dma_display->setBrightness8(255);    // range is 0-255, 0 - 0%, 255 - 100%

  // Allocate memory and start DMA display
  if ( not dma_display->begin() )
    Serial.println("****** !KABOOM! I2S memory allocation failed ***********");

  dma_display->clearScreen();
  delay(500);

  // create FourScanPanellay object based on our newly created dma_display object
  FourScanPanel = new EightPxBasePanel ((*dma_display), NUM_ROWS, NUM_COLS, PANEL_RES_X, PANEL_RES_Y, VIRTUAL_MATRIX_CHAIN_TYPE);

  // THE IMPORTANT BIT BELOW!
  // FOUR_SCAN_16PX_HIGH
  // FOUR_SCAN_32PX_HIGH
// **** You don't need to set PhysicalPanelScanRate when use your own virtual panel class
  //FourScanPanel->setPhysicalPanelScanRate(FOUR_SCAN_16PX_HIGH);
}

// Test the pixel mapping - fill the panel pixel by pixel
void loop() {

  for (int i = 0; i < FourScanPanel->height(); i++)
  {
    for (int j = 0; j < FourScanPanel->width(); j++)
    {

      FourScanPanel->drawPixel(j, i, FourScanPanel->color565(255, 255, 255));
      // Print coordinates
        Serial.print("(");
        Serial.print(j);
        Serial.print(",");
        Serial.print(i);
        Serial.print(") ");
      delay(30);
    }
  }

  delay(2000);
  dma_display->clearScreen();

} // end loop
```

please read my code is something wrong ?

actually when I try using

board707, 2024-06-06:
I'll take a look and write later

mjmokhtar, 2024-06-06:
https://github.com/mrcodetastic/ESP32-HUB75-MatrixPanel-DMA/assets/108962994/5811b9d0-5185-4ff2-b8b2-6707a2cf852d

i remove 

> if ( coords.x & 64) {
>     coords.x -= 64;
>     coords.y +=32;
>   }

and this what i got
but when I didn't remove it panel is blank

board707, 2024-06-06:
Please show a full code

mjmokhtar, 2024-06-06:
here is full code

```
#include "ESP32-HUB75-MatrixPanel-I2S-DMA.h"

/* Use a custom Virtual Display class to re-map co-ordinates such that they draw
   correctly on a 32x16 1/4 Scan panel (or chain of such panels).
*/
#include "ESP32-VirtualMatrixPanel-I2S-DMA.h"

/* ================================================== */
// Define custom class derived from VirtualMatrixPanel
class EightPxBasePanel : public VirtualMatrixPanel
{
  public:
    using VirtualMatrixPanel::VirtualMatrixPanel; // inherit VirtualMatrixPanel's constructor(s)

  protected:

    VirtualCoords getCoords(int16_t x, int16_t y);  // custom getCoords() method for specific pixel mapping

};

inline VirtualCoords EightPxBasePanel ::getCoords(int16_t x, int16_t y) {

  coords = VirtualMatrixPanel::getCoords(x, y); // first call base class method to update coords for chaining approach

  if ( coords.x == -1 || coords.y == -1 ) { // Co-ordinates go from 0 to X-1 remember! width() and height() are out of range!
    return coords;
  } 

  // if ( coords.x & 64) {
  //   coords.x -= 64;
  //   coords.y +=32;
  // }
}
/* ================================================== */
// Panel configuration
#define PANEL_RES_X 64 // Number of pixels wide of each INDIVIDUAL panel module. 
#define PANEL_RES_Y 32 // Number of pixels tall of each INDIVIDUAL panel module.

#define NUM_ROWS 1 // Number of rows of chained INDIVIDUAL PANELS
#define NUM_COLS 2 // Number of INDIVIDUAL PANELS per ROW
#define PANEL_CHAIN NUM_ROWS*NUM_COLS

// ^^^ NOTE: DEFAULT EXAMPLE SETUP IS FOR A CHAIN OF TWO x 1/8 SCAN PANELS

// Change this to your needs, for details on VirtualPanel pls read the PDF!
#define SERPENT true
#define TOPDOWN false
#define VIRTUAL_MATRIX_CHAIN_TYPE CHAIN_TOP_RIGHT_DOWN

// placeholder for the matrix object
MatrixPanel_I2S_DMA *dma_display = nullptr;

// placeholder for the virtual display object
EightPxBasePanel   *FourScanPanel = nullptr;

/******************************************************************************
   Setup!
 ******************************************************************************/
void setup()
{
  delay(250);

  Serial.begin(115200);
  Serial.println(""); Serial.println(""); Serial.println("");
  Serial.println("*****************************************************");
  Serial.println("*         1/32 Scan Panel Demonstration              *");
  Serial.println("*****************************************************");

  /*
       // 62x32 1/8 Scan Panels don't have a D and E pin!

       HUB75_I2S_CFG::i2s_pins _pins = {
        R1_PIN, G1_PIN, B1_PIN, R2_PIN, G2_PIN, B2_PIN,
        A_PIN, B_PIN, C_PIN, D_PIN, E_PIN,
        LAT_PIN, OE_PIN, CLK_PIN
       };
  */
  HUB75_I2S_CFG mxconfig(
    PANEL_RES_X ,            // DO NOT CHANGE THIS
    PANEL_RES_Y * 2,            // DO NOT CHANGE THIS
    PANEL_CHAIN         // DO NOT CHANGE THIS
    //,_pins            // Uncomment to enable custom pins
  );

  mxconfig.clkphase = false; // Change this if you see pixels showing up shifted wrongly by one column the left or right.
  mxconfig.gpio.e = 32;
  //mxconfig.driver   = HUB75_I2S_CFG::FM6126A;     // in case that we use panels based on FM6126A chip, we can set it here before creating MatrixPanel_I2S_DMA object
  
  // OK, now we can create our matrix object
  dma_display = new MatrixPanel_I2S_DMA(mxconfig);

  // let's adjust default brightness to about 75%
  dma_display->setBrightness8(192);    // range is 0-255, 0 - 0%, 255 - 100%

  // Allocate memory and start DMA display
  if ( not dma_display->begin() )
    Serial.println("****** !KABOOM! I2S memory allocation failed ***********");

  

  // create FourScanPanellay object based on our newly created dma_display object
  FourScanPanel = new EightPxBasePanel ((*dma_display), NUM_ROWS, NUM_COLS, PANEL_RES_X, PANEL_RES_Y, VIRTUAL_MATRIX_CHAIN_TYPE);
  
  dma_display->clearScreen();
  delay(500);
  // THE IMPORTANT BIT BELOW!
  // FOUR_SCAN_16PX_HIGH
  // FOUR_SCAN_32PX_HIGH
// **** You don't need to set PhysicalPanelScanRate when use your own virtual panel class
  //FourScanPanel->setPhysicalPanelScanRate(FOUR_SCAN_16PX_HIGH);
}

// Test the pixel mapping - fill the panel pixel by pixel
void loop() {

  for (int i = 0; i < FourScanPanel->height(); i++)
  {
    for (int j = 0; j < FourScanPanel->width(); j++)
    {

      FourScanPanel->drawPixel(j, i, FourScanPanel->color565(255, 255, 255));
      // Print coordinates
        Serial.print("(");
        Serial.print(j);
        Serial.print(",");
        Serial.print(i);
        Serial.print(") ");
      delay(30);
    }
  }

  delay(2000);
  dma_display->clearScreen();

} // end loop
```
I just use 1 panel for code, The second panel should do the same thing but instead, it is blank

mjmokhtar, 2024-06-06:
> HUB75_I2S_CFG mxconfig(
>     PANEL_RES_X ,            // DO NOT CHANGE THIS
>     PANEL_RES_Y * 2,            // DO NOT CHANGE THIS
>     PANEL_CHAIN         // DO NOT CHANGE THIS
>     //,_pins            // Uncomment to enable custom pins
>   );

I still save this config

board707, 2024-06-06:
I may have found a bug, try this
```
#include "ESP32-HUB75-MatrixPanel-I2S-DMA.h"

/* Use a custom Virtual Display class to re-map co-ordinates such that they draw
   correctly on a 32x16 1/4 Scan panel (or chain of such panels).
*/
#include "ESP32-VirtualMatrixPanel-I2S-DMA.h"

/* ================================================== */
// Define custom class derived from VirtualMatrixPanel
class EightPxBasePanel : public VirtualMatrixPanel
{
  public:
    using VirtualMatrixPanel::VirtualMatrixPanel; // inherit VirtualMatrixPanel's constructor(s)

  protected:

    VirtualCoords getCoords(int16_t x, int16_t y);  // custom getCoords() method for specific pixel mapping

};

inline VirtualCoords EightPxBasePanel ::getCoords(int16_t x, int16_t y) {

  coords = VirtualMatrixPanel::getCoords(x, y); // first call base class method to update coords for chaining approach

  if ( coords.x == -1 || coords.y == -1 ) { // Co-ordinates go from 0 to X-1 remember! width() and height() are out of range!
    return coords;
  } 

   if ( coords.x & 64) {
     coords.x -= 64;
     coords.y +=32;
   }
return coords;

}
/* ================================================== */
// Panel configuration
#define PANEL_RES_X 64 // Number of pixels wide of each INDIVIDUAL panel module. 
#define PANEL_RES_Y 32 // Number of pixels tall of each INDIVIDUAL panel module.

#define NUM_ROWS 1 // Number of rows of chained INDIVIDUAL PANELS
#define NUM_COLS 1 // Number of INDIVIDUAL PANELS per ROW
#define PANEL_CHAIN NUM_ROWS*NUM_COLS

// ^^^ NOTE: DEFAULT EXAMPLE SETUP IS FOR A CHAIN OF TWO x 1/8 SCAN PANELS

// Change this to your needs, for details on VirtualPanel pls read the PDF!
#define SERPENT true
#define TOPDOWN false
#define VIRTUAL_MATRIX_CHAIN_TYPE CHAIN_TOP_RIGHT_DOWN

// placeholder for the matrix object
MatrixPanel_I2S_DMA *dma_display = nullptr;

// placeholder for the virtual display object
EightPxBasePanel   *FourScanPanel = nullptr;

/******************************************************************************
   Setup!
 ******************************************************************************/
void setup()
{
  delay(250);

  Serial.begin(115200);
  Serial.println(""); Serial.println(""); Serial.println("");
  Serial.println("*****************************************************");
  Serial.println("*         1/32 Scan Panel Demonstration              *");
  Serial.println("*****************************************************");

  /*
       // 62x32 1/8 Scan Panels don't have a D and E pin!

       HUB75_I2S_CFG::i2s_pins _pins = {
        R1_PIN, G1_PIN, B1_PIN, R2_PIN, G2_PIN, B2_PIN,
        A_PIN, B_PIN, C_PIN, D_PIN, E_PIN,
        LAT_PIN, OE_PIN, CLK_PIN
       };
  */
  HUB75_I2S_CFG mxconfig(
    PANEL_RES_X  ,            // DO NOT CHANGE THIS
    PANEL_RES_Y * 2,            // DO NOT CHANGE THIS
    PANEL_CHAIN         // DO NOT CHANGE THIS
    //,_pins            // Uncomment to enable custom pins
  );

  mxconfig.clkphase = false; // Change this if you see pixels showing up shifted wrongly by one column the left or right.
  mxconfig.gpio.e = 32;
  //mxconfig.driver   = HUB75_I2S_CFG::FM6126A;     // in case that we use panels based on FM6126A chip, we can set it here before creating MatrixPanel_I2S_DMA object
  
  // OK, now we can create our matrix object
  dma_display = new MatrixPanel_I2S_DMA(mxconfig);

  // let's adjust default brightness to about 75%
  dma_display->setBrightness8(192);    // range is 0-255, 0 - 0%, 255 - 100%

  // Allocate memory and start DMA display
  if ( not dma_display->begin() )
    Serial.println("****** !KABOOM! I2S memory allocation failed ***********");

  

  // create FourScanPanellay object based on our newly created dma_display object
  FourScanPanel = new EightPxBasePanel ((*dma_display), NUM_ROWS, NUM_COLS, PANEL_RES_X, PANEL_RES_Y, VIRTUAL_MATRIX_CHAIN_TYPE);
  
  dma_display->clearScreen();
  delay(500);
  // THE IMPORTANT BIT BELOW!
  // FOUR_SCAN_16PX_HIGH
  // FOUR_SCAN_32PX_HIGH
// **** You don't need to set PhysicalPanelScanRate when use your own virtual panel class
  //FourScanPanel->setPhysicalPanelScanRate(FOUR_SCAN_16PX_HIGH);
}

// Test the pixel mapping - fill the panel pixel by pixel
void loop() {

  for (int i = 0; i < FourScanPanel->height(); i++)
  {
    for (int j = 0; j < FourScanPanel->width(); j++)
    {

      FourScanPanel->drawPixel(j, i, FourScanPanel->color565(255, 255, 255));
     
      delay(30);
    }
  }

  delay(2000);
  dma_display->clearScreen();

} // end loop
```

mjmokhtar, 2024-06-07:
the result it is same like my video 

> video5.mp4

board707, 2024-06-07:
At least now it show something...

board707, 2024-06-07:
@mjmokhtar 
I apologize for so many failed tests - but I think you understand how difficult it is to configure the panel remotely without being able to test it live. So if you want to get the result, we will have to do a few more tests.

mjmokhtar, 2024-06-07:
you don't have to apologize about failed tests, I really appreciate you help me doing this remote,
is the issue something about parallel i2s ?

board707, 2024-06-07:
> is the issue something about parallel i2s ?

What is the issue? The video5 is not wrong, it exactly as it should be.

Are you ready to next test?
Please edit the last code changing the `getCoords()` this way:
```
inline VirtualCoords EightPxBasePanel ::getCoords(int16_t x, int16_t y) {

  coords = VirtualMatrixPanel::getCoords(x, y); // first call base class method to update coords for chaining approach

  if ( coords.x == -1 || coords.y == -1 ) { // Co-ordinates go from 0 to X-1 remember! width() and height() are out of range!
    return coords;
  } 

   if ( coords.x & 64) {
     coords.x -= 64;
     coords.y +=32;
   }
  coords.x -= (coords.x / (2 * PANEL_RES_X)) * PANEL_RES_X;
return coords;

}
```
please setup the screen as horizontal row of 3 panels:
```
#define NUM_ROWS 1 // Number of rows of chained INDIVIDUAL PANELS
#define NUM_COLS 3 // Number of INDIVIDUAL PANELS per ROW
```

mjmokhtar, 2024-06-07:
here is the result
the matrixs  just prints in left matrix center and right becomes blank

https://github.com/mrcodetastic/ESP32-HUB75-MatrixPanel-DMA/assets/108962994/8ca6fbc9-d8f7-4611-a9aa-0c3896a659b7

board707, 2024-06-07:
Sorry by asking that - don't you forget to setup a screen as 3x1 panels?

Or try to change the cycle in loop:
```
for (int i = 0; i < FourScanPanel->height(); i++)
  {
    for (int j = 0; j < 192; j++)
    {

      FourScanPanel->drawPixel(j, i, FourScanPanel->color565(255, 255, 255));
     
      delay(30);
    }
  }
```

mjmokhtar, 2024-06-07:
yes, the result same like video6

board707, 2024-06-07:
Sorry, I have no more ideas yet

[42 more replies]