mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #544: After 'pio pkg update' my 4 chained panels now just flicker

jdeltoft, 2023-12-18:
Hi, today I got back to a project I wanted to create with this project so I started working on my 4 panel (each 64x64) chain and everything was going well all day.  The repo was several months old, btw.  

I ran into an issue trying to also have the esp32 download a text file locally over wifi, which led me to an issue (https://github.com/espressif/arduino-esp32/issues/6129) which made me try my first ever `pio pkg update`

This is the log of that.  Now I went from 4 beautifully working panels to just random flicker, which based on the error I think makes sense because it ran out of memory (RAM) for the buffers.  

Did I just run out of memory size and I need to optimize my code?  I can try that but was wondering if the latest versions of things are just that much larger or maybe I need to configure something now that I updated?

```
[  1243][E][ESP32-HUB75-MatrixPanel-I2S-DMA.cpp:58] allocateDMAmemory(): [I2S-DMA] CRITICAL ERROR: Not enough memory for requested colour depth! Please reduce pixel_color_depth_bits value.

[  1249][E][ESP32-HUB75-MatrixPanel-I2S-DMA.cpp:59] allocateDMAmemory(): [I2S-DMA] Could not allocate rowBitStruct 27!.

[  4107][E][WiFiClient.cpp:67] fillBuffer(): Not enough memory to allocate buffer
[  4107][E][WiFiClient.cpp:467] read(): fail on fd 48, errno: 11, "No more processes"
Error code: -7
[Disconnected]
```

Here is the upgrade as it ran also:

```
Justins:Led jeltoft$ pio pkg update
Resolving esp32 dependencies...
Platform Manager: Updating espressif32 @ 6.3.2
Platform Manager: Removing espressif32 @ 6.3.2
Platform Manager: espressif32@6.3.2 has been removed!
Platform Manager: Installing platformio/espressif32 @ 6.4.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Platform Manager: espressif32@6.4.0 has been installed!
Library Manager: Updating Adafruit BusIO @ 1.14.1
Library Manager: Removing Adafruit BusIO @ 1.14.1
Library Manager: Adafruit BusIO@1.14.1 has been removed!
Library Manager: Installing adafruit/Adafruit BusIO @ 1.14.5
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Library Manager: Adafruit BusIO@1.14.5 has been installed!
Library Manager: Updating Adafruit GFX Library @ 1.11.7
Library Manager: Removing Adafruit GFX Library @ 1.11.7
Library Manager: Adafruit GFX Library@1.11.7 has been removed!
Library Manager: Installing adafruit/Adafruit GFX Library @ 1.11.9
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Library Manager: Adafruit GFX Library@1.11.9 has been installed!
Library Manager: Resolving dependencies...
Library Manager: Updating ESP32 HUB75 LED MATRIX PANEL DMA Display @ 3.0.9+sha.3d7c2df
git version 2.43.0
remote: Enumerating objects: 2857, done.
remote: Counting objects: 100% (2857/2857), done.
remote: Compressing objects: 100% (1002/1002), done.
remote: Total 2778 (delta 1803), reused 2665 (delta 1718), pack-reused 0
Receiving objects: 100% (2778/2778), 4.50 MiB | 3.11 MiB/s, done.
Resolving deltas: 100% (1803/1803), completed with 60 local objects.
From https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-I2S-DMA
   3d7c2df..61960e1  master     -> origin/master
 * [new tag]         1.0.2      -> 1.0.2
 * [new tag]         1.1.0      -> 1.1.0
 * [new tag]         1.2.2      -> 1.2.2
 * [new tag]         1.2.4      -> 1.2.4
 * [new tag]         2.0.0      -> 2.0.0
 * [new tag]         2.0.6      -> 2.0.6
 * [new tag]         2.0.7      -> 2.0.7
 * [new tag]         3.0.5      -> 3.0.5
Updating 3d7c2df..61960e1
Fast-forward
 .github/workflows/{esp-idf_with-gfx.yml => esp-idf-5.1.2_with-gfx.yml}       | 14 +++++++-------
 .github/workflows/{esp-idf_without-gfx.yml => esp-idf-5.1.2_without-gfx.yml} | 10 +++++-----
 .github/workflows/{pio_build.yml => pio_arduino_build.yml}                   | 11 ++++++-----
 README.md                                                                    | 31 ++++++++++++++-----------------
[... 171 lines cut ...]
Compiling .pio/build/esp32/FrameworkArduino/main.cpp.o
Compiling .pio/build/esp32/FrameworkArduino/stdlib_noniso.c.o
Compiling .pio/build/esp32/FrameworkArduino/wiring_pulse.c.o
Compiling .pio/build/esp32/FrameworkArduino/wiring_shift.c.o
Archiving .pio/build/esp32/libFrameworkArduino.a
Indexing .pio/build/esp32/libFrameworkArduino.a
Linking .pio/build/esp32/firmware.elf
Retrieving maximum program size .pio/build/esp32/firmware.elf
Checking size .pio/build/esp32/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [===       ]  29.3% (used 95908 bytes from 327680 bytes)
Flash: [=======   ]  74.6% (used 977257 bytes from 1310720 bytes)
Building .pio/build/esp32/firmware.bin
esptool.py v4.5.1
Creating esp32 image...
Merged 25 ELF sections
Successfully created esp32 image.
======================================================================================== [SUCCESS] Took 53.35 seconds ========================================================================================

Environment    Status    Duration
-------------  --------  ------------
esp32          SUCCESS   00:00:53.352
========================================================================================= 1 succeeded in 00:00:53.352 =========================================================================================
Justin:Led jeltoft$ pio run -t upload
```

mrfaptastic, 2023-12-18:
What version of the library did you use before? If the old library worked then I'd suggest sticking with that version.

I don't believe memory consumption has grown much over the versions.

jdeltoft, 2023-12-18:
Ok, just tried with

-platform = espressif32
+platform = espressif32@6.3.2

to go back per the log I captured above, and it's working again. Sorry to bug you, and thanks so much for this great library!!!

Here is my album art so far...

[image]